ravijo/ros_openpose #1: Can't compile.

Gryogor, 2019-10-17:
After cloning  the ros_openpose repo and run catkin_make i receive this output:
Base path: /home/gryogor/catkin_ws
Source space: /home/gryogor/catkin_ws/src
Build space: /home/gryogor/catkin_ws/build
Devel space: /home/gryogor/catkin_ws/devel
Install space: /home/gryogor/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/gryogor/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/gryogor/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/gryogor/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/gryogor/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.7.18
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - realsense2_description
-- ~~  - ros_openpose
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- +++ processing catkin package: 'ros_openpose'
-- ==> add_subdirectory(ros_openpose)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found gflags  (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libgflags.so)
-- Found glog    (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libglog.so)
-- ros_openpose: 4 messages, 0 services
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Create Debug Build.
-- realsense2_camera: 2 messages, 0 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gryogor/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/gryogor/catkin_ws/build"
####
[  0%] Built target _ros_openpose_generate_messages_check_deps_Pixel
[  0%] Built target _ros_openpose_generate_messages_check_deps_Frame
[  0%] Built target _ros_openpose_generate_messages_check_deps_BodyPart
[  0%] Built target _ros_openpose_generate_messages_check_deps_Person
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
Scanning dependencies of target testCameraReader
Scanning dependencies of target rosOpenpose
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  2%] Building CXX object ros_openpose/CMakeFiles/testCameraReader.dir/src/testCameraReader.cpp.o
[  2%] Built target _catkin_empty_exported_target
[  2%] Built target diagnostic_msgs_generate_messages_py
[  4%] Building CXX object ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o
[  4%] Built target sensor_msgs_generate_messages_lisp
[  4%] Built target roscpp_generate_messages_lisp
[  6%] Building CXX object ros_openpose/CMakeFiles/testCameraReader.dir/src/cameraReader.cpp.o
[  6%] Built target roscpp_generate_messages_nodejs
[  6%] Built target rosgraph_msgs_generate_messages_lisp
[  6%] Built target rosgraph_msgs_generate_messages_py
[... 219 lines cut ...]
[ 63%] Generating EusLisp code from ros_openpose/Person.msg
[ 65%] Generating Javascript code from ros_openpose/Person.msg
[ 68%] Generating EusLisp code from ros_openpose/Frame.msg
[ 70%] Generating Javascript code from ros_openpose/Frame.msg
[ 72%] Built target ros_openpose_generate_messages_eus
[ 72%] Built target ros_openpose_generate_messages_nodejs
[ 75%] Generating Lisp code from ros_openpose/BodyPart.msg
[ 79%] Built target realsense2_camera_generate_messages_cpp
[ 81%] Generating Lisp code from ros_openpose/Pixel.msg
[ 90%] Built target realsense2_camera
[ 90%] Built target realsense2_camera_generate_messages
[ 93%] Generating Lisp code from ros_openpose/Person.msg
[ 95%] Generating Lisp code from ros_openpose/Frame.msg
[ 95%] Built target ros_openpose_generate_messages_lisp
[ 95%] Built target ros_openpose_generate_messages
[ 97%] Linking CXX executable /home/gryogor/catkin_ws/devel/lib/ros_openpose/testCameraReader
CMakeFiles/Makefile2:1593: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 97%] Built target testCameraReader
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Does anyone know how to fix it?

ravijo, 2019-10-17:
Hi @Gryogor 

I quickly looked at the error and found the following-
```
error: no match for ‘operator=’ (operand types are ‘op::Matrix’ and ‘const cv::Mat’)
datumPtr->cvInputData = colorImage;
```
I highly doubt that you are using a really old version of OpenPose. Can you please install OpenPose version 1.5.1? _The latest version of OpenPose should also work fine._

Gryogor, 2019-10-17:
Thank you for the answer. I will try it. But how does your wrapper checks, which openpose to choose? Because I have 2 versions installed in different folders.

ravijo, 2019-10-17:
> how does your wrapper checks, which openpose to choose? 

It uses `find_package` as shown below:

https://github.com/ravijo/ros_openpose/blob/709b96287daab9c429397d4278d22bee64fb2763/CMakeLists.txt#L25

> I have 2 versions installed in different folders.

As I said earlier, `find_package` must be able to located the latest OpenPose. To do that you must copy the compiled files into the system directories. You need to go inside the build directory of OpenPose and then execute the following command-

```
cd build/
sudo make install
```

You can look at the detailed instructions [here](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/deployment.md#cmake-windows-ubuntu-and-mac).

Gryogor, 2019-10-18:
Tried the most recent commit and version  1.5.1. Still gives the same error.

Gryogor, 2019-10-18:
@ravijo 
Okey. I tried to install everything on completely clean ubuntu 16.04. This time everything worked without a problem. I am not sure what is the problem, but maybe older versions of installed openpose might be creating a problem.

ravijo, 2019-10-19:
Hi @Gryogor 

> I tried to install everything on completely clean ubuntu 16.04. This time everything worked without a problem. 

I am glad that it worked. 

> maybe older versions of installed openpose might be creating a problem.

Yes, you are right. You can't have two versions of the same software installed system directories (globally). The command `sudo make install` basically overwrites your exiting OpenPose. Furthermore, I don't see any benefit of having multiple versions of OpenPose in one computer. 

Cheers

closed 2019-10-19 by ravijo