ravijo/ros_openpose #2: Question about other cameras

Gryogor, 2019-10-24:
Hello. Are you planning on implementing the support of ordinary RGB cameras for only 2D body image?

ravijo, 2019-10-24:
Hi @Gryogor 

Thanks for reminding me. It is difficult to obtain and rely on depth information acquired by a single RGB camera. There are certain limitations. I do have plan to support (and provide pixel values only) an ordinary RGB camera. However, I am a bit busy preparing for IEEE IROS 2019.

Cheers

Gryogor, 2019-10-24:
Ok. Thank you for the answer.

ravijo, 2019-10-24:
@Gryogor: Thanks for this wonderful idea. I have added enhancement (new feature request) tag on this issue. I will be more than happy to look over it.

Ceepert, 2020-02-27:
> @Gryogor: Thanks for this wonderful idea. I have added enhancement (new feature request) tag on this issue. I will be more than happy to look over it.

Did anything happen on this subject yet? I wanted to use a normal webcam with this wrapper

ravijo, 2020-02-27:
Hi @Ceepert 

> Did anything happen on this subject yet? I wanted to use a normal webcam with this wrapper

Please check [here](https://github.com/ravijo/ros_openpose/issues/7#issuecomment-591995554).

ravijo, 2020-03-02:
Hi @Gryogor

> Hello. Are you planning on implementing the support of ordinary RGB cameras for only 2D body image?

Yes. It is done. Please pull the recent changes from the repository. Now it supports any color camera such as webcam etc. See the [README.md](https://github.com/ravijo/ros_openpose/blob/master/README.md) for more details.

closed 2020-03-02 by ravijo