ravijo/ros_openpose #3: Can't compile

silasalves, 2019-10-29:
I am having the same problem as in Issue #1, and can't compile the project. I am using a fresh Ubuntu 16.04 install with ROS-Kinetic and OpenPose 1.5.1. I never installed any previous version of OpenPose, but I am getting the same problem as in #1:

```
error: no match for ‘operator=’ (operand types are ‘op::Matrix’ and ‘const cv::Mat’)
         datumPtr->cvInputData = colorImage;
```

I noticed that I had two versions of OpenCV (2.4, supplied by Ubuntu, and 3.3, supplied by ROS), so I tried using only both (first try), then I removed 3.3 (which didn't work because the projects are dependent on `ros-kinetic-cv-bridge`). Finally, I removed 2.4 completely and compiled with 3.3 only, but I still got the same errors.

Please, do you have any suggestion?

I am also enclosing the full `catkin_make` output for completion.

Thanks!

*catkin_make output:*
```
silas:~/catkin_ws$ catkin_make
Base path: /home/silas/catkin_ws
Source space: /home/silas/catkin_ws/src
Build space: /home/silas/catkin_ws/build
Devel space: /home/silas/catkin_ws/devel
Install space: /home/silas/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/silas/catkin_ws/build"
####
####
#### Running command: "make -j6 -l6" in "/home/silas/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target _ros_openpose_generate_messages_check_deps_Person
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target _ros_openpose_generate_messages_check_deps_BodyPart
[  0%] Built target _ros_openpose_generate_messages_check_deps_Frame
[  0%] Built target _ros_openpose_generate_messages_check_deps_Pixel
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  2%] Built target std_msgs_generate_messages_lisp
[  2%] Building CXX object ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o
[  2%] Built target _catkin_empty_exported_target
[  2%] Built target sensor_msgs_generate_messages_lisp
[  2%] Built target diagnostic_msgs_generate_messages_py
[  9%] Built target testCameraReader
[  9%] Built target roscpp_generate_messages_nodejs
[  9%] Built target roscpp_generate_messages_lisp
[  9%] Built target rosgraph_msgs_generate_messages_lisp
[  9%] Built target rosgraph_msgs_generate_messages_py
[  9%] Built target sensor_msgs_generate_messages_nodejs
[  9%] Built target rosgraph_msgs_generate_messages_nodejs
[  9%] Built target rosgraph_msgs_generate_messages_eus
[  9%] Built target sensor_msgs_generate_messages_eus
[  9%] Built target roscpp_generate_messages_py
[  9%] Built target rosgraph_msgs_generate_messages_cpp
[  9%] Built target sensor_msgs_generate_messages_py
[  9%] Built target roscpp_generate_messages_cpp
[  9%] Built target roscpp_generate_messages_eus
[  9%] Built target sensor_msgs_generate_messages_cpp
[  9%] Built target tf_generate_messages_cpp
[  9%] Built target actionlib_generate_messages_eus
[  9%] Built target dynamic_reconfigure_generate_messages_py
[  9%] Built target _realsense2_camera_generate_messages_check_deps_IMUInfo
[  9%] Built target bond_generate_messages_lisp
[  9%] Built target _realsense2_camera_generate_messages_check_deps_Extrinsics
[  9%] Built target nodelet_generate_messages_nodejs
[  9%] Built target nav_msgs_generate_messages_lisp
[... 174 lines cut ...]
/home/silas/catkin_ws/src/ros_openpose/src/rosOpenpose.cpp:334:69: error: no matching function for call to ‘op::WrapperStructOutput::WrapperStructOutput(<brace-enclosed initializer list>)’
                                                       FLAGS_udp_port};
                                                                     ^
In file included from /usr/local/include/openpose/wrapper/wrapper.hpp:12:0,
                 from /usr/local/include/openpose/wrapper/headers.hpp:6,
                 from /usr/local/include/openpose/headers.hpp:47,
                 from /home/silas/catkin_ws/src/ros_openpose/src/rosOpenpose.cpp:21:
/usr/local/include/openpose/wrapper/wrapperStructOutput.hpp:150:9: note: candidate: op::WrapperStructOutput::WrapperStructOutput(double, const op::String&, op::DataFormat, const op::String&, const op::String&, int, int, const op::String&, const op::String&, const op::String&, double, bool, const op::String&, const op::String&, const op::String&, const op::String&, const op::String&, const op::String&, const op::String&)
         WrapperStructOutput(
         ^
/usr/local/include/openpose/wrapper/wrapperStructOutput.hpp:150:9: note:   no known conversion for argument 2 from ‘fLS::clstring {aka std::__cxx11::basic_string<char>}’ to ‘const op::String&’
/usr/local/include/openpose/wrapper/wrapperStructOutput.hpp:13:19: note: candidate: op::WrapperStructOutput::WrapperStructOutput(const op::WrapperStructOutput&)
     struct OP_API WrapperStructOutput
                   ^
/usr/local/include/openpose/wrapper/wrapperStructOutput.hpp:13:19: note:   candidate expects 1 argument, 19 provided
/usr/local/include/openpose/wrapper/wrapperStructOutput.hpp:13:19: note: candidate: op::WrapperStructOutput::WrapperStructOutput(op::WrapperStructOutput&&)
/usr/local/include/openpose/wrapper/wrapperStructOutput.hpp:13:19: note:   candidate expects 1 argument, 19 provided
ros_openpose/CMakeFiles/rosOpenpose.dir/build.make:62: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o' failed
make[2]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o] Error 1
CMakeFiles/Makefile2:1593: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed
```

silasalves, 2019-10-29:
After trying different approaches, a friend suggested looking if there were any special branch or tags for OpenPose, and we found out that there is a tag `v1.5.1`.

So we went to OpenPose repo and run the following command to get the correct files for version 1.5.1.

```
$ git checkout tags/v1.5.1 -b v1.5.1
```

After building/installing OpenPose again, I tried compiling ros_openpose and then everything worked.

Maybe adding this information on the installation notes will be helpful for new users.

Thanks!

ravijo, 2019-10-30:
Hi @silasalves 

Thank you very much for the workaround. Yes, you are right. To be more precise, the error `error: no match for ‘operator=’ (operand types are ‘op::Matrix’ and ‘const cv::Mat’)` is coming from OpenPose while you are trying to assign OpenCV matrix to OpenPose matrix. If [you see the source code clearly](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/559a57ec436f28ebdc29b2d27d5d075604fd1239/src/openpose/core/matrix.cpp), you can find that the OpenPose matrix is a wrapper build on the top of OpenCV matrix just to avoid OpenCV as dependency in the headers.

This repository should, therefore, work with >= v1.5.1 . I am adding the information on the installation note to check out the correct/latest version.

Thanks again. I appreciate your efforts.