ravijo/ros_openpose #4: fatal error: ros_openpose/Frame.h: No such file or directory

xh20, 2019-11-28:
Hi ravijo,

when I run catkin_make clean, it appears an error: 
**fatal error: ros_openpose/Frame.h: No such file or directory**
and I only find cameraReader.hpp in the folder ~/catkin_ws/src/ros_openpose/include/ros_openpose. Is it still in programming?

best 
Hao

ravijo, 2019-11-28:
@xh20 , thank you very much.

> when I run catkin_make clean, it appears an error:
> **fatal error: ros_openpose/Frame.h: No such file or directory**

Sorry to hear that you faced the issue. Please note that `catkin_make clean` is going to clean the `build` and other folders. 
Before you clone, `ros_openpose`, make sure `catkin_make` compiles all existing packages on your workspace.  After cloning this package, you should run `catkin_make` a few times. 

> I only find cameraReader.hpp in the folder 

`Frame.h` is defined as a ROS msg file. You can find its definition [here](https://github.com/ravijo/ros_openpose/blob/master/msg/Frame.msg). Furthermore, it is included in [CMakeLists.txt](https://github.com/ravijo/ros_openpose/blob/master/CMakeLists.txt#L38), so that `catkin_make` can find it out.

> Is it still in programming?

The code pushed to GitHub should work fine. 

Let me know if the problem persists.

Cheers!

xh20, 2019-11-28:
@ravijo Thank you for replying. Yes, you are right. I don't need to clean it. It is compiled now.

ravijo, 2019-11-28:
@xh20 Thank you for the quick response. Glad that it compiled. Feel free to create new issues if needed. I will close this issue now.

closed 2019-11-28 by ravijo