ravijo/ros_openpose #5: Problem with 3D visualization inside RVIZ.

OsArkoses, 2020-01-16:
I have a problem  with 3D visualization inside RVIZ

I cam see the skeleton in default window 
But, in rviz only the image

In console show next error:
```
 _File "/home/user/catkin_ws/src/ros_openpose/scripts/visualizer.py", line 169, in frame_callback
    leftHandPart =  person.leftHandParts[point_id]
IndexError: list index out of range_

```

Output in rostopic echo /frame:
  

> 
> ....
>  point: 
>           x: 0.0278618205339
>           y: -0.336625277996
>           z: 0.774000048637
>       - 
>         score: 0.0
>         pixel: 
>           x: 0.0
>           y: 0.0
>         point: 
>           x: -0.193545013666
>           y: 0.103801973164
>           z: 0.628000020981
>       - 
>         score: 0.0
>         pixel: 
>           x: 0.0
>           y: 0.0
>         point: 
>           x: -0.193545013666
>           y: 0.103801973164
>           z: 0.628000020981
>       - 
>         score: 0.0
>         pixel: 
>           x: 0.0
>           y: 0.0
>         point: 
>           x: -0.477131932974
>           y: 0.190523177385
>           z: 1.23500001431
>       - 
>         score: 0.0
>         pixel: 
>           x: 0.0
>           y: 0.0
>         point: 
>           x: -0.477131932974
>           y: 0.190523177385
>           z: 1.23500001431
>       - 
>         score: 0.0
>         pixel: 
>           x: 0.0
>           y: 0.0
>         point: 
>           x: -0.1889847368
>           y: 0.0630785375834
>           z: 0.572000026703
>       - 
>         score: 0.0
>         pixel: 
>           x: 0.0
>           y: 0.0
>         point: 
>           x: -0.1889847368
>           y: 0.0630785375834
>           z: 0.572000026703
>     leftHandParts: []
>     rightHandParts: []

OsArkoses, 2020-01-16:
After installation of ros-visualization-msgs

`sudo apt-get install -y ros-visualization-msgs`

In rviz Human Skeleton, Status: Error:

> Uninitialized quaternion, assuming identity. 
> scale.y and scale.z of LINE_LIST or LINE_STRIP are ignored. 
> Points should not be empty for specified marker type. 
> At least two points are required for a LINE_STRIP marker.

In rviz the whole person is shown in green

output rostopic echo /frame
> point: 
>           x: 0.76700001955
>           y: 0.0799146220088
>           z: -0.0290726330131
>     leftHandParts: 
>       - 
>         score: 0.299114465714
>         pixel: 
>           x: 471.147674561
>           y: 198.045516968
>         point: 
>           x: 0.410511046648
>           y: -0.0188666600734
>           z: 0.705000042915
>       - 
>         score: 0.460736393929
>         pixel: 
>           x: 479.077056885
>           y: 205.023361206
>         point: 
>           x: 0.408933371305
>           y: -0.00515820737928
>           z: 0.677000045776
>       - 
>         score: 0.590507388115
>         pixel: 
>           x: 486.054901123
>           y: 210.415328979
>         point: 
>           x: 0.418776631355
>           y: 0.0048198052682
>           z: 0.672000050545
>       - 
>         score: 0.729315817356
>         pixel: 
>           x: 498.424713135
>           y: 211.049682617
>         point: 
>           x: 0.418776631355
>           y: 0.0048198052682
>           z: 0.672000050545
>       - 
>         score: 0.661788702011
>         pixel: 
>           x: 511.428894043
>           y: 212.318389893
>         point: 
>           x: 4.42572249355e-28
>           y: 3.08355727075e-41
>           z: 0.112212896347
>       - 
>         score: 0.546052455902
>         pixel: 
>           x: 498.424713135
>           y: 188.847442627
>         point: 
[... 150 lines cut ...]
>         pixel: 
>           x: 319.062286377
>           y: 195.99760437
>         point: 
>           x: 0.0592623464763
>           y: -0.0116241201758
>           z: 0.35900002718
>       - 
>         score: 0.017850574106
>         pixel: 
>           x: 315.29196167
>           y: 190.551574707
>         point: 
>           x: 0.11264577508
>           y: -0.0344482399523
>           z: 0.728000044823
>       - 
>         score: 0.017687804997
>         pixel: 
>           x: 323.251525879
>           y: 196.835449219
>         point: 
>           x: 0.11264577508
>           y: -0.0344482399523
>           z: 0.728000044823

ravijo, 2020-01-17:
@OsArkoses: The output given by you in the above two messages is not matching with each other. In the [first message](https://github.com/ravijo/ros_openpose/issues/5#issue-550653763), I can see that parameters `leftHandParts` and `rightHandParts` are empty list. In the [second message](https://github.com/ravijo/ros_openpose/issues/5#issuecomment-575085035), I can see that the parameter `leftHandParts` contains 21 points, which is expected (correct).

1. First of all, I suspect that your ROS installation is broken (or you have not installed a complete ROS package. I mostly install the full version, i.e., `ros-${DISTRO}-desktop-full`. Therefore can you please make sure you have all the packages installed?

    (To confirm the above point, you can simply try using [my another ROS package](https://github.com/ravijo/rviz_skeleton_visualization) which uses the almost same method for visualization. You don't need to install dependency [kinect_anywhere](https://github.com/ravijo/kinect_anywhere) this time. Just clone/download the repository and make python scripts executables. Finally run `roslaunch rviz_skeleton_visualization visualization.launch`. If you see a static human visualization inside RVIZ then you don't need to install extra ROS packages.)

2. Secondly, in the [first message](https://github.com/ravijo/ros_openpose/issues/5#issue-550653763), you reported that the error is at line 169, which is a bit surprising. In the latest version, the line `leftHandPart = person.leftHandParts[point_id]` is at number 165. Can you please pull the latest source code from [GitHub](https://github.com/ravijo/ros_openpose) and try again?

3. The error about `id_text_offset` reported by you via email was a bug which is fixed in the latest code. Please use the latest code from [GitHub](https://github.com/ravijo/ros_openpose) and try again.

4. From the screenshot you shared with me, it seems that the parameter `skeleton_line_width` needs to be set to a much lower value. Please open [run_kinect.launch](https://github.com/ravijo/ros_openpose/blob/master/launch/run_kinect.launch#L48) file and change the value `0.01` to a much lower value such as `0.005` and try again.

5. Finally, please unset the variable `skeleton_hands` if you don't want to see fingers inside the RVIZ by executing the following command-
```
roslaunch ros_openpose run.launch skeleton_hands:=false
```

Let me know if the problem persists.

OsArkoses, 2020-01-20:
Hello Ravi

1. Yes, I have the desktop-full installation
2. I  modificated the file, I wanted to print the content 
3. Thanks, I did the changes.
4. **This was my problem,** I made the changes and now it's going well. 
5. Thanks.

I have a question, Is it normal for the skeleton to be distorted at the time of viewing in Rviz? incorrect lines appear. In the openPose by default if the points are displayed correctly.

ravijo, 2020-01-20:
Hi @OsArkoses 

Glad that you made it working!

> Is it normal for the skeleton to be distorted at the time of viewing in Rviz? incorrect lines appear. In the openPose by default if the points are displayed correctly.

1. 3D point calculation highly depends on depth values received by the camera. I believe that the depth information is fluctuating which is causing the distortion. You can see that the point cloud inside rviz is fluctuating too. If required, you can preprocess 3D points by applying filtering process in order to reduce the fluctuation.

2. You can reduce the `lifetime` of the skeleton lines. The marker property 'lifetime' defines how long the object should last before being automatically deleted.  0 means forever. Please set it to a lower value (more than 0 but less than 1) [here](https://github.com/ravijo/ros_openpose/blob/master/scripts/visualizer.py#L110). 

3. If you see that the skeleton is not overlapping with your body inside the point cloud, then you must calibrate (intrinsic calibration) your camera. Please note that I haven't considered camera distortion while computing 3D point from pixel. 

4. Furthermore, the fluctuation may not be easily visible in image (2D)  because of the small pixel values but in 3D, it gets converted into real values shown in meter. 

Let me know if this helps.

ravijo, 2020-01-27:
Hi @OsArkoses,

Can you please confirm if the problem is fixed? I will close this thread after your acknowledgment.

OsArkoses, 2020-01-27:
Hi Ravi

The problem is fixed

Thank you very much for your valuable help

closed 2020-01-27 by OsArkoses