ravijo/ros_openpose #6: Low VRAM Settings

rocketmax, 2020-01-29:
Hello,

I know that I'm pushing my luck trying to run this on a laptop with a GTX 960M (2 GB), but I'm fairly limited in my hardware right now. The Openpose demo will run if I lower the net_resolution parameter down below -1x224. I've tried to run this with that parameter added to the openpose_args line in the launch file:

`<arg name="openpose_args" value=" --model_folder /home/ubuntafoo/openpose/models/ --net_resolution -1x160 --model_pose BODY_25" />`

I've even tried to lower it down to -1x16 and that didn't help, so I'm wondering if there is something within the system that locks the network resolution or if there is any way to lower the graphics memory needed that you know of.

Thanks for any help you can provide!

Here's my full error log
```
roslaunch ros_openpose run_realsense.launch
... logging to /home/ubuntafoo/.ros/log/d8509af4-42e6-11ea-9c31-441ca8e2c5e3/roslaunch-ubuntafoo-XPS-15-9550-2683.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntafoo-XPS-15-9550:35019/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
[... 135 lines cut ...]
    @     0x7f92f420d6d3  op::WPoseExtractor<>::work()
    @     0x7f92f4247b59  op::Worker<>::checkAndWork()
    @     0x7f92f4247cfb  op::SubThread<>::workTWorkers()
    @     0x7f92f425079b  op::SubThreadQueueInOut<>::work()
    @     0x7f92f424abbb  op::Thread<>::threadFunction()
    @     0x7f92f24739e0  (unknown)
    @     0x7f92f27466db  start_thread
    @     0x7f92f1c9f88f  clone
================================================================================REQUIRED process [rosOpenpose-2] has died!
process has died [pid 2719, exit code -6, cmd /home/ubuntafoo/catkin_ws/devel/lib/ros_openpose/rosOpenpose --hand __name:=rosOpenpose __log:=/home/ubuntafoo/.ros/log/d8509af4-42e6-11ea-9c31-441ca8e2c5e3/rosOpenpose-2.log].
log file: /home/ubuntafoo/.ros/log/d8509af4-42e6-11ea-9c31-441ca8e2c5e3/rosOpenpose-2*.log
Initiating shutdown!
================================================================================
[rviz-6] killing on exit
[camera/realsense2_camera-5] killing on exit
[camera/realsense2_camera_manager-4] killing on exit
[visualizer-3] killing on exit
[rosOpenpose-2] killing on exit
terminate called without an active exception
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

ravijo, 2020-01-30:
Hi @rocketmax, 

I am sorry to hear about the issue. As you said already, you are using a GTX 960M (2 GB) graphics card. It is evident from the following log message that you are running out of memory-

```
Check failed: error == cudaSuccess (2 vs. 0)  out of memory
``` 

This package, i.e. [ros_openpose](https://github.com/ravijo/ros_openpose) doesn't use GPU directly. However, it depends on [OpenPose](https://github.com/CMU-Perceptual-Computing-Lab/openpose) which uses GPU heavily.

Since the error is coming from OpenPose software, I suggest you run the openpose software on example data as shown below-

```
./build/examples/openpose/openpose.bin --video examples/media/video.avi
```

Now, please try to reduce the `--net_resolution` and give it a try with `--model_pose COCO.` Also, please make sure that `cuDNN` is installed correctly. You need to compile OpenPose with it.

I also suggest you read and try this [documentation](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/faq.md#out-of-memory-error).

Let me know if this helps!

rocketmax, 2020-01-30:
Hey @ravijo that helped a lot! The COCO model does use less memory and works well for me. Now I'm having an issue similar to Issue #4 , but I am going to keep working on it.

Thanks again!

closed 2020-01-30 by rocketmax

ravijo, 2020-01-30:
Hi @rocketmax,

Great! Glad that it worked. Feel free to post your error if you can't fix it.