ravijo/ros_openpose #7: Using a webcam

Ceepert, 2020-02-27:
Is it possible to use this wrapper with a normal webcam? When running the launch file it states, that the resource realsense wasnt found (ResourceNotFound: realsense2_camera), but there is no option in that file to put in webcam, just kinect and realsense

The plan was to use an already existing image_raw topic

ravijo, 2020-02-27:
Hi @Ceepert 

Yes, you can use a webcam with it. However, you need to hack the package. So far, I am not getting time to do it by myself. But it is doable by following the steps below.

As you said already, we can use `image_raw` topic. In this package `image_raw` is called `color_topic`. Please follow the steps mentioned below-

1. Inside file src/cameraReader.cpp, change [line number 89](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/cameraReader.cpp#L89) to the following-
```
auto depthPtr = cv_bridge::toCvCopy(depthMsg, sensor_msgs::image_encodings::BGR8);
```
2. Inside the launch file [run_realsense.launch](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/launch/run_realsense.launch)
   1. Look for `depth_topic` and assign it to `color_topic`. In other words, `depth_topic` must be the same as `color_topic` in this special case.
    1. Look for `pointcloud` and assign `false` to it.
    1. Look for `skeleton` and assign `false` to it.
    1. Remove 3D visualization commands by deleting all the lines starting from [76 to 93](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/launch/run_realsense.launch#L76-L93).
3. Comment the line which starts from `mSPtrCameraReader->compute3DPoint`. Therefore, please comment line number [119](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/rosOpenpose.cpp#L119), [149](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/rosOpenpose.cpp#L149) and [152](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/rosOpenpose.cpp#L152) inside [rosOpenpose.cpp](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/rosOpenpose.cpp).

4. Open a terminal and source ros workspace. Please compile the project by executing `catkin_make` as usual.
5. Please connect your webcam and start ros package which is basically going to capture images from your webcam and it is going to publish those images as ROS topic i.e. `image_raw`.
6. Open anotger terminal and source ros workspace. Run the launch file.
```
roslaunch ros_openpose run_realsense.launch
```
7. Open another terminal and source ros workspace. Now run `rostopic echo /frame` to see the output values.

I will add this feature once I get time. Let me know if it works.

Ceepert, 2020-03-01:
> 
> 
> Hi @Ceepert
> 
> Yes, you can use a webcam with it. However, you need to hack the package. So far, I am not getting time to do it by myself. But it is doable by following the steps below.
> 
> As you said already, we can use `image_raw` topic. In this package `image_raw` is called `color_topic`. Please follow the steps mentioned below-
> 
>     1. Inside file src/cameraReader.cpp, change [line number 89](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/cameraReader.cpp#L89) to the following-
> 
> 
> ```
> auto depthPtr = cv_bridge::toCvCopy(depthMsg, sensor_msgs::image_encodings::BGR8);
> ```
> 
>     1. Inside the launch file [run_realsense.launch](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/launch/run_realsense.launch)
>        
>        1. Look for `depth_topic` and assign it to `color_topic`. In other words, `depth_topic` must be the same as `color_topic` in this special case.
>        2. Look for `pointcloud` and assign `false` to it.
>        3. Look for `skeleton` and assign `false` to it.
>        4. Remove 3D visualization commands by deleting all the lines starting from [76 to 93](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/launch/run_realsense.launch#L76-L93).
> 
>     2. Comment the line which starts from `mSPtrCameraReader->compute3DPoint`. Therefore, please comment line number [119](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/rosOpenpose.cpp#L119), [149](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/rosOpenpose.cpp#L149) and [152](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/rosOpenpose.cpp#L152) inside [rosOpenpose.cpp](https://github.com/ravijo/ros_openpose/blob/3efa6d47088d55498ea74dd0c54737c095838607/src/rosOpenpose.cpp).
> 
>     3. Open a terminal and source ros workspace. Please compile the project by executing `catkin_make` as usual.
> 
>     4. Please connect your webcam and start ros package which is basically going to capture images from your webcam and it is going to publish those images as ROS topic i.e. `image_raw`.
> 
>     5. Open anotger terminal and source ros workspace. Run the launch file.
> 
> 
> ```
> roslaunch ros_openpose run_realsense.launch
> ```
> 
>     1. Open another terminal and source ros workspace. Now run `rostopic echo /frame` to see the output values.
> 
> 
> I will add this feature once I get time. Let me know if it works.

Okay thanks, im gonna try that. Just some questions as I am pretty new to ROS. At point 5. you mentioned to start "ros packaged" do you mean a specific one? Or just the one I use when getting the image topic? Cause I wrote one myselfe as I havent found a good one to get the image topic.

And when you mean to source the workspace you mean to source it to my catkin workspace right?

ravijo, 2020-03-01:
Hi @Ceepert 

Glad to hear from you.

> Okay thanks, im gonna try that. Just some questions as I am pretty new to ROS. At point 5. you mentioned to start "ros packaged" do you mean a specific one? Or just the one I use when getting the image topic? Cause I wrote one myselfe as I havent found a good one to get the image topic.

All I was saying is to start your ROS node, which provides images in a ROS topic.

> And when you mean to source the workspace you mean to source it to my catkin workspace right?

Yes. You are right.

Nevertheless, give me a week. I can push these changes smoothly. I planned to improve the efficiency of the code, but for now, I think the important thing is to push the webcam code so that you can use it.

Cheers!

Ceepert, 2020-03-01:
> Hi @Ceepert
>
> Glad to hear from you.
>
>
> All I was saying is to start your ROS node, which provides images in a ROS topic.
>
>
> Yes. You are right.
>
> Nevertheless, give me a week. I can push these changes [...]

Thanks for the quick reply. Im gonna try your steps anyway, just to try it, cause I need this for a school project, but if I know myselfe Im gonna screw up at one step or another anyway

Ceepert, 2020-03-02:
> Hi @Ceepert
>
> Glad to hear from you.
>
>
> All I was saying is to start your ROS node, which provides images in a ROS topic.
>
>
> Yes. You are right.
>
> Nevertheless, give me a week. I can push these changes [...]

oh one more thing. I just noticed that in the details of the wrapper it says Indigo for the ROS version. I was able to compile it on Kinetic. Should it work with kinetic or will there be some issue I will run into?

ravijo, 2020-03-02:
Hi @Ceepert 

> it says Indigo for the ROS version.
It says that the package has been tested on ROS Indigo.

> Should it work with kinetic or will there be some issue I will run into?
Yes. The code doesn't depend on ROS distro.

BTW, pull the recent changes from the repository. Now it supports any color camera such as webcam etc. See the [README.md](https://github.com/ravijo/ros_openpose/blob/master/README.md) for more details.

closed 2020-03-02 by ravijo

Ceepert, 2020-03-02:
gf

> 
> 
> Hi @Ceepert
> 
> > it says Indigo for the ROS version.
> 
> It says that the package has been tested on ROS Indigo.
> 
> > Should it work with kinetic or will there be some issue I will run into?
> 
> Yes. The code doesn't depend on ROS distro version.
> 
> BTW, pull the recent changes from the repository. Now it supports any color camera such as webcam etc. See the [README.md](https://github.com/ravijo/ros_openpose/blob/master/README.md) for more details.

Thank you very much, cant test it right now, but I´ll give it a try as soon as I can