ravijo/ros_openpose #8: Cannot launch node of type error AND file not found

Ceepert, 2020-03-05:
Hello again,
I just tried using your new normal camera feature and I am having some issues with it.
I set the color_topic to my cv_camera/image_raw topic and tried to launch it.
One error states, that ros_openpose/echo.py couldnt be launched and it for some reason cant find the pose_deploay.prototext

I did install OpenPose with sudo make install and the needed file exists in the openpose folder, but it searches in a completely different non exisiting directory.

And then there is the warning in the end, where it states, that the camera frame is empty, even though I can see the image it sends.

######################################
PARAMETERS
 * /echo/pub_topic: /frame
 * /rosOpenpose/cam_info_topic: cam_info_topic
 * /rosOpenpose/color_topic: /cv_camera/image_raw
 * /rosOpenpose/depth_topic: depth_topic
 * /rosOpenpose/frame_id: no_depth
 * /rosOpenpose/no_depth: True
 * /rosOpenpose/openpose_model_dir: /home/ravi/tools/...
 * /rosOpenpose/pub_topic: /frame
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    echo (ros_openpose/echo.py)
    rosOpenpose (ros_openpose/rosOpenpose)

ROS_MASTER_URI=http://localhost:11311

WARNING: Package name "psTutorial" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[rosOpenpose-1]: started with pid [7778]
ERROR: cannot launch node of type [ros_openpose/echo.py]: can't locate node [echo.py] in package [ros_openpose]
[ INFO] [1583401261.208366552]: Starting ros_openpose...
Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.

Error:
Prototxt file not found: /home/ravi/tools/openpose/models/pose/body_25/pose_deploy.prototxt.
Possible causes:
	1. Not downloading the OpenPose trained models.
	2. Not running OpenPose from the same directory where the `model` folder is located.
	3. Using paths with spaces.

Coming from:
- /home/student/Desktop/OpenPose/openpose/src/openpose/net/netCaffe.cpp:ImplNetCaffe():58
- /home/student/Desktop/OpenPose/openpose/src/openpose/net/netCaffe.cpp:ImplNetCaffe():94
- /home/student/Desktop/OpenPose/openpose/src/openpose/pose/poseExtractorCaffe.cpp:addCaffeNetOnThread():108
- /home/student/Desktop/OpenPose/openpose/src/openpose/pose/poseExtractorCaffe.cpp:netInitializationOnThread():198
- /home/student/Desktop/OpenPose/openpose/src/openpose/pose/poseExtractorNet.cpp:initializationOnThread():98
- /home/student/Desktop/OpenPose/openpose/src/openpose/pose/poseExtractor.cpp:initializationOnThread():34
- /home/student/Desktop/OpenPose/openpose/include/openpose/pose/wPoseExtractor.hpp:initializationOnThread():57
- /home/student/Desktop/OpenPose/openpose/include/openpose/thread/worker.hpp:initializationOnThreadNoException():77
[ WARN] [1583401261.231449402]: Empty color image frame detected. Ignoring...

Ceepert, 2020-03-05:
Considering it mentiones "ravi" in the directory, could it be that you by accident hard coded the path somewhere? Considering your name is ravijo

ravijo, 2020-03-05:
Hi @Ceepert 

I updated the instructions in the ReadMe file just now. Please do the following-

* You need to make Python scripts executable as stated [here](https://github.com/ravijo/ros_openpose#installation).
* Please edit `run.launch` to update `openpose_model_dir` as stated [here](https://github.com/ravijo/ros_openpose#configuration).

Let me know if the problem persists.

Ceepert, 2020-03-05:
> Hi @Ceepert
>
> I updated the instructions in the ReadMe file just now. Please do the following-
>
>     * You need to make Python scripts executable as stated [...]

Yeah the old errors disappeared, but now the node crashes after a second with this:

SUMMARY
========

PARAMETERS
 * /echo/pub_topic: /frame
 * /rosOpenpose/cam_info_topic: cam_info_topic
 * /rosOpenpose/color_topic: /cv_camera/image_raw
 * /rosOpenpose/depth_topic: depth_topic
 * /rosOpenpose/frame_id: no_depth
 * /rosOpenpose/no_depth: True
 * /rosOpenpose/openpose_model_dir: /home/student/Des...
 * /rosOpenpose/pub_topic: /frame
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    echo (ros_openpose/echo.py)
    rosOpenpose (ros_openpose/rosOpenpose)

ROS_MASTER_URI=http://localhost:11311

WARNING: Package name "psTutorial" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[rosOpenpose-1]: started with pid [15685]
process[echo-2]: started with pid [15686]
[ INFO] [1583408577.440885207]: Starting ros_openpose...
Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.
[ WARN] [1583408577.467630423]: Empty color image frame detected. Ignoring...
================================================================================REQUIRED process [rosOpenpose-1] has died!
process has died [pid 15685, exit code -11, cmd /home/student/catkin_workspace/devel/lib/ros_openpose/rosOpenpose __name:=rosOpenpose __log:=/home/student/.ros/log/5ad528ae-5ed6-11ea-a7cf-48a4727f27d1/rosOpenpose-1.log].
log file: /home/student/.ros/log/5ad528ae-5ed6-11ea-a7cf-48a4727f27d1/rosOpenpose-1*.log
Initiating shutdown!
================================================================================
[echo-2] killing on exit
[rosOpenpose-1] killing on exit

ravijo, 2020-03-05:
1. The package is unable to get images from your camera. As per your question, the `color_topic` is `/cv_camera/image_raw`. Therefore, please check if the topic is correct by executing the following command-
```
rosrun image_view image_view image:=/cv_camera/image_raw
```
A window should pop up and you should see realtime images.

2. You have a package called `psTutorial` which is showing the warning.

3. Nevertheless, ros_openpose should not stop even if there is no image.

Ceepert, 2020-03-05:
>     1. The package is unable to get images from your camera. As per your question, the `color_topic` is `/cv_camera/image_raw`. Therefore, please check if the topic is correct by executing the following command-
> 
> 
> ```
> rosrun image_view image_view image:=/cv_camera/image_raw
> ```
> 
> A window should pop up and you should see realtime images.
> 
>     1. You have a package called `psTutorial` which is showing the warning.
> 
>     2. Nevertheless, ros_openpose should not stop even if there is no image.

I tried that and I do see the realtime image of my webcam.
And the psTutorial warning as far as i can see is only a namingconvetion warning as it has upper case letters in it

ravijo, 2020-03-05:
Can you please post the complete output as shown in terminal?

Ceepert, 2020-03-05:
> Can you please post the complete output as shown in terminal?

Sure:(man, why is the warning in bold...)

student@student-GL73-8SE:~$ roslaunch ros_openpose run.launch camera:=nodepthWARNING: Package name "psTutorial" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/student/.ros/log/16a64760-5eec-11ea-bf7c-48a4727f27d1/roslaunch-student-GL73-8SE-2540.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "psTutorial" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://student-GL73-8SE:39071/

SUMMARY
========

PARAMETERS
 * /echo/pub_topic: /frame
 * /rosOpenpose/cam_info_topic: cam_info_topic
 * /rosOpenpose/color_topic: /cv_camera/image_raw
 * /rosOpenpose/depth_topic: depth_topic
 * /rosOpenpose/frame_id: no_depth
 * /rosOpenpose/no_depth: True
 * /rosOpenpose/openpose_model_dir: /home/student/Des...
 * /rosOpenpose/pub_topic: /frame
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    echo (ros_openpose/echo.py)
    rosOpenpose (ros_openpose/rosOpenpose)

ROS_MASTER_URI=http://localhost:11311

WARNING: Package name "psTutorial" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[rosOpenpose-1]: started with pid [2557]
process[echo-2]: started with pid [2558]
[ INFO] [1583417885.332686373]: Starting ros_openpose...
Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.
[ WARN] [1583417885.375693820]: Empty color image frame detected. Ignoring...
================================================================================REQUIRED process [rosOpenpose-1] has died!
process has died [pid 2557, exit code -11, cmd /home/student/catkin_workspace/devel/lib/ros_openpose/rosOpenpose --face __name:=rosOpenpose __log:=/home/student/.ros/log/16a64760-5eec-11ea-bf7c-48a4727f27d1/rosOpenpose-1.log].
log file: /home/student/.ros/log/16a64760-5eec-11ea-bf7c-48a4727f27d1/rosOpenpose-1*.log
Initiating shutdown!
================================================================================
[echo-2] killing on exit
[rosOpenpose-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

ravijo, 2020-03-05:
> Sure:(man, why is the warning in bold...)

The output should be written inside three backquotes.

Anyway, the output is strange. The package is not able to get the RGB image frame. Are you sure that the webcam images are RGB color (not monochrome)? 

Furthermore, it is better to run the code inside a debugger for better understanding. Please do the following-

1. Please add `launch-prefix="gdb -ex run --args"` attribute to the <node> tag inside `core.launch` file. Please see below the example-
```
<node name="rosOpenpose" pkg="ros_openpose" type="rosOpenpose" output="screen" required="true" args="$(arg openpose_args)" >
```
Please change it to the following-
```
<node launch-prefix="gdb -ex run --args" name="rosOpenpose" pkg="ros_openpose" type="rosOpenpose" output="screen" required="true" args="$(arg openpose_args)" >
```
2. Generate symbol table for better debugging by recompiling the package using following command-
```
catkin_make  -DCMAKE_BUILD_TYPE=RelWithDebInfo --pkg ros_openpose
```
If this doesn't work, please use the following command-
```
catkin_make -DCMAKE_BUILD_TYPE=RelWithDebInfo --only-pkg-with-deps ros_openpose
```
3. Run the main launch file as usual by using the following command-
```
roslaunch ros_openpose run.launch camera:=nodepth
```

When you see that the program stopped, please type `bt full` inside the GDB terminal. Please share the complete information via email. My email is mentioned [here](https://github.com/ravijo/ros_openpose/blob/master/package.xml).

ravijo, 2020-03-06:
Hi @Ceepert 

I suspect the OpenPose installation in your machine. Can you please make sure that OpenPose is working on your machine. 

Also, are you sure that the webcam images are in RGB colors (not monochrome)? If this is the case, then I can put a conditional statement to check the number of channels in the input image instead of assuming it RGB (to be precise `bgr8` in OpenCV terms). Please confirm.

Well, I have tested the package with a webcam and it works like a charm. I only needed to change `color_topic` inside `config_nodepth.launch` file. _(I used [usb_cam](http://wiki.ros.org/usb_cam) package to get the images from the webcam.)_

I also checked the package with incorrect `color_topic`  and found that the OpenPose window pops up but reports the following message in the terminal `Empty color image frame detected. Ignoring...`. Please note that the program doesn't die and waits for the images, as expected.

closed 2020-03-06 by ravijo

Ceepert, 2020-03-07:
> Hi @Ceepert
>
> I suspect the OpenPose installation in your machine. Can you please make sure that OpenPose is working on your machine.
>
> Also, are you sure that the webcam images are in RGB colors (not [...]

Hi,
I dont have access to the computer I am working on during the weekend, so to test the things you wrote i´ll need to wait till next week.
But what I can say is, that I tested my camera to show the captured image and it is in normal colors.

I havent tested openpose in the last couple of days, but the last time I checked it worked, but I can try it again next week as it wouldnt have been the first time it just broke.

It too used to wait for the image frame and not crash in the beginning of this issue, but after the first fix it started to crash.

Does the usb_cam package work with integrated webcams from a laptop? If so I will try using that instead of my own node to capture the image and try that instead.

Ceepert, 2020-03-09:
> Hi @Ceepert
>
> I suspect the OpenPose installation in your machine. Can you please make sure that OpenPose is working on your machine.
>
> Also, are you sure that the webcam images are in RGB colors (not [...]

Ive sent you an email with the requested output.

Ceepert, 2020-03-09:
I am also getting this warning in my camera node. Could this cause the issues? 
'''rosrun cv_camera cv_camera_node 
[ INFO] [1583768876.916197239]: using default calibration URL
[ INFO] [1583768876.916484102]: camera calibration URL: file:///home/student/.ros/camera_info/camera.yaml
[ INFO] [1583768876.916744572]: Unable to open camera calibration file [/home/student/.ros/camera_info/camera.yaml]
[ WARN] [1583768876.916827507]: Camera calibration file /home/student/.ros/camera_info/camera.yaml not found.'''

ravijo, 2020-03-09:
Assuming you are receiving images on `/cv_camera/image_raw` topic, can you please share initial few frames from your webcam? Please execute the following command and share the file via email:

```
rosbag record -l 1000 /cv_camera/image_raw
```

Ceepert, 2020-03-09:
> Assuming you are receiving images on `/cv_camera/image_raw` topic, can you please share initial few frames from your webcam? Please execute the following command and share the file via email:
> 
> ```
> rosbag record -l 1000 /cv_camera/image_raw
> ```

where is that file being saved?

Also I noticed that the wrapper does not crash, if no image is given and it just waits for an image

ravijo, 2020-03-09:
Hi @Ceepert 

> where is that file being saved?

The file is stored in your current working directory. For example, if you are executing the command from ROS workspace, you can see the file there. 

Please share 2-3 such files via email. If the file size is huge, use your preferred cloud service such as Google Drive, Dropbox etc.

Ceepert, 2020-03-09:
> Hi @Ceepert
>
>
> The file is stored in your current working directory. For example, if you are executing the command from ROS workspace, you can see the file there.
>
> Please share 2-3 such files via [...]

Ive sent you the email, it took quite a while to upload them

ravijo, 2020-03-10:
First of all, please use the latest code as recently I removed the ambiguous argument `openpose_model_dir`. From this version you need to set `model_folder` under `openpose_args` as stated [here](https://github.com/ravijo/ros_openpose#configuration).

We couldn't get much useful information from the debugger. Below are the findings:
1. The error is generated from OpenPose in function `op::resizeFixedAspectRatio`. Please make sure you have installed OpenPose as a library correctly. In short, please execute `sudo make install` to install the OpenPose system-wide.
1. I see the following message inside the debugger window: `[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]`. Unfortunately, it is a bug in gdb as stated [here](https://github.com/ros/ros_comm/issues/223). Please modify `core.launch` and put `launch-prefix="xterm -e gdb --args"` instead of launch-prefix="gdb -ex run --args" as mentioned [here](https://answers.ros.org/question/48056/using-roslaunch-with-gdb-gives-error-tcsetpgrp-failed-in-terminal_inferior-inappropriate-ioctl-for-device/?answer=218309#post-id-218309). After executing `roslaunch`, you need to type `run`. If the program stops, please type `bt full`.
1. As the error is coming from OpenPose, we can't say much right now. The debugger can't find symbol table of OpenPose as it has reported `No symbol table info available`. Therefore, you should recompile OpenPose in debugging mode and install it system-wide by using `sudo make install`.

> I did everything as you wrote, but the catkin_make commands dont work as I have used "catkin build" this far.

Open `CMakeLists.txt` file inside `ros_openpose`. After `project(ros_openpose)` line, enter the following line `set(CMAKE_BUILD_TYPE RelWithDebInfo)`

> I am also getting this warning in my camera node.
> ```
> [ INFO] [1583768876.916744572]: Unable to open camera calibration file [/home/student/.ros/camera_info/camera.yaml]
> [ WARN] [1583768876.916827507]: Camera calibration file /home/student/.ros/camera_info/camera.yaml not found.
> ```
>  Could this cause the issues?

No. Please ignore it.

> Ive sent you the email, it took quite a while to upload them

I appreciate your kind coordination. I have downloaded and run rosbag files provided by you. I have confirmed the camera is providing RGB images.
```
rostopic echo /cv_camera/image_raw | grep encoding
encoding: bgr8
```

**Finally, I have confirmed that ros_openpose is working fine.** 
Please see the attached screenshot in your email (Due to privacy reasons, I have cropped the screenshot before using it in GitHub).

[image]

_Note that the problem which you are facing could be due to a bug in OpenPose. Therefore, please keep trying and let me know the progress._

Smilels, 2020-08-27:
> Yeah the old errors disappeared, but now the node crashes after a second with this:
>
> # SUMMARY
> PARAMETERS
>
> * /echo/pub_topic: /frame
> * /rosOpenpose/cam_info_topic: cam_info_topic
> * [...]

I guess you have solved this problem. But I post the answer in here anyway.
The error comes from the opencv issue. Openpose compiles with the system opencv. But this ros package compiles with the  ros opencv.
The solution is that compiling openpose against the ros opencv instead of the system opencv.
You can OpenCV_DIR=/opt/ros/kinetic/share/OpenCV-3.3.1-dev when running cmake.
I used cmake to build and install openpose:
RUN mkdir -p $OPEN_POSE/build && cd $OPEN_POSE/build && cmake -DOpenCV_DIR=/opt/ros/kinetic/share/OpenCV-3.3.1-dev .. && cd $OPEN_POSE/build && make -j8 && make install

ravijo, 2020-08-28:
Hi @Smilels 

Thank you very much for the information.

Indeed, you are right. ros_openpose uses ROS OpenCV and [CvBridge](http://wiki.ros.org/cv_bridge) to get the data from the camera. Therefore, one needs to be careful while compiling ros_openpose and OpenPose.

Actually, I am using ROS Melodic in Ubuntu 18.04 LTS. After installing ROS, I can easily use OpenCV by `find_package(OpenCV REQUIRED)`. I did not install OpenCV separately, because ROS (full package) installs OpenCV anyways. I complied OpenPose using default OpenCV and it works perfectly.

Cheers!