ravijo/ros_openpose #9: ros_openpose with ZED2 camera

ruksen25, 2020-04-15:
Hi ravijo

I am working on getting this package working with my ZED2 camera. 
I have copied the config_realsense. launch file and tried to adapt it to my camera. 
I also similarly copied and modified the rviz file. 

Currently, I seem to be able to obtain a skeleton, and I can visualise in Rviz. However, the visualisation appears all weird. I can visualise my point cloud just fine, but of course this has also not been through manipulation in your wrapper.  It is mainly the skeleton which is visualised weird. 
The scale of it does not match the point cloud (or vice versa) , and I even in some cases seem to have a deformed skeleton. 

My question to you is if you have some parameters or variables in any parts of your wrapper that you think might be associated with the problem. Or maybe asked in another way: Can you line out the parameters/variables in your code that you think I need to look at and adapt in order to make it work with a different depth camera? 

I greatly appreciate any input you have.

ravijo, 2020-04-16:
Hi @ruksen25 

Thank you very much.

> I have copied the config_realsense. launch file and tried to adapt it to my camera.
> I also similarly copied and modified the rviz file.

Perfect, this is what I would suggest in the first place.

> Currently, I seem to be able to obtain a skeleton, and I can visualise in Rviz. However, the visualisation appears all weird. I can visualise my point cloud just fine, but of course this has also [...]

The visualization is done by python script `visualizer.py` which you can find inside `scripts` directory. The skeleton is defined by a combination of line strips. More preciously, the skeleton is decomposed into 3 line strips as following- 1) upper_body (from nose to hip), 2) hands  (from left-hand wrist to right-hand wrist), and 3) legs (from left foot toe to right foot toe). Each line segment has a width, which is set by the parameter `skeleton_line_width`. Please set it to a lower value if the lines are appearing huge. Check [here](https://github.com/ravijo/ros_openpose/blob/f8e56368444c95067aa467c178b1a032fff74d58/launch/run.launch#L37), please.

A similar question was asked previously, please go through it. Check [here](https://github.com/ravijo/ros_openpose/issues/5), please. You can find the discussion on the mentioned URL very useful.

_PS: If you can share a screenshot of the RVIZ window, we can discuss it better. You can email me the screenshot if privacy is concerned!_

Let me know if the problem persists.

ruksen25, 2020-04-16:
@ravijo 
Thank you for your reply and recommendations. 
The lines are not too thick. 
I already looked at the issue you linked to, but there seem to be some fundamental differences between my and that issue. I also looked into my camera calibration following your recomendation, but the ZED2 camera comes with a very precise calibration from the factory which they do not recommend you re-calibrate as you most likely will get a less precise result. 
after inspecting the topics published by your wrapper I have noticed that the Z value in all the points in the frame messages is always either z: 0.0010000000475 or  z: 65.5350036621 (no other values have been published). This gives me the impression that something is wrong before the visualisation. Nonetheless I have a feeling that I have a two fold issue. I attach a few images from my visualisation. The observations I have done myself is that the it seems like all the lines meet in one point (which of course should not be the case).  Secondly, this point is not aligned with the point cloud.

If I may I have a few questions to the input you use from the realsense: 
is your depth map from the realsense in mm or m and how many bits is it? 
Also which resolution are your input images? - And should I give this as an argument to the wrapper somewhere?

Explanation of the images: 
One shows how it seems like a lot of the lines are contained in one point when you zoom in. 
One other shows how all lines seem to meet in one point with an offset from the point cloud.
The last show the general view where it also looks like the scale between the skeleton and the point cloud does not match. 

Thank you for taking the time. I appreciate your help.

[image]
[image]
[image]

ravijo, 2020-04-16:
Hi @ruksen25 

Thank you very much for the information. The following information is required from the camera-

```
<!-- rostopic to subscribe for color images -->
<arg name="color_topic" default="/camera/color/image_raw" />

<!-- rostopic to subscribe for depth images -->
<arg name="depth_topic" default="/camera/aligned_depth_to_color/image_raw" />

<!-- rostopic to subscribe for camera calibration parameters -->
<arg name="cam_info_topic" default="/camera/color/camera_info" />

<!-- frame id for point cloud and 3D skeleton data -->
<arg name="frame_id" default="camera_depth_optical_frame" />
```

The trick here is that depth images are aligned to the color images. Also depth and color are having the same dimension. In this way, each pixel from the color image can be mapped to corros[doing depth pixel at the same x, y location. The `cam_info_topic` is containing camera calibration parameters supplied by the manufacturer.

Please see `compute3DPoint` function [here](https://github.com/ravijo/ros_openpose/blob/f8e56368444c95067aa467c178b1a032fff74d58/include/ros_openpose/cameraReader.hpp#L111) to understand the 2D to 3D conversation.  

> is your depth map from the realsense in mm or m?

It is in mm. You can see [here](https://github.com/ravijo/ros_openpose/blob/f8e56368444c95067aa467c178b1a032fff74d58/include/ros_openpose/cameraReader.hpp#L130) I am converting mm into m (SI Units).

> how many bits is it?

16 Bits. See [here](https://github.com/ravijo/ros_openpose/blob/f8e56368444c95067aa467c178b1a032fff74d58/src/cameraReader.cpp#L72) `TYPE_16UC1` is used as image encoding.

> Also which resolution are your input images? 

I don't remember right now. However, you can find this information online. The topic `/camera/color/image_raw` is published by RealSense Camera D435 model. The trick is to have Depth and Color in the same size and aligned as well. If not aligned, the `compute3DPoint` function has to be rewritten.

> And should I give this as an argument to the wrapper somewhere?

No. Not required.

Let me know the progress, please.

ruksen25, 2020-04-16:
Thank you very much for your help. I think I have it running now. 

Turns out I had to change a parameter in the ROS-ZED-camera node to get depth in mm. 
I had actually already tried this, but apparently it also changes the name of the depth topic which I was not aware of. When I set this parameter and passed the right topic name it worked! 

Again thank you for your time. And thank you for making this wrapper available.

closed 2020-04-16 by ruksen25

ravijo, 2020-04-17:
Hi @ruksen25 

Thank you very much. I am glad you made it work!

Since I have not used a ZED camera, I wasn't aware of its parameters. I am sorry for the inconvenience. The depth images are mostly stored in 16 Bit (Kinect also follows the same convention).

1. Meanwhile, can you please share the changes you have done to make it? 
1. For example, the `color_topic`, `depth_topic`, `cam_info_topic` and `frame_id`? 
1. How did you change the ROS-ZED-camera node to get depth in mm?
1. Are you using [this wrapper](https://github.com/stereolabs/zed-ros-wrapper) for your camera?

After receiving this information, I will update (and create `config_zed.launch`) the documentation of `ros_openpose`.

Looking forward to hearing from you.

ruksen25, 2020-04-20:
No worries! I did not expect you to know about this problem. As I already thought I tried to make these changes I thought this was not the issue and that the problem was somewhere else in the interfacing with your code.  I knew that my questions were a bit out of your scope, but I thought you might have some inputs that could hint me towards the solution. I very much appreciate your efforts in helping me find a solution. 

I already was thinking about making my zed2 solution available and thought of creating a git side on my own with the modified files and instructions.  but maybe it is nicer to have everything together in your repository. 

Maybe I can write out a section describing the use of the zed camera and send you that and the files I have created (modified from yours), which you can then add to the repository?

ravijo, 2020-04-20:
Hi @ruksen25 

> No worries! I did not expect you to know about this problem. As I already thought I tried to make these changes I thought this was not the issue and that the problem was somewhere else in the [...]

Thank you very much.

> Maybe I can write out a section describing the use of the zed camera and send you that and the files I have created (modified from yours), which you can then add to the repository?

Perfect. I agree!

AswinkarthikeyenAK, 2022-04-13:
``Hi @ruksen25,
I am trying to use the zed2 for openpose. At the moment, I am able to launch openpose fine with 
roslaunch ros_openpose run.launch synchronous:=true camera:=zed2, however, there is no skeleton visible in RViz.

I changed the openni_depth_mode:    true   

While testing with the intel camera, it seems to be fine. Whereas testing the same with ZED2, I have trouble at the moment. 
Can you please let me some other toubleshooting methods to have this fixed?
Below you will find the terminal output. 

`
SUMMARY
========

PARAMETERS
 * /echo/pub_topic: /frame
 * /rosOpenpose/cam_info_topic: /zed2/zed_node/rg...
 * /rosOpenpose/color_topic: /zed2/zed_node/rg...
 * /rosOpenpose/depth_topic: /zed2/zed_node/de...
 * /rosOpenpose/frame_id: zed2_left_camera_...
 * /rosOpenpose/no_depth: False
 * /rosOpenpose/pub_topic: /frame
 * /rosOpenpose/py_openpose_path: /usr/local/python
 * /rosdistro: melodic
 * /rosversion: 1.15.14
 * /visualizer/frame_id: zed2_left_camera_...
 * /visualizer/id_text_offset: -0.05
 * /visualizer/id_text_size: 0.2
 * /visualizer/pub_topic: /frame
 * /visualizer/skeleton_hands: False
 * /visualizer/skeleton_line_width: 0.01
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.5
 * /zed2/zed_node/depth/openni_depth_mode: True
 * /zed2/zed_node/depth/quality: 1
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
 * /zed2/zed_node/gain: 100
 * /zed2/zed_node/gamma: 8
 * /zed2/zed_node/general/base_frame: base_link
 * /zed2/zed_node/general/camera_flip: False
 * /zed2/zed_node/general/camera_model: zed2
 * /zed2/zed_node/general/camera_name: zed2
 * /zed2/zed_node/general/gpu_id: -1
 * /zed2/zed_node/general/grab_frame_rate: 30
 * /zed2/zed_node/general/resolution: 1
 * /zed2/zed_node/general/self_calib: True
 * /zed2/zed_node/general/serial_number: 0
 * /zed2/zed_node/general/svo_compression: 2
 * /zed2/zed_node/general/verbose: False
 * /zed2/zed_node/general/zed_id: 0
 * /zed2/zed_node/hue: 0
 * /zed2/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2/zed_node/mapping/mapping_enabled: False
 * /zed2/zed_node/mapping/max_mapping_range: -1
 * /zed2/zed_node/mapping/resolution: 0.05
 * /zed2/zed_node/object_detection/body_fitting: True
 * /zed2/zed_node/object_detection/confidence_threshold: 50
 * /zed2/zed_node/object_detection/max_range: 15.0
 * /zed2/zed_node/object_detection/mc_animal: True
 * /zed2/zed_node/object_detection/mc_bag: True
[... 194 lines cut ...]
[ INFO] [1649853417.749580722]: Advertised on topic /zed2/zed_node/temperature/left
[ INFO] [1649853417.749765227]: Advertised on topic /zed2/zed_node/temperature/right
[ INFO] [1649853417.749985530]: Advertised on topic /zed2/zed_node/left_cam_imu_transform [LATCHED]
[ INFO] [1649853417.751877495]: Published static transform 'zed2_imu_link' -> 'zed2_left_camera_frame'
[ INFO] [1649853417.752675049]: *** Starting Positional Tracking ***
[ INFO] [1649853417.752698187]:  * Waiting for valid static transformations...
[ INFO] [1649853417.752755762]: Static transform Sensor to Base [zed2_left_camera_frame -> base_link]
[ INFO] [1649853417.752771770]:  * Translation: {0.000,-0.060,0.000}
[ INFO] [1649853417.752786160]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1649853417.752806949]: Static transform Sensor to Camera Center [zed2_left_camera_frame -> zed2_camera_center]
[ INFO] [1649853417.752824077]:  * Translation: {0.000,-0.060,0.000}
[ INFO] [1649853417.752848295]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1649853417.752875310]: Static transform Camera Center to Base [zed2_camera_center -> base_link]
[ INFO] [1649853417.752891062]:  * Translation: {0.000,0.000,0.000}
[ INFO] [1649853417.752912857]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1649853417.853025242]: Initial ZED left camera pose (ZED pos. tracking): 
[ INFO] [1649853417.853086771]:  * T: [0,0.06,0]
[ INFO] [1649853417.853099107]:  * Q: [0,0,0,1]
[ WARN] [1649853417.853141856]: area_memory_db_path [/home/aswin/.ros/zed_area_memory.area] doesn't exist or is unreachable. 
[ INFO] [1649853418.028469968]: Positional tracking -> Searching for a known position [SEARCHING]
[ INFO] [1649853418.028535316]: Odometry aligned to last tracking pose
[ INFO] [1649853418.040120201]: Positional tracking -> OK [OK]

Thanks
Please let me know if you need more info

ravijo, 2022-04-14:
@AswinkarthikeyenAK 

>  however, there is no skeleton visible in RViz.

- Please make sure the camera is providing depth image
- Please check the depth values. I hope you are receiving depth values in millimeters or meters.
- Please print joint values using the following command:
    ```
    rostopic echo /frame
    ```

Hope it helps!

AswinkarthikeyenAK, 2022-04-14:
Hi @ravijo ,
Thanks. 
I check the points you have pointed out. 
I tried to add a print statement in the echo, and it did not get to the script. Openpose seems to work with the ZED2 camera. but for some reason when trying to launch using zed it isn't displaying anything or opening openpose for instance. I also tested with Azure, and seems to be fine. 

Thanks

ruksen25, 2022-04-24:
Hi @AswinkarthikeyenAK 

Has your issue been resolved? 

Otherwise I have a suspicion that the problem is that Stereolabs have changed the name of the depth topic published when the openni mode is used. 

Try to change the depth_topic parameter in the config_zed2.launch file to depth_registered. 

Thus, change it from: 

 <arg name="depth_topic" default="/zed2/zed_node/depth/depth_raw_registered" />

to 

 <arg name="depth_topic" default="/zed2/zed_node/depth/depth_registered" />

If this tip solves your issue, please let us know. Then @ravijo can update the launch file. 

Best regards