ravijo/ros_openpose #10: Error while compiling : error: no match for ‘operator=’ (operand types are ‘op::Matrix’ and ‘const cv::Mat’)

magi-dalmau, 2020-04-23:
Hi, i'm trying to compile ros_openpose with this hardware and software:

Ubuntu 16.04

NVIDIA GeForce 840M 
CUDA Version: 10.2 
CUDNN 7.6.5 (But I've disabled compiling caffe and openpose with cudnn , not working...) 

OpenCV 2.4.9
Caffe (openpose branch)
Openpose 1.5.1 

ROS KINETIC 

I've seen in your troubleshooting section the same error that i'm getting from my compiler but i've tryed the proposed solution (install a recent version of openpose) unsuccesfully...

I would really appreciate if you could you help me please.  I remain at your disposal for any questions do you have. 

Plese find bellow all the error throwed by the compilator:
(*NOTE: (---) = catkin workspace directory)

Scanning dependencies of target ros_openpose_generate_messages_lisp
[ 78%] Generating Lisp code from ros_openpose/Pixel.msg
(---) /src/ros_openpose/src/rosOpenpose.cpp: In member function ‘virtual sPtrVecSPtrDatum WUserInput::workProducer()’:
(---) /src/ros_openpose/src/rosOpenpose.cpp:74:33: error: no match for ‘operator=’ (operand types are ‘op::Matrix’ and ‘const cv::Mat’)
           datumPtr->cvInputData = colorImage;
                                 ^
In file included from /usr/local/include/openpose/core/array.hpp:7:0,
                 from /usr/local/include/openpose/core/common.hpp:10,
                 from /usr/local/include/openpose/3d/cameraParameterReader.hpp:4,
                 from /usr/local/include/openpose/3d/headers.hpp:5,
                 from /usr/local/include/openpose/headers.hpp:5,
                 from (---) /src/ros_openpose/src/rosOpenpose.cpp:23:
/usr/local/include/openpose/core/matrix.hpp:95:18: note: candidate: op::Matrix& op::Matrix::operator=(const op::Matrix&)
     class OP_API Matrix
                  ^
/usr/local/include/openpose/core/matrix.hpp:95:18: note:   no known conversion for argument 1 from ‘const cv::Mat’ to ‘const op::Matrix&’
/usr/local/include/openpose/core/matrix.hpp:95:18: note: candidate: op::Matrix& op::Matrix::operator=(op::Matrix&&)
/usr/local/include/openpose/core/matrix.hpp:95:18: note:   no known conversion for argument 1 from ‘const cv::Mat’ to ‘op::Matrix&&’
(---) /src/ros_openpose/src/rosOpenpose.cpp: In function ‘void configureOpenPose(op::Wrapper&, const std::shared_ptr<ros_openpose::CameraReader>&, const ros::Publisher&, const string&, bool)’:
(---) /src/ros_openpose/src/rosOpenpose.cpp:263:5: error: ‘check’ is not a member of ‘op’
     op::check(0 <= FLAGS_logging_level && FLAGS_logging_level <= 255,
     ^
(---) /src/ros_openpose/src/rosOpenpose.cpp:263:5: note: suggested alternative:
In file included from /opt/ros/kinetic/include/ros/ros.h:52:0,
                 from (---) /src/ros_openpose/src/rosOpenpose.cpp:15:
/opt/ros/kinetic/include/ros/master.h:80:18: note:   ‘ros::master::check’
 ROSCPP_DECL bool check();
                  ^
(---) /src/ros_openpose/src/rosOpenpose.cpp:274:78: error: invalid initialization of reference of type ‘const op::String&’ from expression of type ‘fLS::clstring {aka std::__cxx11::basic_string<char>}’
     const auto outputSize = op::flagsToPoint(FLAGS_output_resolution, "-1x-1");
                                                                              ^
In file included from /usr/local/include/openpose/utilities/headers.hpp:10:0,
                 from /usr/local/include/openpose/headers.hpp:44,
                 from (---) src/ros_openpose/src/rosOpenpose.cpp:23:
/usr/local/include/openpose/utilities/flagsToOpenPose.hpp:41:23: note: in passing argument 1 of ‘op::Point<int> op::flagsToPoint(const op::String&, const op::String&)’
     OP_API Point<int> flagsToPoint(const String& pointString, const String& pointExample = String("1280x720"));
                       ^
(---) /src/ros_openpose/src/rosOpenpose.cpp:277:78: error: invalid initialization of reference of type ‘const op::String&’ from expression of type ‘fLS::clstring {aka std::__cxx11::basic_string<char>}’
     const auto netInputSize = op::flagsToPoint(FLAGS_net_resolution, "-1x368");
                                                                              ^
In file included from /usr/local/include/openpose/utilities/headers.hpp:10:0,
                 from /usr/local/include/openpose/headers.hpp:44,
                 from (---) /src/ros_openpose/src/rosOpenpose.cpp:23:
/usr/local/include/openpose/utilities/flagsToOpenPose.hpp:41:23: note: in passing argument 1 of ‘op::Point<int> op::flagsToPoint(const op::String&, const op::String&)’
     OP_API Point<int> flagsToPoint(const String& pointString, const String& pointExample = String("1280x720"));
                       ^
(---) /src/ros_openpose/src/rosOpenpose.cpp:280:106: error: invalid initialization of reference of type ‘const op::String&’ from expression of type ‘fLS::clstring {aka std::__cxx11::basic_string<char>}’
     const auto faceNetInputSize = op::flagsToPoint(FLAGS_face_net_resolution, "368x368 (multiples of 16)");
                                                                                                          ^
[... 91 lines cut ...]
/usr/local/include/openpose/wrapper/wrapperStructOutput.hpp:13:19: note:   candidate expects 1 argument, 19 provided
[ 82%] Generating Lisp code from ros_openpose/Person.msg
[ 85%] Generating Lisp code from ros_openpose/BodyPart.msg
[ 89%] Generating Lisp code from ros_openpose/Frame.msg
[ 92%] Building CXX object ros_openpose/CMakeFiles/rosOpenpose.dir/src/cameraReader.cpp.o
[ 92%] Built target ros_openpose_generate_messages_lisp
Scanning dependencies of target ros_openpose_generate_messages
[ 92%] Built target ros_openpose_generate_messages
ros_openpose/CMakeFiles/rosOpenpose.dir/build.make:82: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o' failed
make[2]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 96%] Linking CXX executable (---) /devel/lib/ros_openpose/testCameraReader
/usr/bin/ld: cannot find -lThreads::Threads
collect2: error: ld returned 1 exit status
ros_openpose/CMakeFiles/testCameraReader.dir/build.make:177: recipe for target '(---) 
 /devel/lib/ros_openpose/testCameraReader' failed
make[2]: *** [(---) /devel/lib/ros_openpose/testCameraReader] Error 1
CMakeFiles/Makefile2:1442: recipe for target 'ros_openpose/CMakeFiles/testCameraReader.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/testCameraReader.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1555: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

ravijo, 2020-04-25:
Hi @magi-dalmau 

The error is generated due to the improper installation of OpenPose software. 

1. Can you first remove OpenPose completely
1. And then try to checkout OpenPose version 1.5.1

Do not forget to run `sudo make install` to install the OpenPose system-wide. I highly recommend you to look at a similar question [here](https://github.com/ravijo/ros_openpose/issues/1).

Let me know the outcome, please!

ravijo, 2020-05-06:
Hi @magi-dalmau 

Since I couldn't hear from you, I am going to close the issue. Please feel free to open and comment if the issue persists.

closed 2020-05-06 by ravijo

kledom, 2020-05-06:
I'm having the same problem with 
    Ubuntu 18.04
    ROS Melodic
    Openpose > 1.5.1
    OpenCV 3.2
    Caffe (openpose branch)

Downgrading to OpenPose 1.5.1 fixed the issue.

ravijo, 2020-05-06:
Hi @kledom 

Thank you very much for the information. 

I checked the [deployment](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/deployment.md) and [release notes](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/release_notes.md#openpose-160-apr-26-2020) of OpenPose 1.6.0 and found that OpenPose has modified the code to get rid of OpenCV in its headers. 

OpenPose 1.6.0 has introduced many new preprocessors. The preprocessor `OP_CV2OPCONSTMAT` is important to us. I believe that if we change [this line](https://github.com/ravijo/ros_openpose/blob/e91fda9c8dd26800316bd7310995e45bc44e176a/src/rosOpenpose.cpp#L74) with the following code, we should be good to go again-

```
datumPtr->cvInputData = OP_CV2OPCONSTMAT(colorImage);
```

_However, keep in mind that due to the COVID-19 pandemic, I cannot test the above workaround. I do not have access to my PC at the laboratory._

kledom, 2020-05-06:
I think there will be some more probems with 1.6.0.
For instance, I think op::check was replace with op::checkBool.

ravijo, 2020-05-06:
Hi @kledom

Thank you very much.

> For instance, op::check was replace with op::checkBool.

Noted it. I will make some workaround. For now, we should keep using OpenPose 1.5.1.