ravijo/ros_openpose #14: fatal error: ros_openpose/Frame.h: No such file or directory

LeviVasconcelos, 2020-05-19:
ros_openpose/src/rosOpenpose.cpp:19:10: fatal error: ros_openpose/Frame.h: No such file or directory
 #include <ros_openpose/Frame.h>

Is this a missing file?

LeviVasconcelos, 2020-05-20:
Hi, fixed it by adding this at line 88 of CMakeLists.txt:
add_dependencies(rosOpenpose ros_openpose_generate_messages_cpp)

closed 2020-05-20 by LeviVasconcelos

ravijo, 2020-05-20:
Thanks a lot, @LeviVasconcelos 

Normally, I re-execute the `catkin` command if such errors are shown, and then it disappears in the next run. However, it is good to have a fix for it. 

Added your suggestion in the main code. Please see [here](https://github.com/ravijo/ros_openpose/commit/583d09dfb82a3a393190825920ffa644255ff099)