ravijo/ros_openpose #15: Can't Compile

IsmaelElHouas, 2020-05-28:
Hi!

I m trying to implement ROS wrapper with Openpose tracking, but I cannot compile when I try to run with version 1.5.0, what can i do to solve the errors?

I attach the catkin_make output:
`Base path: /home/nvidia/catkin_ws
Source space: /home/nvidia/catkin_ws/src
Build space: /home/nvidia/catkin_ws/build
Devel space: /home/nvidia/catkin_ws/devel
Install space: /home/nvidia/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/nvidia/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/nvidia/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[ 10%] Built target testCameraReader
[ 10%] Built target geometry_msgs_generate_messages_eus
[ 10%] Built target geometry_msgs_generate_messages_nodejs
[ 10%] Built target std_msgs_generate_messages_nodejs
[ 10%] Built target _ros_openpose_generate_messages_check_deps_Frame
[ 10%] Built target _ros_openpose_generate_messages_check_deps_Person
[ 10%] Built target geometry_msgs_generate_messages_cpp
[ 10%] Built target std_msgs_generate_messages_cpp
[ 10%] Built target _ros_openpose_generate_messages_check_deps_Pixel
[ 10%] Built target _ros_openpose_generate_messages_check_deps_BodyPart
[ 57%] Built target ros_openpose_generate_messages_py
[ 57%] Built target ros_openpose_generate_messages_cpp
[ 57%] Built target ros_openpose_generate_messages_lisp
[ 71%] Built target ros_openpose_generate_messages_nodejs
[ 89%] Built target ros_openpose_generate_messages_eus
Scanning dependencies of target rosOpenpose
[ 89%] Built target ros_openpose_generate_messages
[ 92%] Building CXX object ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o
In file included from /usr/include/opencv/opencv2/core/core_c.h:48:0,
                 from /usr/include/opencv/opencv2/imgproc/types_c.h:46,
                 from /opt/ros/melodic/include/cv_bridge/cv_bridge.h:45,
                 from /home/nvidia/catkin_ws/src/ros_openpose/include/ros_openpose/cameraReader.hpp:17,
                 from /home/nvidia/catkin_ws/src/ros_openpose/src/rosOpenpose.cpp:20:
/usr/include/opencv/opencv2/core/types_c.h:905:0: warning: "CV_TERMCRIT_ITER" redefined
 #define CV_TERMCRIT_ITER    1
 
In file included from /usr/local/include/openpose/core/array.hpp:7:0,
                 from /usr/local/include/openpose/core/common.hpp:10,
                 from /usr/local/include/openpose/3d/cameraParameterReader.hpp:5,
                 from /usr/local/include/openpose/3d/headers.hpp:5,
                 from /usr/local/include/openpose/headers.hpp:5,
                 from /home/nvidia/catkin_ws/src/ros_openpose/src/rosOpenpose.cpp:16:
/usr/local/include/openpose/core/macros.hpp:119:0: note: this is the location of the previous definition
     #define CV_TERMCRIT_ITER cv::TermCriteria::Type::MAX_ITER
 
In file included from /usr/include/opencv/opencv2/core/core_c.h:48:0,
                 from /usr/include/opencv/opencv2/imgproc/types_c.h:46,
                 from /opt/ros/melodic/include/cv_bridge/cv_bridge.h:45,
                 from /home/nvidia/catkin_ws/src/ros_openpose/include/ros_openpose/cameraReader.hpp:17,
                 from /home/nvidia/catkin_ws/src/ros_openpose/src/rosOpenpose.cpp:20:
/usr/include/opencv/opencv2/core/types_c.h:907:0: warning: "CV_TERMCRIT_EPS" redefined
 #define CV_TERMCRIT_EPS     2
 
In file included from /usr/local/include/openpose/core/array.hpp:7:0,
                 from /usr/local/include/openpose/core/common.hpp:10,
                 from /usr/local/include/openpose/3d/cameraParameterReader.hpp:5,
                 from /usr/local/include/openpose/3d/headers.hpp:5,
                 from /usr/local/include/openpose/headers.hpp:5,
                 from /home/nvidia/catkin_ws/src/ros_openpose/src/rosOpenpose.cpp:16:
[... 563 lines cut ...]
                 from /home/nvidia/catkin_ws/src/ros_openpose/src/rosOpenpose.cpp:16:
/usr/include/opencv/opencv2/imgproc.hpp:250:5: note: previous declaration ‘cv::InterpolationFlags INTER_CUBIC’
     INTER_CUBIC          = 2,
     ^~~~~~~~~~~
In file included from /opt/ros/melodic/include/cv_bridge/cv_bridge.h:45:0,
                 from /home/nvidia/catkin_ws/src/ros_openpose/include/ros_openpose/cameraReader.hpp:17,
                 from /home/nvidia/catkin_ws/src/ros_openpose/src/rosOpenpose.cpp:20:
/usr/include/opencv/opencv2/imgproc/types_c.h:372:25: error: invalid conversion from ‘int’ to ‘<unnamed enum>’ [-fpermissive]
     CV_INTER_AREA      =3,
                         ^
/usr/include/opencv/opencv2/imgproc/types_c.h:373:25: error: invalid conversion from ‘int’ to ‘<unnamed enum>’ [-fpermissive]
     CV_INTER_LANCZOS4  =4
                         ^
/usr/include/opencv/opencv2/imgproc/types_c.h:374:1: error: expected ‘,’ or ‘;’ before ‘}’ token
 };
 ^
/usr/include/opencv/opencv2/imgproc/types_c.h:374:1: error: expected declaration before ‘}’ token
ros_openpose/CMakeFiles/rosOpenpose.dir/build.make:62: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o' failed
make[2]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o] Error 1
CMakeFiles/Makefile2:1246: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

ravijo, 2020-05-29:
Hello @IsmaelElHouas 

Thank you for using ros_openpose. I am sorry that you faced an issue.

Looking at the output, I find the following piece of error-

```
error: ‘cv’ redeclared as different kind of symbol
```

Below is more error trace-
```
In file included from /usr/local/include/openpose/core/array.hpp:7:0,
                 from /usr/local/include/openpose/core/common.hpp:10,
                 from /usr/local/include/openpose/3d/cameraParameterReader.hpp:5,
                 from /usr/local/include/openpose/3d/headers.hpp:5,
                 from /usr/local/include/openpose/headers.hpp:5,
                 from /home/nvidia/catkin_ws/src/ros_openpose/src/rosOpenpose.cpp:16:
/usr/local/include/openpose/core/macros.hpp:116:0: note: this is the location of the previous definition
     #define CV_L2 cv::NORM_L2
 
/usr/include/opencv/opencv2/imgproc/types_c.h:115:5: error: ‘cv’ redeclared as different kind of symbol
     CV_BGR2RGB     =4,
```

At line number 16 of [rosOpenpose.cpp](https://github.com/ravijo/ros_openpose/blob/583d09dfb82a3a393190825920ffa644255ff099/src/rosOpenpose.cpp#L16) OpenPose headers `#include <openpose/headers.hpp>` are included.

It seems that your OpenPose is not installed properly. The error you are facing is coming from OpenPose. 

To solve the issue, I would recommend you to update OpenPose installation because OpenPose version 1.5.0 is really old. At the same time, I would suggest you check the OpenCV compatibility with OpenPose. You may have to upgrade your OpenCV library too.

Let me know if the problem persists.

ravijo, 2020-05-29:
Closing this issue for now. Please feel free to open and comment if the problem persists.

closed 2020-05-29 by ravijo