ravijo/ros_openpose #17: catkin_make fails

sh0w, 2020-07-24:
hi!

we have [openpose-staf](https://github.com/soulslicer/openpose/tree/staf) running on our 
- Jetson Xavier with
- Ubuntu 18.04
- running ROS Melodic 

we are now trying to use your `openpose_ros` to read from a camera stream (image is sent as ROS Message) and send joint info via ros message to another package.

unfortunately 
`catkin_make` fails with this output:
```
rider@rider:~/Desktop/catkin_ws$ catkin_make ros_openpose
Base path: /home/rider/Desktop/catkin_ws
Source space: /home/rider/Desktop/catkin_ws/src
Build space: /home/rider/Desktop/catkin_ws/build
Devel space: /home/rider/Desktop/catkin_ws/devel
Install space: /home/rider/Desktop/catkin_ws/install
####
#### Running command: "cmake /home/rider/Desktop/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/rider/Desktop/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/rider/Desktop/catkin_ws/install -G Unix Makefiles" in "/home/rider/Desktop/catkin_ws/build"
####
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Using CATKIN_DEVEL_PREFIX: /home/rider/Desktop/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/rider/Desktop/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.26
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - ros_openpose
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ros_openpose'
-- ==> add_subdirectory(ros_openpose)
CMake Error at ros_openpose/CMakeLists.txt:9 (target_link_libraries):
  Cannot specify link libraries for target "ros_openpose_node" which is not
  built by this project.

-- Configuring incomplete, errors occurred!
See also "/home/rider/Desktop/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/rider/Desktop/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

any advice on this?
thanks and all the best!!
sarah

sh0w, 2020-07-24:
I fixed this error by adding [findOpenPose.cmake](https://github.com/robotology/human-sensing/blob/master/yarpOpenPose/cmake/Findopenpose.cmake#L15) to `ros_openpose/cmake/Modules`.

at the top of this file i added the line
`set(openpose_ROOT_DIR *path to my openpose*)`

then i changed  `OpenPose_LIBS` to `OpenPose_LIBRARIES` in `target_link_libraries` in `CMakeLists.txt`
so it is now finding my OpenPose.

I also hat to change 
`  ${OpenPose_INCLUDE_DIRS}
`to
`  ${OPENPOSE_INCLUDE_DIRS} 
`

closed 2020-07-24 by sh0w

ravijo, 2020-08-11:
I tested `ros_openpose` with latest OpenPose (master branch) and with OpenPose version 1.6.0. I found that there is no need to set openpose_ROOT_DIR by modifying the CMake file.

Furthermore, a change in the variable `${OpenPose_INCLUDE_DIRS}` is not needed.

I guess you forgot to execute `sudo make install` after compiling OpenPose.

_Please note that I tested these compilations inside fresh Ubuntu 18.04 LTS inside VM._

sh0w, 2020-08-11:
hello!

thank you, that makes sense now. (you're right, i forgot to run `make install`!)

thanks
sarah