ravijo/ros_openpose #21: /usr/bin/ld: cannot find -lThreads::Threads Catkin_make error

AswinkarthikeyenAK, 2020-08-15:
Hi, 
I am currently using Jetson Nano board to run ros_openpose. 
I have installed open pose version 1.5.1. 
While doing catkin_make, I am facing the following error at 97%. 

` [ 97%] Linking CXX executable /home/lams/catkin_ws/devel/lib/ros_openpose/testCameraReader
/usr/bin/ld: cannot find -lThreads::Threads
collect2: error: ld returned 1 exit status
ros_openpose/CMakeFiles/testCameraReader.dir/build.make:184: recipe for target '/home/xxyzx/catkin_ws/devel/lib/ros_openpose/testCameraReader' failed
make[2]: *** [/home/xxyzx/catkin_ws/devel/lib/ros_openpose/testCameraReader] Error 1
CMakeFiles/Makefile2:680: recipe for target 'ros_openpose/CMakeFiles/testCameraReader.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/testCameraReader.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs...`

Kindly let me know if there is a way to solve this issue please?

Thanks

ravijo, 2020-08-16:
Hi @AswinkarthikeyenAK 

I am not sure why this error is coming. Perhaps you have used custom Caffe with OpenPose installation.

Nevertheless, I always recommend checking the existing issues and troubleshooting. Please see point # 2 [here](https://github.com/ravijo/ros_openpose#troubleshooting) for the fix.

Please feel free to open the issue and comment if the problem persists.

Cheers!

closed 2020-08-16 by ravijo

AswinkarthikeyenAK, 2020-09-27:
Hi @ravijo ,
I am another issue similar to this while doing catkin_make in ros_openpose. 

`/usr/bin/ld: cannot find -lopencv_dnn
/usr/bin/ld: cannot find -lopencv_gapi
collect2: error: ld returned 1 exit status
`
Can you kindly let me know how to fix this please?

ravijo, 2020-09-28:
@AswinkarthikeyenAK 

I suspect you are compiling it against custom build OpenCV. `ros_openpose` uses OpenCV, mostly for `cv_bridge`. You need to check other forums where you can find a way to make ROS working with custom OpenCV. Probably, you need to compile ROS from the source too!

The issue doesn't seem related to ros_openpose though.