ravijo/ros_openpose #22: libcaffe.so.1.0.0 missing

Alex-Beh, 2020-08-27:
Hello, may I know whether the caffe library is needed? I try `roslaunch ros_openpose run.launch camera:=nodepth` but raise the following error:
`error while loading shared libraries: libcaffe.so.1.0.0: cannot open shared object file: No such file or directory
`

I saw you have commented those lines related to caffe library in CMakeLists.txt.

ravijo, 2020-08-27:
Hi @Alex-Beh 

The ros_openpose doesn't use Caffe directly. Instead, ros_openpose uses OpenPose, which requires the Caffe library. 

I believe that you have compiled ros_openpose successfully. Are you able to invoke OpenPose examples? I suspect that you have installed Caffe separately, which is why this issue is coming. 

Nevermind, please set the `LD_LIBRARY_PATH` variable in your terminal, just before running ros_openpose. If it works, you may keep it in your `.bashrc` file as shown below-
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/your/custom/path/libcaffe.so
```

Cheers

Alex-Beh, 2020-08-27:
Thanks for your instant reply. I try search in my directory and found that the libcaffe.so is under /usr/local/lib. I try run the following command before the roslaunch:
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib/libcaffe.so
`

But still raise the same error. Btw how to invoke the OpenPose examples? I have make install successfully for the OpenPose dependency.

ravijo, 2020-08-27:
Hi @Alex-Beh 

> I try run the following command before the roslaunch:
> `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib/libcaffe.so `

You just need to type `LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib/libcaffe.so` (without `export`). `export` is used when you write it inside `.bashrc` file.
 
> How to invoke the OpenPose examples? I have make install successfully for the OpenPose dependency.

First, I assume you have executed `sudo make install` (after `make` command). If not, please do it now. Next, to invoke OpenPose examples, please go inside OpenPose directory and execute the following command-

```
./build/examples/openpose/openpose.bin --video examples/media/video.avi
```
More info. can be found [here](https://github.com/CMU-Perceptual-Computing-Lab/openpose#quick-start).

Cheers!

Alex-Beh, 2020-08-28:
I think it would be better to run the OpenPose example and make sure all things work first before try to use ros_openpose.

I try the example and raise the following issue:
```
Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.
F0828 10:07:18.946646  7438 syncedmem.cpp:71] Check failed: error == cudaSuccess (2 vs. 0)  out of memory
```

I check the issues and they said need at least 4GB GPU memory to run the openpose.

I use nvidia-smi and below is the output:
```
|===============================+======================+======================|
|   0  GeForce MX250       Off  | 00000000:2D:00.0 Off |                  N/A |
| N/A   45C    P8    N/A /  N/A |    524MiB /  2002MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
```

May I know the minimum GPU memory needed to run the openpose example?

ravijo, 2020-08-28:
Hi @Alex-Beh 

> I think it would be better to run the OpenPose example and make sure all things work first before try to use ros_openpose.

Yes, you are right. Make sure that OpenPose is working fine first. Because ros_openpose uses OpenPose.

> I try the example and raise the following issue:
>
> ```
> Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.
> F0828 10:07:18.946646  7438 syncedmem.cpp:71] Check [...]

The example `build/examples/openpose/openpose.bin` is nothing but OpenPose itself. This is what their developer says-

```
This example summarizes all the functionality of the OpenPose library. It can ...
    1. Read a frame source (images, video, webcam, 3D stereo Flir cameras, etc.).
    2. Extract and render body/hand/face/foot keypoint/heatmap/PAF of that image.
    3. Save the results on disk.
    4. Display the rendered pose.
```
So technically, if your `build/examples/openpose/openpose.bin` works, ros_openpose should work without any issue.

Now, back to your question, I am using [GeForce GTX 1080 having 4 GB of GPU memory](https://github.com/ravijo/ros_openpose#note) and I am using the default skeleton model provided by OpenPose.  There are certain workarounds suggested by OpenPose to make it work with 2GB memory as well. Please read [here](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/faq.md#out-of-memory-error) for more info.

Cheers!

Alex-Beh, 2020-09-01:
Hello, I tried   `<arg name="openpose_args" value="--hand --disable_multi_thread --model_pose COCO --model_folder /home/alex-beh/code/openpose/models/" />` in the run.launch file which you have suggested in the README.md but still facing the same issue.

Do you have any other suggestion for the openpose to work in low memory GPU?

ravijo, 2020-09-02:
Hi @Alex-Beh 

It seems you are not following the documentation carefully. I have shared the link to OpenPose documentation in my post above. I am sharing the link again. [Please read it.](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/faq.md#out-of-memory-error)

https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/faq.md#out-of-memory-error

afoxtier, 2021-06-12:
Hello @ravijo 

I've tried setting the LD_LIBRARY_PATH to libcaffe.so.1.0.0 and libcaffe.so that are located in both usr/local/lib and those found in openpose/build/caffe/lib to no avail. I can get the openpose examples to work just fine, and I have run `sudo make install` successfully.

Thank you for your time. I hope that we can solve this problem together so that others in the future don't have to struggle.

ravijo, 2021-06-12:
@afoxtier 

Sorry for the error. Can you please explain a bit more?

Are you able to compile the `ros_openpose`? If not, then you need to set `link_directories` with the location of `libcaffe.so` as shown here
https://github.com/ravijo/ros_openpose/blob/5ccc72e91403c4c9ea4f470a2f048d5b6aa22550/CMakeLists.txt#L81-L82

Next, if you are facing the error while running the launch file, then you must check `LD_LIBRARY_PATH` once again, please. There may be a case that you are having multiple installation of OpenPose in your machine. In this situation, you can execute the OpenPose examples without any issue but `ros_openpose` may have difficulty finiding the location of correct OpenPose installation directory.

In my test machine, I can find the OpenPose is installed at the following location:
```
test@vm:~$ find /usr -name OpenPoseConfig.cmake
/usr/local/lib/OpenPose/OpenPoseConfig.cmake

test@vm:~$ find /usr -name libcaffe.so
/usr/lib/x86_64-linux-gnu/libcaffe.so
```

The `OpenPoseConfig.cmake` is the most important file which is used while compiling the  `ros_openpose`.

afoxtier, 2021-06-14:
@ravijo 

Thank you for the very quick response!

I can compile `ros_openpose` just fine by following the readme instructions and the subsequent FAQ.

I wanted to double check some stuff before proceeding. It seems like everything is working just fine. I just had to properly set up my camera in `config_nodepth.launch` to the proper camera. I apologize for the unnecessary pings.

Thank you for your time.

Now to just figure out how to use the coordinates topic for my project. :)

ravijo, 2021-06-14:
@afoxtier 

Thank you very much. Based on your feedback, I am closing this issue now. If you face any issues, please let me know via [the issue page](https://github.com/ravijo/ros_openpose/issues).

closed 2021-06-14 by ravijo