ravijo/ros_openpose #24: Ros Node Dies after few seconds - ZED2 camera

AswinkarthikeyenAK, 2020-09-29:
Hi @ravijo ,
I am currently using ros_openpose with zed2 camera. 
I had tested it couple of days back where everything seemed to work fine.
But unfortunately I am running into issues now with regards to the ros node.

After I launch the ros file `roslaunch ros_openpose run.launch camera:=zed2`, 

The camera connection is successful but after few seconds I get the following msg:
`EQUIRED process [rosOpenpose-2] has died!
process has died [pid 21215, exit code -11, cmd /home/lams/catkin_ws/devel/lib/ros_openpose/rosOpenpose --model_folder /home/lams/openpose/models/ --net_resolution -256x96 __name:=rosOpenpose __log:=/home/lams/.ros/log/178878be-023a-11eb-8de0-d8c0a6261973/rosOpenpose-2.log].
log file: /home/lams/.ros/log/178878be-023a-11eb-8de0-d8c0a6261973/rosOpenpose-2*.log
Initiating shutdown!
` 
I cannot not figure out why this error is occurring. Openpose works fine with the camera which i tested again.

I am attaching the log files for your reference. Kindly let me know if you can help me fix this issue.

Thanks
[echo-4.log](https://github.com/ravijo/ros_openpose/files/5299658/echo-4.log)
[master.log](https://github.com/ravijo/ros_openpose/files/5299660/master.log)
[master.log](https://github.com/ravijo/ros_openpose/files/5299661/master.log)
[roslaunch-lams-desktop-21169.log](https://github.com/ravijo/ros_openpose/files/5299662/roslaunch-lams-desktop-21169.log)
[roslaunch-lams-desktop-21169.log](https://github.com/ravijo/ros_openpose/files/5299663/roslaunch-lams-desktop-21169.log)
[rosout.log](https://github.com/ravijo/ros_openpose/files/5299664/rosout.log)
[rosout.log](https://github.com/ravijo/ros_openpose/files/5299665/rosout.log)
[rosout-1-stdout.log](https://github.com/ravijo/ros_openpose/files/5299667/rosout-1-stdout.log)
[rosout-1-stdout.log](https://github.com/ravijo/ros_openpose/files/5299668/rosout-1-stdout.log)
[visualizer-3.log](https://github.com/ravijo/ros_openpose/files/5299670/visualizer-3.log)
[visualizer-3.log](https://github.com/ravijo/ros_openpose/files/5299671/visualizer-3.log)

ravijo, 2020-09-30:
Hi @AswinkarthikeyenAK 

It is difficult to say what is wrong here. Unfortunately, I don't own a ZED2 camera. So I can not debug your issue exactly. 

However, looking into your [master.log](https://github.com/ravijo/ros_openpose/files/5299661/master.log) file, this is what I found-

```
[rosmaster.master][INFO] 2020-09-29 10:57:57,299: publisherUpdate[/tf_static] -> http://lams-desktop:33065/ ['http://lams-desktop:37247/']: sec=0.00, exception=<Fault -1: 'publisherUpdate: unknown method name'>
[rosmaster.threadpool][ERROR] 2020-09-29 10:57:57,307: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1316, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1493, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 800, in close
    raise Fault(**self._stack[0])
Fault: <Fault -1: 'publisherUpdate: unknown method name'>
```

This is what causing the ROS node to shutdown. I am sorry, you need to debug from here. Please let me know the progress and your fix for it.

Thank you very much for your kind understanding.

AswinkarthikeyenAK, 2020-09-30:
Hi @ravijo 
Thanks for your reply. 
I'll let you know once i debug the issue.

AswinkarthikeyenAK, 2020-10-07:
Hi @ravijo 
I had initially managed to fix the issue and everything seemed to work fine. 
But since couple of days back I am not getting proper 3d points in Rviz. 
Please find the image attached.
@ruksen25, Do you think you can help me out here if possible,please?
Really appreciate your assistance
I have checked and made sure I made the necessary changes to the ZED Ros_wrapper and changed this 
`openni_depth_mode:          true                           # 'false': 32bit float meters, 'true': 16bit uchar millimeters` 

Thanks 

[image]

ravijo, 2020-10-08:
Hi @AswinkarthikeyenAK 

I am glad to hear from you.

> I had initially managed to fix the issue and everything seemed to work fine.

Can you please tell us what the problem was? It will help us to understand the issue properly.

> But since couple of days back I am not getting proper 3d points in Rviz.

Do you mean that earlier, you were getting correct a 3D view? It is difficult to understand the issue right now. You should better tell us more.

Right now, I see that the actual scene is hidden behind a green color object. I suspect that this green color is coming from ros_openpose. Can you please decrement the values of the following variables-

- `skeleton_line_width` in [run.launch](https://github.com/ravijo/ros_openpose/blob/f089d7f0c6a2cfbe4b5adf421421e9f20d965932/launch/run.launch#L37)
- `id_text_size` in [run.launch](https://github.com/ravijo/ros_openpose/blob/f089d7f0c6a2cfbe4b5adf421421e9f20d965932/launch/run.launch#L40)

It would help if you also gave it a try by upsetting the `skeleton` flag in [run.launch](https://github.com/ravijo/ros_openpose/blob/f089d7f0c6a2cfbe4b5adf421421e9f20d965932/launch/run.launch#L15)

ravijo, 2020-10-16:
Hi @AswinkarthikeyenAK 

I assume your problem is fixed now. Therefore, I am closing this issue now. Feel free to open if the issue persist!

-
Cheers!

closed 2020-10-16 by ravijo

AswinkarthikeyenAK, 2020-10-18:
Hi @ravijo,
Yes it is fixed now. Sorry wasn't able to update it. 
I dont know what caused this issue. 
After reinstalling ros_openpose it seemed to have fix the problem now.

vipulgarg1998, 2020-12-19:
>  I had initially managed to fix the issue and everything seemed to work fine.

How you managed to resolve this? I am facing same error and got stuck with this

ravijo, 2020-12-21:
@vipulgarg1998:

It is always suggested to share detailed information about the issue. Moreover, we do not face such an issue in our dev machine.

Kindly reinstall and comment back if the problem persists.