ravijo/ros_openpose #25: Question about Depth when using IntelRealsense

Shunichi09, 2020-10-12:
Hi. Thank you for your great code!! I installed and ran this repository successfully due to your kind instruction.

However, the depth value which is contained in Frame topic might be strange...
(the value only has  1.0 or 2.0 or 3.0 or 4.0 or 5.0) 
It's supposed to have float values like 1.1.

Do you have any ideas about this things?

When I echoed the Frame, I got the following results

```
score: 0.218902811408
pixel: 
  x: 444.264190674
  y: 146.671981812
point: 
  x: 0.198683813214
  y: -0.153367757797
  z: 1.0  # Here
[INFO] [1602521053.435068]: Human Point estimation bodypart score: 0.393784701824
pixel: 
  x: 382.929473877
  y: 318.964385986
point: 
  x: 0.198683813214
  y: -0.153367757797
  z: 1.0  # Here
[INFO] [1602521053.435608]: Human Point estimation bodypart score: 0.100059762597
pixel: 
  x: 365.940765381
  y: 444.267089844
point: 
  x: -2.12089323997
  y: -1.57605838776
  z: 4.0  # Here
[INFO] [1602521053.476026]: Human Point estimation bodypart score: 0.224799275398
pixel: 
  x: 458.610015869
  y: 140.16998291
point: 
  x: 0.22222109139
  y: -0.16403567791
  z: 1.0  # Here
[INFO] [1602521053.476766]: Human Point estimation bodypart score: 0.341619074345
pixel: 
  x: 397.282806396
  y: 318.949279785
point: 
  x: 0.22222109139
  y: -0.16403567791
  z: 1.0  # Here
```

ravijo, 2020-10-13:
Hi @Shunichi09 

Thank you very much. I am glad you could follow the instructions and made it work.

> (the value only has 1.0 or 2.0 or 3.0 or 4.0 or 5.0)
> It's supposed to have float values like 1.1.

This is surprising. Last time, I checked everything was working fine. I doubt maybe the values you are seeing are just a coincidence? Did you check it many times with multiple frames? Furthermore, if the RVIZ visualization is working fine, then there should not be any problem.

Now back to your query, the values are represented in SI units inside the frame topic. Therefore, if the depth values are in millimeters, a conversion from millimeter to the meter is required. This conversation is done based on the image encoding and can be seen [here](https://github.com/ravijo/ros_openpose/blob/f089d7f0c6a2cfbe4b5adf421421e9f20d965932/src/cameraReader.cpp#L80).

Let me know if the issue persists. I will check once again at my side.

Shunichi09, 2020-10-13:
Thank you for your reply.
In my case the RVIZ visualization worked fine, but the depth information only took int values like 1.0, 2.0...
It means that the human bone which was drawn by green line in RVIZ moved not smoothly.
(only Z position, X and Y position worked fine!!)

> Did you check it many times with multiple frames? 

Yes, I did...

> Now back to your query, the values are represented in SI units inside the frame topic. Therefore, if the depth values are in millimeters, a conversion from millimeter to the meter is required. This conversation is done based on the image encoding and can be seen here.

I'm using D435i of IntelRealSense so I think depth values are in millimeters.
Does it mean your code should work fine ??

ravijo, 2020-10-14:
Hello @Shunichi09 

I checked the code and found the same issue. It is a bug introduced in the recent commit. Therefore, can you please use an old commit as shown below-
```
git checkout 583d09dfb82a3a393190825920ffa644255ff099
```

Meanwhile, I am working on the bug! Stay tuned.

Shunichi09, 2020-10-14:
@ravijo Thank you for your support!!
Ok, I will use the old version of your repository. I'm looking forward to your updating.
Again, Thank you very much.

ravijo, 2020-10-16:
Hello @Shunichi09 

Can you please pull the latest changes from the repository and check if the problem is solved?

Thanks

Shunichi09, 2020-10-19:
I ran your latest code and there was no errors!! Thank you for fixing the bugs !!

ravijo, 2020-10-19:
Hi @Shunichi09 

Thanks for the confirmation. I am closing this issue now. Feel free to report issues if you find any!

closed 2020-10-19 by ravijo