ravijo/ros_openpose #27: My development is on the win10, can I not use the ROS system and transplant to Win10, visualization with open3d or any other solution instead?

WestbrookZero, 2020-12-24:


ravijo, 2020-12-25:
@WestbrookZero 

Hi,

You can use ROS in Windows 10. A simple google search returned the following 2 results in my workstation:

1. http://wiki.ros.org/Installation/Windows
2. https://raspberry-valley.azurewebsites.net/ROS-on-Windows-10/

Once you have ROS up and working in Windows 10, you can use RViz easily. Why waste time on using other tools when there are simpler tools? However, yes, you can use Open3D. They provide Python APIs.

Cheers!

closed 2020-12-25 by ravijo

WestbrookZero, 2020-12-25:
Thanks for your answer, I install foxy ros version on win10, but run python have the error : ModuleNotFoundError: No module named 'rospy' , i don't know how to solve it.  and also 'catkin_init_workspace' is not an internal or external command, nor is it a running program or batch file.

ravijo, 2020-12-26:
@WestbrookZero 

To be honest, I have never used ROS inside Windows. I am a Linux guy, for that matter.

The errors reported by you truly indicates that your ROS installation is broken. Kindly do a Google Search and follow the results. By now, I believe that ROS support for Windows is not buggy.

Good luck!