ravijo/ros_openpose #28: Using Xtion Camera

ROSNovice, 2021-01-05:
Hi @ravijo  thank you for your sharing code
i have tried to run your code using the Xtion camera. but the marker array not appear in the point cloud2 Rviz. here my setting

  <arg name="color_topic" default="/camera/rgb/image_raw" />

  
  <arg name="depth_topic" default="/camera/depth_registered/image_raw" />

  
  <arg name="cam_info_topic" default="/camera/rgb/camera_info" />

  
  <arg name="frame_id" default="camera_link" />

  
  <arg name="no_depth" default="true" />

the RGB work very well but the depth not showing any result of marker array

ravijo, 2021-01-06:
Hello, @ROSNovice 

Thanks for your kind appreciation.

Unfortunately, I don't have an Xtion camera. However, do not worry. This package is written in such a way that it is possible to use any kind of camera supported by ROS. As per the information you have provided, it seems that the ros_openpose is not receiving any depth images from ROS. I strongly recommend you to check if `color_topic`, `depth_topic`, `cam_info_topic`, and `frame_id` are set properly. In this regard, let me share more information for a better understanding. Please check [faq](https://github.com/ravijo/ros_openpose#faq) and [issue 9](https://github.com/ravijo/ros_openpose/issues/9).

I hope you can make the Xtion camera with ros_openpose. Furthermore, please don't hesitate commenting below if you face any problem in this problem.

Cheers!

ravijo, 2021-01-22:
Hello, @ROSNovice

I am assuming that the problem has been resolved. Therefore, I am closing this issue. Please feel free to open it and comment in case of trouble.

Cheers!

closed 2021-01-22 by ravijo