ravijo/ros_openpose #29: Error running on Ubuntu 20.04 with RealSense

SM30000, 2021-01-20:
Hello @ravijo,

first of all thanks for your great work, this is exactly what i was searching for!

 I'm trying to run the ROS wrapper on Ubuntu 20.04, ROS noetic and OpenPose 1.7 (using CUDA 11.1 and cuDNN 8.05) with Intel RealSense 435. Running OpenPose on it's own is working fine so far. The problem is when i try to run your project within my ros workspace it doesn't work, i think due to those errors: `[visualizer-3] process has died [pid 22718, exit code 127, cmd /home/fabian/catkin_ws/src/ros_openpose/scripts/visualizer.py` and `process has died [pid 22719, exit code 127, cmd /home/fabian/catkin_ws/src/ros_openpose/scripts/echo.py`. See attached log files for more information.

Is this issue related to my OpenPose version since you've recommended version 1.6?

Logs:
[master.log](https://github.com/ravijo/ros_openpose/files/5841899/master.log)
[roslaunch-fabian-MS-7C76-22692.log](https://github.com/ravijo/ros_openpose/files/5841901/roslaunch-fabian-MS-7C76-22692.log)
[rosout.log](https://github.com/ravijo/ros_openpose/files/5841902/rosout.log)
[rosout-1-stdout.log](https://github.com/ravijo/ros_openpose/files/5841903/rosout-1-stdout.log)
[rviz-7-stdout.log](https://github.com/ravijo/ros_openpose/files/5841904/rviz-7-stdout.log)

ravijo, 2021-01-21:
Hello @SM30000 

I am glad that you find it useful.

Yeah, you are right. The `visualizer.py` and `echo.py` are somehow unable to work as expected.

After `OpenPose v1.6`, I couldn't check the code with newer versions so far. But as you said, the `OpenPose` standalone works without any issue, so together we need to find out the culprit. Therefore, I am looking for your kind support in this matter.

Luckily, it is possible to disable `visualizer.py` and `echo.py` by command line. Can you please execute the following command-

```
roslaunch ros_openpose run.launch camera:=realsense skeleton:=false print:=false
```

Furthermore, I would like you to manually invoke realsenese camera to see if it is all right. To do so, kindly print the color, depth and camera info topics as suggested in the instructions below:

**Color Topic**
```
rostopic echo /camera/color/image_raw
```

**Depth Topic**
```
rostopic echo /camera/aligned_depth_to_color/image_raw
```

**Camera Info. Topic**
```
rostopic echo /camera/color/camera_info
```

`ros_openpose` uses these 3 information from the realsense camera. It passes the color image to `OpenPose` and then uses depth + camera info to compute cartesian coordinates in 3 dimensions.

Therefore, it is necessary above 3 `rostopic echo` command must print something in the terminal. Please use <kbd>Ctrl</kbd> + <kbd>C</kbd> to stop these echo commands.

Let me know the outcome, please.

SM30000, 2021-01-28:
Hello @ravijo,

Sorry for the delay.

Since I'm using ROS Noetic and therefore Python3, I just had to call the scripts `echo.py` and `visualizer.py` with Python3 instead of Python2 by editing the first line of the scripts to `#!/usr/bin/env python3` and it's working fine right now.

Thank's a lot!

Regards,
Fabian

closed 2021-01-28 by SM30000

ravijo, 2021-01-29:
@SM30000 

Glad you made it work :+1: