ravijo/ros_openpose #30: run.launch error occurr

RollingInTheBed, 2021-01-22:
roslaunch ros_openpose run.launch
... logging to /home/peanut/.ros/log/1e6eeb3a-5c8b-11eb-99ad-48b02d2f863c/roslaunch-peanut-desktop-9587.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://peanut-desktop:34859/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 316 lines cut ...]
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 48, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 8, in <module>
    from loader import *
  File "/usr/lib/python2.7/dist-packages/yaml/loader.py", line 4, in <module>
    from reader import *
  File "/usr/lib/python2.7/dist-packages/yaml/reader.py", line 47, in <module>
    class Reader(object):
  File "/usr/lib/python2.7/dist-packages/yaml/reader.py", line 140, in Reader
    NON_PRINTABLE = re.compile(u'[^\x09\x0A\x0D\x20-\x7E\x85\xA0-\uD7FF\uE000-\uFFFD\U00010000-\U0010ffff]')
  File "/usr/lib/python2.7/re.py", line 194, in compile
    return _compile(pattern, flags)
  File "/usr/lib/python2.7/re.py", line 249, in _compile
    p = sre_compile.compile(pattern, flags)
  File "/usr/lib/python2.7/sre_compile.py", line 576, in compile
    code = _code(p, flags)
  File "/usr/lib/python2.7/sre_compile.py", line 558, in _code
    _compile_info(code, p, flags)
  File "/usr/lib/python2.7/sre_compile.py", line 536, in _compile_info
    _compile_charset(charset, flags, code)
  File "/usr/lib/python2.7/sre_compile.py", line 232, in _compile_charset
    flags & SRE_FLAG_UNICODE):
  File "/usr/lib/python2.7/sre_compile.py", line 274, in _optimize_charset
    r = range(av[0], av[1]+1)
KeyboardInterrupt

ravijo, 2021-01-22:
Hello @RollingInTheBed 

Sorry to know that you couldn't make it run.  

Thank you for sharing the logs. From these logs, I can see the following critical information:
 
```
error while loading shared libraries: libcaffe.so.1.0.0: cannot open shared object file: No such file or directory
```

It means that OpenPose software is not properly installed. 

* First of all, please try to invoke OpenPose software standalone by executing the `openpose.bin` file. [Check here for more info](https://github.com/CMU-Perceptual-Computing-Lab/openpose#quick-start). 

* Next, please make sure that you have executed `sudo make install`.

* Finally, if you have installed the Caffe library by yourself, you need to provide the location of it. To be precise, you need to provide the location of `libcaffe.so` file inside your `CMakeLists.txt` file. [Please see here for more information](https://github.com/ravijo/ros_openpose/blob/cd92d0d7828de3ab9e2f49d867a189531d2dcdcd/CMakeLists.txt#L79).

Please feel free to revert back, if the issue persists.

RollingInTheBed, 2021-01-22:
problem solved by adding the following command in the Ubuntu 18.04 ~/.bashrc:
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/usr/local/lib

RollingInTheBed, 2021-01-22:
@ravijo 
Thx for the reply!
Thank you very much~

closed 2021-01-22 by RollingInTheBed

ravijo, 2021-01-23:
@RollingInTheBed :+1: 

Glad you made it work so quickly