ravijo/ros_openpose #32: "make -j8 -l8" Error!

sacittezcanlar, 2021-02-21:
Hi Ravijo,
I have an issue with catkin_make; I've done the steps that you say but unf. got a error like this: 

[ 93%] Built target ros_openpose_generate_messages_cpp
Scanning dependencies of target rosOpenpose
[ 94%] Building CXX object ros_openpose/CMakeFiles/rosOpenpose.dir/src/cameraReader.cpp.o
[ 96%] Building CXX object ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o
[ 96%] Built target ros_openpose_generate_messages_eus
Scanning dependencies of target ros_openpose_generate_messages
[ 96%] Built target ros_openpose_generate_messages
[ 98%] Linking CXX executable /home/sacit/catkin_ws/devel/lib/ros_openpose/testCameraReader
/usr/bin/ld: cannot find -lThreads::Threads
collect2: error: ld returned 1 exit status
ros_openpose/CMakeFiles/testCameraReader.dir/build.make:197: recipe for target '/home/sacit/catkin_ws/devel/lib/ros_openpose/testCameraReader' failed
make[2]: *** [/home/sacit/catkin_ws/devel/lib/ros_openpose/testCameraReader] Error 1
CMakeFiles/Makefile2:1448: recipe for target 'ros_openpose/CMakeFiles/testCameraReader.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/testCameraReader.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable /home/sacit/catkin_ws/devel/lib/ros_openpose/rosOpenpose
/usr/bin/ld: cannot find -lThreads::Threads
collect2: error: ld returned 1 exit status
ros_openpose/CMakeFiles/rosOpenpose.dir/build.make:197: recipe for target '/home/sacit/catkin_ws/devel/lib/ros_openpose/rosOpenpose' failed
make[2]: *** [/home/sacit/catkin_ws/devel/lib/ros_openpose/rosOpenpose] Error 1
CMakeFiles/Makefile2:1873: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/all] Error 2
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

My system infos:
Ubuntu 18.04/ Ros - Melodic / Cuda 10

Thank you already.

ravijo, 2021-02-22:
Hello @sacittezcanlar 

Sorry to hear that you couldn't compile it properly. From the above logs, I can find the following error:
```
/usr/bin/ld: cannot find -lThreads::Threads
```
[Please look at point number 2 in the Troubleshooting section](https://github.com/ravijo/ros_openpose#troubleshooting).

Let me know if the issue persists.

sacittezcanlar, 2021-02-22:
Thank you so much i've complied it successfully. Can i use it with my laptop camera and how ?

ravijo, 2021-02-22:
@sacittezcanlar it is always advised to read the documentation carefully first. [Please check here.](https://github.com/ravijo/ros_openpose#steps-to-run-with-any-color-camera-such-as-webcam-etc)

Thanks

sacittezcanlar, 2021-02-22:
Thank you so much for your interest, i'm such a new guy here ^^

closed 2021-02-22 by sacittezcanlar

ravijo, 2021-02-23:
No problem @sacittezcanlar Glad you could find the relevant information.