ravijo/ros_openpose #33: OpenPose 1.7.0

sacittezcanlar, 2021-02-24:
Hello again dear Ravijo,
[image]

`sacit@sacit-N550JK:~/catkin_ws$ source devel/setup.bash 
sacit@sacit-N550JK:~/catkin_ws$ roslaunch ros_openpose run.launch camera:=nodepth
... logging to /home/sacit/.ros/log/18ec5e7a-76be-11eb-bb0f-8019347b1b23/roslaunch-sacit-N550JK-23959.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sacit-N550JK:36235/

SUMMARY
========

PARAMETERS
 * /echo/pub_topic: /frame
 * /rosOpenpose/cam_info_topic: cam_info_topic
 * /rosOpenpose/color_topic: /image_view/output
 * /rosOpenpose/depth_topic: depth_topic
 * /rosOpenpose/frame_id: no_depth
 * /rosOpenpose/no_depth: True
 * /rosOpenpose/pub_topic: /frame
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    echo (ros_openpose/echo.py)
    rosOpenpose (ros_openpose/rosOpenpose)

auto-starting new master
process[master]: started with pid [23970]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 18ec5e7a-76be-11eb-bb0f-8019347b1b23
process[rosout-1]: started with pid [23982]
started core service [/rosout]
process[rosOpenpose-2]: started with pid [23985]
process[echo-3]: started with pid [23990]
[ INFO] [1614184441.773653131]: Starting ros_openpose...
Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.
[ WARN] [1614184441.823631514]: Waiting for color image frame...
[ WARN] [1614184441.823919102]: Waiting for datum...
`
Normally i can reach to OpenPose and it works but when i'm trying to connect by roslaunch i have a problem like this.
Can you help me please? Thank you very much again.

ravijo, 2021-02-25:
Can you please check if there is any data available in the `color_topic` simply by using the following command:
```
rostopic echo /image_view/output
```
If you don't see anything, you need to check if you have already started your camera (webcam) via ROS. If you have not started, please start it first before starting `ros_openpose`. If you have already started your camera, please make sure to assign the correct  value of `color_topic`

sacittezcanlar, 2021-02-25:
Dear Ravijo, 
I've had with usb_cam; my version was broken and  i uninstalled, installed again and I can work it now. I have 2.8fps but i'll fix the params and i'll make it much optimized. :) Thank you so much your kindness and support.

closed 2021-02-25 by sacittezcanlar