ravijo/ros_openpose #35: error: no matching function for call to ‘op::WrapperStructPose::WrapperStructPose

KeundongKang, 2021-02-27:
Hello, I'm having trouble with catkin_make because the error below keeps occuring.

`[ 98%] Built target transform_nodelet
/home/isr_gd/isr_gd_ws/src/ros_openpose/src/rosOpenpose.cpp: In function ‘void configureOpenPose(op::Wrapper&, const std::shared_ptr<ros_openpose::CameraReader>&, const ros::Publisher&, const string&, bool)’:
/home/isr_gd/isr_gd_ws/src/ros_openpose/src/rosOpenpose.cpp:392:70: error: no matching function for call to ‘op::WrapperStructPose::WrapperStructPose(<brace-enclosed initializer list>)’
                                                   enableGoogleLogging};
                                                                      ^
In file included from /usr/local/include/openpose/wrapper/wrapper.hpp:13:0,
                 from /usr/local/include/openpose/wrapper/headers.hpp:6,
                 from /usr/local/include/openpose/headers.hpp:47,
                 from /home/isr_gd/isr_gd_ws/src/ros_openpose/src/rosOpenpose.cpp:16:
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:221:9: note: candidate: op::WrapperStructPose::WrapperStructPose(op::PoseMode, const op::Point<int>&, double, const op::Point<int>&, op::ScaleMode, int, int, int, float, op::RenderMode, op::PoseModel, bool, float, float, int, const op::String&, const std::vector<op::HeatMapType>&, op::ScaleMode, bool, float, int, bool, double, const op::String&, const op::String&, float, bool)
         WrapperStructPose(
         ^~~~~~~~~~~~~~~~~
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:221:9: note:   no known conversion for argument 3 from ‘const op::Point<int>’ to ‘double’
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note: candidate: op::WrapperStructPose::WrapperStructPose(const op::WrapperStructPose&)
     struct OP_API WrapperStructPose
                   ^~~~~~~~~~~~~~~~~
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note:   candidate expects 1 argument, 26 provided
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note: candidate: op::WrapperStructPose::WrapperStructPose(op::WrapperStructPose&&)
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note:   candidate expects 1 argument, 26 provided
ros_openpose/CMakeFiles/rosOpenpose.dir/build.make:75: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o' failed
make[2]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/src/rosOpenpose.cpp.o] Error 1
CMakeFiles/Makefile2:5279: recipe for target 'ros_openpose/CMakeFiles/rosOpenpose.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenpose.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed
`
It would be really thankful to solve this issue.

ravijo, 2021-02-28:
@KeundongKang 

I suspect about the OpenPose version you are using. Can you please use find the version of the OpenPose installed on your machine? Please read point number 3 in the [FAQ](https://github.com/ravijo/ros_openpose#faq) for the same.

So far, I have not tested `ros_openpose` with OpenPose 1.7.0. Please check [Supported OpenPose Versions](https://github.com/ravijo/ros_openpose#supported-openpose-versions) which are confirmed already. Well, my intuition says that the newer version (i.e., OpenPose 1.7.0) should work well too because there are no changes in APIs as stated [here](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/08_release_notes.md#openpose-170-nov-15-2020).

Finally, If you have OpenPose v != 1.6.0 and you still face the problem, I recommend using OpenPose v1.6.0. Please revert back with the outcome.

KeundongKang, 2021-02-28:
I found that my openpose version was 1.7.0.
So I changed it to 1.6.0 and now it works.

Thanks for the kind reply!

ravijo, 2021-02-28:
Hi @KeundongKang 

I debug the issue in detail. Below is what I found:

I guess you cloned the master branch of OpenPose. The `master` branch is the current branch. [Although it says OpenPose v1.7.0](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/1f1aa9c59fe59c90cca685b724f4f97f76137224/include/openpose/core/macros.hpp#L13), you should always check out from Git `tag` to confirm the [actual OpenPose 1.7.0](https://github.com/CMU-Perceptual-Computing-Lab/openpose/tree/v1.7.0). Please read point #1 in [troubleshooting](https://github.com/ravijo/ros_openpose/blob/master/README.md#troubleshooting). 

Just now, I have pushed a few changes to `ros_openpose`. These changes support OpenPose 1.7.0. You may consider pulling the changes. However, make sure to checkout tag 1.7.0. 

Closing this issue now.

closed 2021-02-28 by ravijo