ravijo/ros_openpose #36: Current ros_openpose_synchronous.py crashes with openpose 1.7.0

luorui93, 2021-03-29:
This is probably due to the change of emplaceAndPop() in new op_wrapper.
I've updated the code in my fork here:
https://github.com/luorui93/ros_openpose/commit/97ad917bebb30818280e4b4ab11474a800e6822a

I also made some changes to the code so the program won't crash when nobody or only partial body is detected. 
Should I make a pull request or something? Sorry I'm not very familiar with the standard procedure.

ravijo, 2021-03-30:
Hello @luorui93 

> This is probably due to the change of emplaceAndPop() in new op_wrapper.
> I've updated the code in my fork here:
> [luorui93@97ad917](https://github.com/luorui93/ros_openpose/commit/97ad917bebb30818280e4b4ab11474a800e6822a)

Thank you very much for reporting this issue.

Unfortunately, the ros_openpose_synchronous.py was tested with OpenPose 1.6.0 only. Please see the first item under [Implementation Versions Info. ](https://github.com/ravijo/ros_openpose#implementation-versions-info) section. So it is understandable that this bug may have arrived.

> I also made some changes to the code so the program won't crash when nobody or only partial body is detected.
> Should I make a pull request or something? Sorry I'm not very familiar with the standard procedure.

Yes, please make a pull request. 

Meanwhile, I am tagging @QuantuMope who kindly provided this version.

@luorui93 
As you said, based on your recent [commit](https://github.com/luorui93/ros_openpose/commit/97ad917bebb30818280e4b4ab11474a800e6822a), the `emplaceAndPop` is changed. I am writing down the changes so that QuantuMope can understand quickly without reading your commit.

**Before**
```
op_wrapper.emplaceAndPop([datum])
```

**Now**
```
op_wrapper.emplaceAndPop(op.VectorDatum([datum]))
```

The above changes once done may break support for OpenPose older than 1.7. I can think of a workaround using a flag inside the Python code as shown below (this is a pseudo-code):
```
OPENPOSE1POINT7_OR_HIGHER = op.__version__ >= 1.7.0

if OPENPOSE1POINT7_OR_HIGHER:
    op_wrapper.emplaceAndPop(op.VectorDatum([datum]))
else:
    op_wrapper.emplaceAndPop([datum])
```

- Please suggest if there is a better way!
- I think other changes such as checking a null object, i.e., `None` should work without any workaround.

Looking forward to hearing opinions from everyone.

QuantuMope, 2021-03-30:
@ravijo, Yeah, I remember seeing this difference between Openpose 1.7 and 1.6. I was originally planning on incorporating something like your pseudocode (similar to how the asynchronous module was fixed to handle Openpose 1.7), but I must have forgotten about it.

Good news is that the only code that needs to be changed are the lines you identified so it's a simple fix.

Just submitted request #37. Should work now for versions 1.6+.

@luorui93, please submit a pull request to fix the nobody or partial body crashing. I didn't catch that problem when I was testing it but glad that you did.

ravijo, 2021-03-30:
@QuantuMope Thank you so much for the quick response! 

@luorui93 Please do the needful as suggested by @QuantuMope. Please do not forget to [pull the latest changes](https://github.com/ravijo/ros_openpose/commit/3b8eb672b56a6b5a7da0df1472d319d11e16bcbe) before you make a pull request.

luorui93, 2021-03-30:
@ravijo Merged the latest changes and submitted a new pull request.

QuantuMope, 2021-03-31:
@luorui93
If the synchronous node is now working on your end for Openpose 1.7.0, don't forget to close the issue.

closed 2021-03-31 by luorui93