ravijo/ros_openpose #41: Errors running synchronous API

chrisywong, 2021-04-01:
Continuation of https://github.com/ravijo/ros_openpose/issues/31

Original problem:
```
[ERROR] [1617024825.486954]: OpenPose library could not be found. Did you enable `BUILD_PYTHON` in CMake and have this Python script in the right folder?
ImportError: cannot import name pyopenpose
```
Solved this issue by making sure I was building using similar Python versions (I have both 2.7 and 3.6 on my computer). Probably not the most kosher method, but CMake was still picking up remnants of a 3.7 install, so I had to manually modify values in `build/CMakeCache.txt` to change Python version to 2.7).

I did a lot of other things in between so I'm not sure if those helped too. 

Once that built properly and I could `import pyopenpose` properly, ran into the next problem:
```
  File "/home/cwong/src_UdeS/catkin_ws/src/ros_openpose/scripts/ros_openpose_synchronous.py", line 104, in callback
    self.op_wrapper.emplaceAndPop([datum])
TypeError: emplaceAndPop(): incompatible function arguments. The following argument types are supported:
    1. (self: openpose.pyopenpose.WrapperPython, arg0: std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > >) -> bool

Invoked with: <openpose.pyopenpose.WrapperPython object at 0x7fac4ffe2e30>, [<openpose.pyopenpose.Datum object at 0x7fac4ff475b0>]

Did you forget to `#include <pybind11/stl.h>`? Or <pybind11/complex.h>, <pybind11/functional.h>, <pybind11/chrono.h>, etc. Some automatic conversions are optional and require extra headers to be included when compiling your pybind11 module.
```
Solved it via https://github.com/CMU-Perceptual-Computing-Lab/openpose/issues/1718 by editing Line 104 in `ros_openpose_synchronous.py` from `self.op_wrapper.emplaceAndPop([datum])` to `self.op_wrapper.emplaceAndPop(op.VectorDatum([datum]))`

Current problem:
```
  File "/home/cwong/src_UdeS/catkin_ws/src/ros_openpose/scripts/ros_openpose_synchronous.py", line 111, in callback
    if pose_kp.shape == ():
AttributeError: 'NoneType' object has no attribute 'shape'
```
I solved this by changing it to `If pose_kp == None`, but I don't know if this is a valid way to do it, or I'll run into problems later on? (The same error happens below with `lhand_kp` and `rhand_kp`, and can be bypassed the same way too.)

(Still haven't got it running, but they're Caffe-related errors)

ravijo, 2021-04-01:
@chrisywong 

Thank you very much for explaining the problem.

Yes, you are on the right track. We must use the same python which is being used by ROS. You did it well :+1: 

Next, about the `emplaceAndPop`, this is a change in OpenPose API v1.7.0 onwards. This issue was discussed by us as well [here](https://github.com/ravijo/ros_openpose/issues/36).

Finally, `pose_kp.shape == ()` issue was also discussed here and [fixed too](https://github.com/ravijo/ros_openpose/blob/e85fe34fc93050f18862cfacbc7167aec176f42a/scripts/ros_openpose_synchronous.py#L118).

Please pull the latest commit from the `ros_openpose` and feel free to report in case of any issues.

Thanks

chrisywong, 2021-04-02:
Oooh oops I didn't see those other issues/latest updates. And yes, pulling the latest version fixed everything and now I have synchronous images! Thanks for the help!

ravijo, 2021-04-02:
Thanks for the update. I am glad that you made it work. I am closing this issue now. Please feel free to comment or create a new one if any problem occurs!

closed 2021-04-02 by ravijo