ravijo/ros_openpose #44: no matching function for call to ‘op::WrapperStructPose::WrapperStructPose(<brace-enclosed initializer list>)’ round 2

AntonBock, 2021-04-30:
Hello.

I have the same error as you address in your troubleshoot guide point #1, and issue #35. However, the solution to checkout v. 1.7.0 is not working for me. Am I missing something?

Br
Anton

ravijo, 2021-05-01:
Hey @AntonBock 

Are you sure you are using OpenPose v1.7.0? I guess you forgot to execute `sudo make install` in OpenPose. Therefore, the OpenPose inside your `/usr` directory is an old one.

Please read [here](https://github.com/ravijo/ros_openpose/issues/35#issuecomment-787436302) and add the following in your CMakeLists.txt file:

```
MESSAGE(STATUS "OpenPose VERSION: " ${OpenPose_VERSION})
```

Or you can use the `get_openpose_version.sh` script as stated in the above hyperlink. Please let me know your findings if you still can't make it work.

Cheers!

AntonBock, 2021-05-03:
Hey @ravijo, 

Thanks, I'll try that :D

ravijo, 2021-05-22:
Closing this issue, as there is no response from the author of this issue. Please feel free to comment if the issue still exists.

closed 2021-05-22 by ravijo