ravijo/ros_openpose #46: Fix suggestion for skeleton markers not showing on RViz when using ros_openpose_synchronous.py with azurekinect camera

jyp0802, 2021-06-03:
First of all, thanks for your work.

When launching `run.launch` with `synchronous:=true` and `camera:=azurekinect`,
the skeleton markers doesn't show on the point clouds in RViz.
Instead if you look carefully at the origin (0,0) you can see markers all together.
It's probably because the depth values of azurekinect are at a different scale than for realsense.

**So when I changed the lines 150, 154, 158 in `ros_openpose_synchronous.py` to multiply 1000, it worked fine.**

**Like this -> `b_XYZ = self.compute_3D_vectorized(pose_kp, depth) * 1000`**

But this is only for the azurekinect. For the realsense camera, it needs to be without the `* 1000`.
(I haven't tried the other cameras though)

Also, when I don't use `ros_openpose_synchronous.py` (i.e. `synchronous:=false`) then it works fine without any changes.

ravijo, 2021-06-03:
@jyp0802 

Thank you so much for reporting issues and contributing to improve ros_openpose.

I remember that the depth encoding is different in the case of azure kinect. The non-synchronous version, which was implemented in C++ keeps a note of this information. Please [see the highlighted code here](https://github.com/ravijo/ros_openpose/blob/13a5bdf24c30d561010d4eeb1b6eda2c47b04512/src/cameraReader.cpp#L80-L87).

As you can notice that in the case of '16UC1' encoding, the depth values are represented in millimeters. Therefore a manual conversion from millimeter to meters is required.

In short, you have to implement a similar conditional statement in the `scripts/ros_openpose_synchronous.py` file.

Please let me know if you need more help. Furthermore, if you can incorporate the workaround, I kindly request you to make a pull request so that it can be merged here.

Finally, I am tagging contributors @QuantuMope  and @luorui93 to keep them informed about the possible changes!

ravijo, 2021-06-08:
fixed in https://github.com/ravijo/ros_openpose/commit/5ccc72e91403c4c9ea4f470a2f048d5b6aa22550

@QuantuMope Thank you very much for the pull request!

@jyp0802 Please let us know if it works fine.

I am keeping this issue open for a few days so that @jyp0802 can comment easily

jyp0802, 2021-06-09:
I've just checked the new commit with the azurekinect and it works fine! Thanks a lot :smile:

closed 2021-06-09 by jyp0802