ravijo/ros_openpose #47: Question about Azure Kinect Depth information

jmberd02, 2021-06-05:
Hi, I apologize if this is an ignorant question, however I am having trouble understanding how to get depth information from the Azure Kinect using this package. I definitely get 2d output in the console, however, I do not see a Z entry, is there some way to get that information?

Here is a sample output of the information while testing at my desk. 

Thank you so much for your help!!

``` [INFO] [1622857957.348458]: [x: 0.0
y: 0.0, x: 1377.77819824
y: 515.407470703, x: 1670.69921875
y: 620.101135254, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 1105.66638184
y: 415.028503418, x: 682.829711914
y: 674.433959961, x: 385.547424316
y: 850.206054688, x: 1189.40771484
y: 1164.23413086, x: 1369.34387207
y: 1160.03320312, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 1628.87231445
y: 301.859985352, x: 1319.22668457
y: 117.640922546, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0]

[INFO] [1622857957.372276]: [x: 0.0
y: 0.0, x: 1377.73901367
y: 515.395507812, x: 1679.06237793
y: 628.458557129, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 1101.47973633
y: 419.004608154, x: 691.223449707
y: 670.295593262, x: 0.0
y: 0.0, x: 1189.33251953
y: 1118.19421387, x: 1344.23791504
y: 1126.5501709, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 1624.6829834
y: 281.055999756, x: 1294.05151367
y: 88.4597396851, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0, x: 0.0
y: 0.0]
```

ravijo, 2021-06-06:
@jmberd02 

Thanks a lot for creating this issue. I hope you are running the following command:

```
roslaunch ros_openpose run.launch camera:=azurekinect
```

> I definitely get 2d output in the console, however, I do not see a Z entry

You are right. That's because we have set to print only pixel information. Please see the code below:

https://github.com/ravijo/ros_openpose/blob/13a5bdf24c30d561010d4eeb1b6eda2c47b04512/scripts/echo.py#L14-L15

You can notice that `bodyPart.pixel` is used here to print in the console. It denotes 2D information.

> is there some way to get that information?

Yes. There is. Please edit `bodyPart.pixel` to `bodyPart.point` as shown below:

```
text = [bodyPart.point for person in msg.persons for bodyPart in person.bodyParts]
rospy.loginfo('%s\n' % text)
```

Extra Information
-------

Just in case, if you are running the synchronous API version, you must be using it in the following way:
```
roslaunch ros_openpose run.launch camera:=azurekinect synchronous:=true py_openpose_path:=absolute_path_to_py_openpose
```
However, recently a bug has been discovered in the synchronous API version. Please see [here](https://github.com/ravijo/ros_openpose/issues/46) for more information.

jmberd02, 2021-06-07:
That worked thank you so much for your help!!!

closed 2021-06-07 by jmberd02