ravijo/ros_openpose #49: Expected FPS from ROS Openpose

jmberd02, 2021-06-10:
Hi, 
    I am sorry if this question has been answered, however, I am trying to gauge what performance I should be seeing from the following config:

-  i7-7700K
- Ubuntu 18.04
- Cuda 11.3
- Geforce GTX Titan X
I get about 5 fps according to the openpose gui that pops up.0

Prior to the titan (which I installed today) I was using a 1060 6gb and I was getting 3 fps. Does it make sense for there to be such a small performance difference?

Thank you for your help!

ravijo, 2021-06-11:
@jmberd02 

Thanks for creating the issue. At present, I don't remember the exact FPS value, sorry about it. But it was more than 10. FPS depends on many factors. 5 FPS is defiantly not a good sign.

To answer the FPS value, I just searched across the issue and found this https://github.com/ravijo/ros_openpose/issues/23

I never used GTX Titan X. By the way, which camera are you using? I suspect that either your camera is set to lower FPS or the camera resolution is too high.

I strongly recommend you first execute the OpenPose demo, which is a standalone program. Notice the FPS during this time as it will give you a baseline.

Please feel free to revert back.

jmberd02, 2021-06-11:
@ravijo 

Thank you so much for your quick reply, I am running an Azure Kinect so I will try altering the camera config. I also just ran the Openpose video example and was getting only 1.8 fps. Which I guess tells me that there is something wrong with my openpose installation? Hopefully I dont have to reinstall everything again...

ravijo, 2021-06-14:
@jmberd02 

Thank you very much.

>  I am running an Azure Kinect so I will try altering the camera config. 

Azure Kinect provides HD (or larger) images. In fact, I also noticed the miserable fps while using Azure Kinect. I remember discussing it with users in [the issue section](https://github.com/ravijo/ros_openpose/issues) in the past. Therefore, you should configure Azure Kinect to output smaller images (Or else you need to increase your hardware).

> I also just ran the Openpose video example and was getting only 1.8 fps. 

This is unpleasant. I can't believe it!!! Are you sure it is 1.8 fps ? It should be 18 or more! Kindly check once again. Make sure you are using the latest version of OpenPose. Make sure you have cuDnn installed.

jmberd02, 2021-06-22:
@ravijo 

I apologize, I didnt get notified on your previous comment. when running the video example I am getting the following
[image]

I am using the following
[image]

The following is the openpose from cmake-gui is as follows
[image]

Please let me know if this is helpful!

Thank you!

ravijo, 2021-06-23:
@jmberd02

Thank you very much for sharing more information.

By seeing the output of your `nvidia-smi`, it is clear that OpenPose is not using GPU. Instead it is using CPU and thus the FPS is low (expected).

Although, you have checked CUDA options in cmake, but I am worried that OpenPose could not build properly with CUDA. 

- I highly recommend you to do a fresh install and keep monitoring the build log.
- Please check OpenPose documentation to find the supported CUDA/CuDNN version for your GPU.

ravijo, 2021-07-11:
Closing this issue as no activity is observed for 15 days.

closed 2021-07-11 by ravijo