ravijo/ros_openpose #50: No Bone Visualization in RViz with 2D camera

afoxtier, 2021-06-17:
Problem:
When I open RViz, I cannot find any visualizations by topic that display the skeleton in the viewer.

Launch Procedure:
`$roscore
$rosparam set cv_camera/device_id 2
$rosrun cv_camera cv_camera_node
$roslaunch ros_openpose run.launch
$rviz`

Listed Topics while Running:
`/clicked_point
/cv_camera/camera_info
/cv_camera/image_raw
/cv_camera/image_raw/compressed
/cv_camera/image_raw/compressed/parameter_descriptions
/cv_camera/image_raw/compressed/parameter_updates
/cv_camera/image_raw/compressedDepth
/cv_camera/image_raw/compressedDepth/parameter_descriptions
/cv_camera/image_raw/compressedDepth/parameter_updates
/cv_camera/image_raw/theora
/cv_camera/image_raw/theora/parameter_descriptions
/cv_camera/image_raw/theora/parameter_updates
/frame
/initialpose
/move_base_simple/goal
/rosOpenpose/cam_info_topic
/rosOpenpose/depth_topic
/rosout
/rosout_agg
/tf
/tf_static`

It should be noted that none of the /rosOpenpose/... topics are visible in RViz at all even with the non-displayable topics selection checked.

Lastly, I am very new to ROS as a whole and so I feel like there may just be a step to running all of this that I am missing due to a gap in my knowledge. Thank you for your time.

ravijo, 2021-06-17:
Hello @afoxtier 

To see the bone (skeleton) visualization in RViz, we need depth information too. For example, using Intel RealSense camera or any other depth camera. 

In your case, you are using a normal camera, it is difficult to get the depth information. Therefore, you can not see bone (skeleton) visualization in RViz.

Nevertheless, you can still get the skeleton 2D/pixel data by running the following command:

```
rostopic echo /frame
```

Hope it helps!

afoxtier, 2021-06-17:
Dear @ravijo 

Thank you for your help. It seems that I may need to get a stereo / depth camera to proceed with this project much further.

closed 2021-06-17 by afoxtier