ravijo/ros_openpose #52: Please confirm the newly added citation info.

ravijo, 2021-06-30:
Dear all,

I recently added [citation info.](https://github.com/ravijo/ros_openpose#citation) so that people in academics can cite this work. I am tagging everyone so that you can receive a notification from GitHub. 

- @ruksen25 
- @xiangzhi 
- @QuantuMope 
- @luorui93 

Can you please check if I entered your name correctly? 

FYI, I followed this notation: `author = {Last Name, First Name Middle Name}`. The Last name is basically your family name, sometimes know as sir name too.

As always, thank you very much for your contribution.

xiangzhi, 2021-07-01:
Looks good to me. Thanks!

QuantuMope, 2021-07-02:
Also looks good to me!

ravijo, 2021-07-17:
I am closing this issue because of no activity for some time. Please feel free to comment if needed.

closed 2021-07-17 by ravijo

ruksen25, 2021-09-01:
Sorry for the Kate reply.  My name should be van den Broek, Marike K.

ravijo, 2021-09-01:
Done! Thanks @ruksen25