ravijo/ros_openpose #54: Very Messy Skeleton in RViz But Clean in Openpose

afoxtier, 2021-07-24:
Hello,

Does anyone have any suggestions on how to clean up my skeleton besides creating a noise filter in ROS to deal with the below noise? It seems like Openpose is not having any trouble with the skeleton, but somewhere along the way the skeleton gets mixed up. 

Thank you for your help.

[image]

ravijo, 2021-07-24:
@afoxtier 

Thank you for sharing the video. It makes easy to understand the problem. I have following two suggestions:

1. Please see the left side in the RViz window, where you have checkbox in front of each visualization item. From you video, it seems that skeleton markers are facing some issue. That is why you can notice that the markers are shown as "warning" and sometimes change to "error". Please investigate the cause of it. Most probably some of the marker data is invalid.

2. Please run the Synchronous API  version via setting `synchronous:=true` while invoking launch file. Please read [Implementation Versions Info.](https://github.com/ravijo/ros_openpose#implementation-versions-info) for more info.

Let me know if the problem persists.

afoxtier, 2021-08-06:
Thank you for your suggestions. After following the first part of #41 to enable synchronous mode, I was able to get to the below state. I still have those errors listed about the skeleton (they mention uninitialized quaternions). I think there is another issue thread that mentions that problem.

Thank you again! 

[image]

ravijo, 2021-08-09:
@afoxtier 

I just pushed a fix for "uninitialized quaternions" warning shown in Rviz. Please pull the latest changes and try once more.

Let me know if the issue persists.

Cheers!

afoxtier, 2021-08-10:
@ravijo 

Thank you for addressing this problem. However, I cannot seem to get the new version to work. When I run it `rosrun ros_openpose run.launch` I get the error:

Synchronous = true
`ERROR: cannot launch node of type [ros_openpose/rosOpenposeSync]: Cannot locate node of type [rosOpenposeSync] in package [ros_openpose]. Make sure file exists in package path and permission is set to executable (chmod +x)
` 
or 

Synchronous = false
`ERROR: cannot launch node of type [ros_openpose/rosOpenposeAsync]: Cannot locate node of type [rosOpenposeAsync] in package [ros_openpose]. Make sure file exists in package path and permission is set to executable (chmod +x)
`
All of the files in `ros_openpose/scripts` have execution permissions. I don't see any other issues that mention these terms and don't recall having to fix them earlier. Perhaps my installation notes missed a step...

ravijo, 2021-08-10:
@afoxtier 

It is strange. However no changes were done to `rosOpenposeSync` and `rosOpenposeAsync` file. I believe you have not executed the following command:

```
source devel/setup.bash
````

Please compile the repository again and let me know.

afoxtier, 2021-08-10:
@ravijo

I neglected to run `catkin_make` before. I've amended my notes to include that step of installation. I no longer have the errors above, but now I'm getting many `[Error] [DepthPacketStreamParser] Packet buffer is NULL` errors. The errors start right after Freenect2DeviceImpl submits depth transfers.

There are other errors afterwards but I suspect that they are results of this one for the most part. Your advice would be appreciated. Thank you.

ravijo, 2021-08-11:
@afoxtier 

> I neglected to run catkin_make before. I've amended my notes to include that step of installation. I no longer have the errors above, 

Glad to know

> but now I'm getting many `[Error] [DepthPacketStreamParser] Packet buffer is NULL` errors. 

It seems you are using Kinect v2 with `iai_kinect2` package. The error you mentioned is related to the  `iai_kinect2` package. I found following relevant information to fix your error:

- https://github.com/code-iai/iai_kinect2/issues/499
- https://github.com/r9y9/pylibfreenect2/issues/34

Hope it helps

afoxtier, 2021-08-11:
@ravijo 

After solving the packet buffer issue, the console feed made it clear that that was not actually the issue. I had improperly written the path to the openpose models folder.

Now that that works, here is the current state of things with the default mode set to synchronous:
[image]

You can see that it is much better than the first attempt, but not as stable as it was before your patch. Occasionally some errors in RViz will show up about empty points and not being able to draw lines. I'll post the exact errors once I switch back to Ubuntu.

ravijo, 2021-08-12:
@afoxtier 

> Points should not be empty for specified marker type.
> At least two points are required for a LINE_STRIP marker.

These errors are obvious to me. From your video above, we can see that there are times when few body parts are occluded and OpenPose can not find them. That is why Rviz can not display and shows the error.

You may consider adding a conditional statement to confirm that there are at least 2 points before publishing the skeleton to Rviz. Please edit [visualizer.py](https://github.com/ravijo/ros_openpose/blob/af1fc8f0c2edb435648e5cf1e6ddaeaa121e79fa/scripts/visualizer.py)

Cheers!

ravijo, 2021-09-01:
I am closing this issue because of no activity for some time. Please feel free to comment.

closed 2021-09-01 by ravijo

AswinkarthikeyenAK, 2022-04-10:
Hi, 
I am experiencing the same issue with open pose 1.7.0. 
[image]

I have set the <arg name="synchronous" default="true" /> in run.launch

How to fix this issue?

ravijo, 2022-04-11:
@AswinkarthikeyenAK 

There could be many reasons for this messy skeleton. The `ros_openpose` uses depth values to compute the 3D points. Can you please check if the depth values aren't noisy in the first place? For example, depending on the environment, a depth camera may perform poorer.

If you are using a RealSense D-4xx camera, I suggest you decrease the alpha parameter of the temporal filter. It helps to reduce the fluctuation between depth images.

Hope it helps!

AswinkarthikeyenAK, 2022-04-12:
Thanks for your reply. 
Do you mean to set the temporal filter in the rs_camera.launch?
Also, I noticed that the joint data is good until 4 m and it becomes jittery after 4m. Is this something you have experienced ? From what I understood, the depth of intel camera is good for longer distances, or is it normal to experience this jitters? 

Thanks again

ravijo, 2022-04-14:
@AswinkarthikeyenAK 

> Do you mean to set the temporal filter in the rs_camera.launch?

Yes. I think the default value for alpha is 0.4. Please make it smaller.

> Also, I noticed that the joint data is good until 4 m and it becomes jittery after 4m. Is this something you have experienced? 

No. Unfortunately, so far, I haven't used it for longer distances.

> From what I understood, the depth of the intel camera is good for longer distances, or is it normal to experience this jitters?

It is completely a normal phenomenon. However, depending on your application, I suggest you fine-tune camera parameters.

> Also, after setting temporal filter, does the data publish in the default ros topic or does it publish the filtered data as new topic to which ros_openpose should subscribe to? 

It is published on the same/default topic.

> At the moment I am subscribing to <arg name="depth_topic" default="/camera/aligned_depth_to_color/image_raw"

Correct. 

On a side note, please visualize the depth images before and after applying the temporal filter. You can get a good idea about jitters.

@AswinkarthikeyenAK

There could be many reasons for this messy skeleton. The ros_openpose uses depth values to compute the 3D points. Can you please check if the depth values aren't noisy in the first place? For example, depending on the environment, a depth camera may perform poorer.

If you are using a RealSense D-4xx camera, I suggest you decrease the alpha parameter of the temporal filter. It helps to reduce the fluctuation between depth images.

Hope it helps!

AswinkarthikeyenAK, 2022-04-14:
Perfect!! I understand. 
Would you be able to share your rs_camera.launch file, please? I can set the filter, but I am not sure where to change the alpha parameters. It doesn't seem to be a straight forward step to do it. Thanks

ravijo, 2022-04-14:
@AswinkarthikeyenAK 

There are many ways to set camera parameters in Intel RealSense. I should tell you the interactive way by using GUI. Please bring up your camera and in another terminal execute the following command:

```
rosrun rqt_reconfigure rqt_reconfigure
```

In the third terminal, you should keep monitoring the quality of depth images.

AswinkarthikeyenAK, 2022-04-14:
@ravijo ,
I did the following:
1. In the rs_camera.launch, I added `<arg name="filters"                   default="temporal"/>`
2. Launched the realsense camera `roslaunch realsense2_camera rs_camera.launch`
3. ran rosrun rqt_reconfigure rqt_reconfigure in another terminal.
4. I was able to change the alpha and typing in the following command was able to see that the there was change in the alpha factor.  `rosrun dynamic_reconfigure dynparam get /camera/temporal`, result `{'filter_smooth_alpha': 0.63, 'stream_index_filter': -1, 'groups': {'filter_smooth_alpha': 0.63, 'parent': 0, 'stream_index_filter': -1, 'groups': {}, 'id': 0, 'name': 'Default', 'parameters': {}, 'state': True, 'stream_format_filter': 1, 'stream_filter': 1, 'filter_smooth_delta': 20, 'type': '', 'holes_fill': 3, 'frames_queue_size': 16}, 'stream_format_filter': 1, 'stream_filter': 1, 'filter_smooth_delta': 20, 'holes_fill': 3, 'frames_queue_size': 16}`
5. I am unable to notice changes in the depth images in the other terminal -> `rosrun rqt_image_view rqt_image_view` subscribed to the ` "/camera/aligned_depth_to_color/image_raw"`. Is there some other way to notice these changes or the quality of the depth images. 

Thank you for your time!!!

mobeeniqbal101, 2022-11-17:
@ravijo  Hi, I am a noob in ROS and Linux, can you please explain what will be the absolute directory for my openpose python node ? which has to be passed in order to run in synchronous mode. In my openpose directory I can see a python folder which has the following contents :
[image]

I have tried giving this path as "/home/openpose/python/openpose/" but it gives me the following error:
[image]

I do remember enabling python in CMAKE, do you have any suggestions, what could be the issue ?

mobeeniqbal101, 2022-11-18:
@ravijo  I read in one of your codes that the default directory should be usr/local/python . I tried this directory as well but still I am getting the same error, can you please look into this issue ?

[image]

ravijo, 2022-11-19:
@mobeeniqbal101 

> ... what will be the absolute directory for my openpose python node ? 

After compiling OpenPose, you should have a build directory, where you should be able to see a openpose.so (or pyopenpose.so) file. Go to that directory and try to import OpenPose inside python terminal as shown below:

```
from openpose import pyopenpose as op
```

If this works, you need to remember that directory and provide it as `py_openpose_path`.

By the way, please always create a new issue. You are commenting under an issue, which is marked as "closed".