ravijo/ros_openpose #57: compile problem!

Mni-coder, 2021-12-28:
Hello ravijo,
hier I met problems by compiling ROS-wrapper for openpose in your project. Hier is what I met:
`[ 86%] Linking CXX executable /home/jade/catkin_openpose/devel/lib/ros_openpose/rosOpenposeSync
[ 90%] Linking CXX executable /home/jade/catkin_openpose/devel/lib/ros_openpose/rosOpenposeAsync
/usr/bin/ld: warning: libopencv_imgcodecs.so.3.2, needed by /opt/ros/melodic/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.405
CMakeFiles/rosOpenposeAsync.dir/src/rosOpenposeAsync.cpp.o：在函数‘configureOpenPose(op::WrapperT<op::Datum, std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > >, std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > >, std::shared_ptr<op::Worker<std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > > > > >&, std::shared_ptr<ros_openpose::CameraReader> const&, ros::Publisher const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)’中：
rosOpenposeAsync.cpp:(.text+0x29f)：对‘op::ConfigureLog::setPriorityThreshold(op::Priority)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x2af)：对‘op::Profiler::setDefaultX(unsigned long long)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x2c5)：对‘op::String::String(char const*)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x2de)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x2fe)：对‘op::flagsToPoint(op::String const&, op::String const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x332)：对‘op::String::String(char const*)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x34b)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x36b)：对‘op::flagsToPoint(op::String const&, op::String const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x39f)：对‘op::String::String(char const*)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x3b8)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x3d8)：对‘op::flagsToPoint(op::String const&, op::String const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x40c)：对‘op::String::String(char const*)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x425)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x445)：对‘op::flagsToPoint(op::String const&, op::String const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x470)：对‘op::flagsToPoseMode(int)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x48f)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x49e)：对‘op::flagsToPoseModel(op::String const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x5ec)：对‘op::flagsToScaleMode(int)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x61f)：对‘op::flagsToHeatMaps(bool, bool, bool)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x62c)：对‘op::flagsToHeatMapScaleMode(int)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x646)：对‘op::flagsToDetector(int)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x659)：对‘op::flagsToDetector(int)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x6ed)：对‘op::WrapperT<op::Datum, std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > >, std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > >, std::shared_ptr<op::Worker<std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > > > > >::setWorker(op::WorkerType, std::shared_ptr<op::Worker<std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > > > > const&, bool)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x73c)：对‘op::WrapperT<op::Datum, std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > >, std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > >, std::shared_ptr<op::Worker<std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > > > > >::setWorker(op::WorkerType, std::shared_ptr<op::Worker<std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > > > > const&, bool)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x78f)：对‘op::flagsToRenderMode(int, bool, int)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x7eb)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x84f)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x868)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x924)：对‘op::WrapperStructPose::WrapperStructPose(op::PoseMode, op::Point<int> const&, op::Point<int> const&, op::ScaleMode, int, int, int, float, op::RenderMode, op::PoseModel, bool, float, float, int, op::String const&, std::vector<op::HeatMapType, std::allocator<op::HeatMapType> > const&, op::ScaleMode, bool, float, int, bool, double, op::String const&, op::String const&, float, bool)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x96e)：对‘op::WrapperT<op::Datum, std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > >, std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > >, std::shared_ptr<op::Worker<std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > > > > >::configure(op::WrapperStructPose const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x990)：对‘op::flagsToRenderMode(int, bool, int)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x9de)：对‘op::WrapperStructFace::WrapperStructFace(bool, op::Detector, op::Point<int> const&, op::RenderMode, float, float, float)’未定义的引用
rosOpenposeAsync.cpp:(.text+0x9f7)：对‘op::WrapperT<op::Datum, std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > >, std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > >, std::shared_ptr<op::Worker<std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > > > > >::configure(op::WrapperStructFace const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xa34)：对‘op::flagsToRenderMode(int, bool, int)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xa8a)：对‘op::WrapperStructHand::WrapperStructHand(bool, op::Detector, op::Point<int> const&, int, float, op::RenderMode, float, float, float)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xaa3)：对‘op::WrapperT<op::Datum, std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > >, std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > >, std::shared_ptr<op::Worker<std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > > > > >::configure(op::WrapperStructHand const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xadf)：对‘op::WrapperStructExtra::WrapperStructExtra(bool, int, bool, int, int)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xaf8)：对‘op::WrapperT<op::Datum, std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > >, std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > >, std::shared_ptr<op::Worker<std::shared_ptr<std::vector<std::shared_ptr<op::Datum>, std::allocator<std::shared_ptr<op::Datum> > > > > > >::configure(op::WrapperStructExtra const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xb18)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xb27)：对‘op::stringToDataFormat(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xb43)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xb5c)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
rosOpenposeAsync.cpp:(.text+0xb83)：对‘op::String::String(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
[... cut ...]
make[2]: *** [/home/jade/catkin_openpose/devel/lib/ros_openpose/rosOpenposeSync] Error 1
CMakeFiles/Makefile2:1527: recipe for target 'ros_openpose/CMakeFiles/rosOpenposeSync.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenposeSync.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`
I don't know if it's because I didn't install Openpose with the choice "with OpenCV  with OpenGL"

thank you for the help

the chinese charactors above means "no matching to"

ravijo, 2021-12-29:
@Mni-coder 

Thanks for posting this issue.

I am having a hard time due to the many Chinese characters in the error message. Anyway, I suspect checking the following:

1. ROS Melodic by default uses OpenCV 3.2. It seems that you have OpenCV 4.05 as well. The linker is showing a warning!
2. Please make sure that you have installed OpenPose by executing the following command: `sudo make install`
3. Please make sure to check out the latest OpenPose tag using the following command `git checkout tags/v1.7.0`

Mni-coder, 2021-12-29:
I just check on my OpenCV Version with ' pkg-config --modversion opencv' and it's said what I have installed is right 3.2.0. And I am sure step 2&3 are correct. Just feel no idea about it....

ravijo, 2021-12-30:
@Mni-coder 

The errors you have mentioned above are coming during the linking process. The compiler can not find matching symbols.

Can you please show what command was executed and what was reported at the terminal from the beginning?

ravijo, 2022-01-15:
Closing this issue as no future comments received. Feel free to comment if the issue still exists.

closed 2022-01-15 by ravijo