ravijo/ros_openpose #58: Can't compile.

ChrisTiberius, 2022-02-08:
Hello.

I have OpenPose version 1.7.0
I get the expected error when i execute catkin_make.
"git checkout tags/v1.7.0" does not help.
Al tough i have to say i am not sure whats meant with "root directory of OpenPose installation"
I ran it at:
/home/christian/openpose    and
/home/christian/openpose/build

Any help is appreciated.

ChrisTiberius, 2022-02-08:
ROS_HOSTNAME=>10.0.0.121<
ROS_IP=>10.0.0.121<
christian@christian-P64-HJ-HK1:~/catkin_ws$ catkin_make
Base path: /home/christian/catkin_ws
Source space: /home/christian/catkin_ws/src
Build space: /home/christian/catkin_ws/build
Devel space: /home/christian/catkin_ws/devel
Install space: /home/christian/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/christian/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/christian/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  3%] Linking CXX executable /home/christian/catkin_ws/devel/lib/ros_openpose/testCameraReader
[  3%] Built target geometry_msgs_generate_messages_eus
[  3%] Built target std_msgs_generate_messages_eus
[  3%] Built target std_msgs_generate_messages_nodejs
[  3%] Built target _ros_openpose_generate_messages_check_deps_Person
[  3%] Built target _ros_openpose_generate_messages_check_deps_Frame
[  3%] Built target _ros_openpose_generate_messages_check_deps_Pixel
[  3%] Built target geometry_msgs_generate_messages_nodejs
[  3%] Built target _ros_openpose_generate_messages_check_deps_BodyPart
[ 20%] Built target ros_openpose_generate_messages_py
[ 36%] Built target ros_openpose_generate_messages_eus
[ 53%] Built target ros_openpose_generate_messages_cpp
[ 63%] Built target ros_openpose_generate_messages_nodejs
[ 76%] Built target ros_openpose_generate_messages_lisp
[ 80%] Building CXX object ros_openpose/CMakeFiles/rosOpenposeSync.dir/src/rosOpenposeSync.cpp.o
[ 83%] Building CXX object ros_openpose/CMakeFiles/rosOpenposeAsync.dir/src/rosOpenposeAsync.cpp.o
/usr/bin/ld: cannot find -lThreads::Threads
collect2: error: ld returned 1 exit status
ros_openpose/CMakeFiles/testCameraReader.dir/build.make:192: recipe for target '/home/christian/catkin_ws/devel/lib/ros_openpose/testCameraReader' failed
make[2]: *** [/home/christian/catkin_ws/devel/lib/ros_openpose/testCameraReader] Error 1
CMakeFiles/Makefile2:432: recipe for target 'ros_openpose/CMakeFiles/testCameraReader.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/testCameraReader.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/christian/catkin_ws/src/ros_openpose/src/rosOpenposeSync.cpp: In function ‘void configureOpenPose(op::Wrapper&)’:
/home/christian/catkin_ws/src/ros_openpose/src/rosOpenposeSync.cpp:260:74: error: no matching function for call to ‘op::WrapperStructPose::WrapperStructPose(<brace-enclosed initializer list>)’
                                                       enableGoogleLogging};
                                                                          ^
In file included from /usr/local/include/openpose/wrapper/wrapper.hpp:13:0,
                 from /usr/local/include/openpose/wrapper/headers.hpp:6,
                 from /usr/local/include/openpose/headers.hpp:47,
                 from /home/christian/catkin_ws/src/ros_openpose/src/rosOpenposeSync.cpp:10:
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:221:9: note: candidate: op::WrapperStructPose::WrapperStructPose(op::PoseMode, const op::Point<int>&, double, const op::Point<int>&, op::ScaleMode, int, int, int, float, op::RenderMode, op::PoseModel, bool, float, float, int, const op::String&, const std::vector<op::HeatMapType>&, op::ScaleMode, bool, float, int, bool, double, const op::String&, const op::String&, float, bool)
         WrapperStructPose(
         ^~~~~~~~~~~~~~~~~
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:221:9: note:   no known conversion for argument 3 from ‘const op::Point<int>’ to ‘double’
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note: candidate: op::WrapperStructPose::WrapperStructPose(const op::WrapperStructPose&)
     struct OP_API WrapperStructPose
                   ^~~~~~~~~~~~~~~~~
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note:   candidate expects 1 argument, 26 provided
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note: candidate: op::WrapperStructPose::WrapperStructPose(op::WrapperStructPose&&)
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note:   candidate expects 1 argument, 26 provided
/home/christian/catkin_ws/src/ros_openpose/src/rosOpenposeAsync.cpp: In function ‘void configureOpenPose(op::Wrapper&, const std::shared_ptr<ros_openpose::CameraReader>&, const ros::Publisher&, const string&, bool)’:
/home/christian/catkin_ws/src/ros_openpose/src/rosOpenposeAsync.cpp:398:70: error: no matching function for call to ‘op::WrapperStructPose::WrapperStructPose(<brace-enclosed initializer list>)’
                                                   enableGoogleLogging};
                                                                      ^
In file included from /usr/local/include/openpose/wrapper/wrapper.hpp:13:0,
                 from /usr/local/include/openpose/wrapper/headers.hpp:6,
                 from /usr/local/include/openpose/headers.hpp:47,
                 from /home/christian/catkin_ws/src/ros_openpose/src/rosOpenposeAsync.cpp:16:
[... cut ...]
ros_openpose/CMakeFiles/rosOpenposeSync.dir/build.make:75: recipe for target 'ros_openpose/CMakeFiles/rosOpenposeSync.dir/src/rosOpenposeSync.cpp.o' failed
make[2]: *** [ros_openpose/CMakeFiles/rosOpenposeSync.dir/src/rosOpenposeSync.cpp.o] Error 1
CMakeFiles/Makefile2:649: recipe for target 'ros_openpose/CMakeFiles/rosOpenposeSync.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenposeSync.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

ravijo, 2022-02-09:
@ChrisTiberius 

I am sorry that you faced compilation issues. Based on point number 3 in [troubleshooting](https://github.com/ravijo/ros_openpose#troubleshooting), can you please add the following at [CMakeLists.txt#L28](https://github.com/ravijo/ros_openpose/blob/9d221f9fb571fbb0625401f16946ae8698bdb9c6/CMakeLists.txt#L28)

```
find_package(Threads REQUIRED)
```

ravijo, 2022-02-09:
@ChrisTiberius 

Please let me know if it helps!

heiqiyingzi, 2022-02-09:
Hello ravijo, I have met the same situation with OpenPose version 1.7.0. 
I have done both `git checkout tags/v1.7.0`  and added the`find_package(Threads REQUIRED)`. 
But after that the error remains :
```
error: no matching function for call to ‘op::WrapperStructPose::WrapperStructPose(<brace-enclosed initializer list>)’
```

The whole output is like this.
```
slamdog@slamdog-desktop:~/openpose_ws$ catkin_make
Base path: /home/slamdog/openpose_ws
Source space: /home/slamdog/openpose_ws/src
Build space: /home/slamdog/openpose_ws/build
Devel space: /home/slamdog/openpose_ws/devel
Install space: /home/slamdog/openpose_ws/install
####
#### Running command: "cmake /home/slamdog/openpose_ws/src -DCATKIN_DEVEL_PREFIX=/home/slamdog/openpose_ws/devel -DCMAKE_INSTALL_PREFIX=/home/slamdog/openpose_ws/install -G Unix Makefiles" in "/home/slamdog/openpose_ws/build"
####
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/slamdog/openpose_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/slamdog/openpose_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
CMake Deprecation Warning at /usr/src/googletest/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googlemock/CMakeLists.txt:41 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googletest/CMakeLists.txt:48 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
[... 112 lines cut ...]
In file included from /usr/local/include/openpose/wrapper/wrapper.hpp:13:0,
                 from /usr/local/include/openpose/wrapper/headers.hpp:6,
                 from /usr/local/include/openpose/headers.hpp:47,
                 from /home/slamdog/openpose_ws/src/ros_openpose/src/rosOpenposeSync.cpp:10:
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:221:9: note: candidate: op::WrapperStructPose::WrapperStructPose(op::PoseMode, const op::Point<int>&, double, const op::Point<int>&, op::ScaleMode, int, int, int, float, op::RenderMode, op::PoseModel, bool, float, float, int, const op::String&, const std::vector<op::HeatMapType>&, op::ScaleMode, bool, float, int, bool, double, const op::String&, const op::String&, float, bool)
         WrapperStructPose(
         ^~~~~~~~~~~~~~~~~
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:221:9: note:   no known conversion for argument 3 from ‘const op::Point<int>’ to ‘double’
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note: candidate: op::WrapperStructPose::WrapperStructPose(const op::WrapperStructPose&)
     struct OP_API WrapperStructPose
                   ^~~~~~~~~~~~~~~~~
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note:   candidate expects 1 argument, 26 provided
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note: candidate: op::WrapperStructPose::WrapperStructPose(op::WrapperStructPose&&)
/usr/local/include/openpose/wrapper/wrapperStructPose.hpp:18:19: note:   candidate expects 1 argument, 26 provided
ros_openpose/CMakeFiles/rosOpenposeSync.dir/build.make:75: recipe for target 'ros_openpose/CMakeFiles/rosOpenposeSync.dir/src/rosOpenposeSync.cpp.o' failed
make[2]: *** [ros_openpose/CMakeFiles/rosOpenposeSync.dir/src/rosOpenposeSync.cpp.o] Error 1
CMakeFiles/Makefile2:1527: recipe for target 'ros_openpose/CMakeFiles/rosOpenposeSync.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenposeSync.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1501: recipe for target 'ros_openpose/CMakeFiles/rosOpenposeAsync.dir/all' failed
make[1]: *** [ros_openpose/CMakeFiles/rosOpenposeAsync.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

ChrisTiberius, 2022-02-10:
@ravijo 

Thank you for your help.
Unfortunately the issue remains.

Any ideas?

ravijo, 2022-02-10:
@ChrisTiberius and @heiqiyingzi 

The ros_openpose 1.7.0 has been tested countless times. Anyway, today, I installed OpenPose 1.7 (CPU Mode) and then installed this repository. The compilation worked flawlessly.

I suggest you carefully follow the documentation. For example, after you checked out 1.7.0, it is mandatory to re-compile the OpenPose. After successful compilation, you must execute `sudo apt install` so that it can be seen by other projects.

Anyway, I used Ubuntu 18.04 LTS today. Below, I am sharing all the commands, I used:

```
# Preparation and installation of CMake
sudo apt install libssl-dev
sudo apt-get install qtbase5-dev
cd Downloads/cmake-3.21.5/
./configure --qt-gui
./bootstrap && make -j`nproc` && sudo make install -j`nproc`

# Preparation and installation of OpenPose
sudo apt install git
git clone https://github.com/CMU-Perceptual-Computing-Lab/openpose
cd openpose/
git checkout tags/v1.7.0
git submodule update --init --recursive --remote
sudo apt install libgoogle-glog-dev 
sudo apt-get install libgflags-dev
sudo apt-get install protobuf-compiler libprotobuf-dev
sudo apt-get install libopencv-dev
sudo apt install libboost-all-dev
sudo apt-get install libhdf5-dev 
sudo apt install  liblapacke-dev 
sudo apt install libatlas-base-dev
mkdir build/
cd build/
cmake-gui ..
make -j`nproc`
sudo make install

# Preparation and installation of ROS Melodic
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt install curl
curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
sudo apt update
sudo apt install ros-melodic-desktop-full
sudo apt install python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential
sudo apt install python-rosdep
sudo rosdep init
rosdep update
echo "source /opt/ros/melodic/setup.bash" >> ~/.bashrc
source ~/.bashrc 
mkdir -p ~/ros_ws/src
cd ros_ws/
catkin_make

# Preparation and installation of ros_openpose
cd ~/ros_ws/src
git clone https://github.com/ravijo/ros_openpose.git
cd ~/ros_ws/
catkin_make
```

Please share your environment details and other necessary information, if the issue persists.

ravijo, 2022-02-26:
@ChrisTiberius and @heiqiyingzi 

Can you please share an update on this issue? I will close it if it is solved already.

heiqiyingzi, 2022-02-26:
>For me it's solved with compiling process. It's the problem of make install. Thank you very much!

ravijo, 2022-02-26:
@heiqiyingzi 

Thanks a lot for the update. I am glad to see that you made it work. Great!

Now, I am waiting to hear from @ChrisTiberius

ChrisTiberius, 2022-03-03:
"sudo make install" did the trick for me. so i simply had not installed openpose the correct way.
@ravijo than you for your help!

closed 2022-03-03 by ravijo