ravijo/ros_openpose #59: Webcam warning Waiting for datum

robo-bug-blog, 2022-02-20:
Hi, @ravijo 
Greetings
I am using the package to run pose detection with webcam

## roslaunch command

`roslaunch ros_openpose run.launch camera:=nodepth`

I already followed the issue #33 
the images are being published on the rostopic /image_view/output as well as /usb_cam/image_raw 
so I am using the /usb_cam/image_raw changed it in the config_nodepth.launch file in the color_topic argument

but still getting this warning and the frames are not getting published on the /frame topic

## Log  
```bash
SUMMARY
========

PARAMETERS
 * /echo/pub_topic: /frame
 * /rosOpenpose/cam_info_topic: /usb_cam/camera_info
 * /rosOpenpose/color_topic: /usb_cam/image_raw
 * /rosOpenpose/depth_topic: depth_topic
 * /rosOpenpose/frame_id: no_depth
 * /rosOpenpose/no_depth: True
 * /rosOpenpose/pub_topic: /frame
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    echo (ros_openpose/echo.py)
    rosOpenpose (ros_openpose/rosOpenposeAsync)

ROS_MASTER_URI=http://127.0.0.1:11311

process[rosOpenpose-1]: started with pid [25834]
process[echo-2]: started with pid [25835]
[ INFO] [1645378321.619527110]: Starting ros_openpose...
Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.
[ WARN] [1645378321.650289606]: Waiting for color image frame...
[ WARN] [1645378321.650860940]: Waiting for datum...
[ WARN] [1645378331.660552974]: Waiting for datum...
[ WARN] [1645378341.667158112]: Waiting for datum...
[ WARN] [1645378351.670812492]: Waiting for datum...
[ WARN] [1645378361.676992992]: Waiting for datum...
[ WARN] [1645378371.684219336]: Waiting for datum...
^C[echo-2] killing on exit
[rosOpenpose-1] killing on exit
[rosOpenpose-1] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
```

can you please help me?
Thank you :heart:

robo-bug-blog, 2022-02-21:
Never mind

It just worked fine. I guess, it takes some time to start

Thank you, its a great package :heart:

closed 2022-02-21 by robo-bug-blog

ravijo, 2022-02-26:
Glad to see your message, @robo-bug-blog