ravijo/ros_openpose #60: None of keypoints detected from my own data

akamob, 2022-03-08:
Hi, Thank you for the awesome work.

I have a VLP-16 lidar and I want to use it for activity recognition, I convert my point clouds into RGB images and record as a video. First, I use [this](https://github.com/ros-drivers/video_stream_opencv) package with command: `roslaunch video_stream_opencv video_file.launch` to read my video and publish **/videofile/image_raw** topic. Then I use your package by running: `roslaunch ros_openpose run.launch camera:=nodepth`. But none of keypoints were detected such as this video shown: [Links](1GruwepwFfngTKTFJ-S_0Vh1A2jCY39gf)

I made some modifies before using your code:
1. In **run.launch**, I added `--net_resolution -1x128` in **openpose_args**.
2. In **config_nodepth.launch**, I changed **color_topic** to my topic:` /videofile/image_raw`.
In addition, after referring to [this](https://github.com/ravijo/ros_openpose/issues/7) issue, I changed **auto depthPtr** to `cv_bridge::toCvCopy(depthMsg, sensor_msgs::image_encodings::BGR8)`, but I still got same result (none of keypoints).

However, when I use [openpose](https://github.com/CMU-Perceptual-Computing-Lab/openpose) **01_body_from_image.py** to test a single image, it can detect keypoints:
[image]

May i have your suggestions?
Any help is much appreciated:)

ravijo, 2022-03-08:
@akamob 

Thanks for opening this issue. It is quite interesting at first. Because at the backend, the same OpenPose is used in both cases.. Basically, the OpenPose when used with this package is detecting 0 people. The image quality is poor but the python binding of OpenPose is detecting 1 person. 

I suggest you do a little more debugging to find the actual cause of this confusing behavior. 

1. Please use the video (mp4) file on the complied OpenPose to see if Openpose detects anything. Please refer to[ Quick Start Overview](https://github.com/CMU-Perceptual-Computing-Lab/openpose#quick-start-overview)
```
./build/examples/openpose/openpose.bin --video examples/media/video.avi \
--face --hand --write_json output_json_folder/
```

2. We have seen the python bindings are working fine, can you please use the python bindings with this package by enabling synchronous mode as mentioned in [Implementation Versions Info.](https://github.com/ravijo/ros_openpose#implementation-versions-info)
```
roslaunch ros_openpose run.launch camera:=nodepth synchronous:=true \
py_openpose_path:=absolute_path_to_py_openpose
```
3. Can you please confirm that the test image for `01_body_from_image.py` is taken from the recorded video (and has exactly the same resolution as the video)?
4. Can you please confirm that you have added "--net_resolution -1x128" to your `01_body_from_image.py` file?

akamob, 2022-03-09:
Hi, @ravijo, thank you very much for your reply:)

I forgot to mention I build OpenPose in **cpu_only** mode due to the limited computational power. And the following are my test results:

1. I test my video file on OpenPose and it can detect some keypoints. Although the skeleton is incomplete and sometimes OpenPose still detect 0 people.
[image]

[image]

2. I'm not sure if I'm doing it right, **py_openpose_path** means to access the absolute path shown in the following picture?
[image]
I run this command:
`roslaunch ros_openpose run.launch camera:=nodepth synchronous:=true \ py_openpose_path:=**/home/lk3696/yi_ros_openpose/src/openpose/build/python/openpose/**`

And I got the warn:
```
[ WARN] [1646801737.361694890]: Shutdown request received.
[ WARN] [1646801737.361725319]: Reason given for shutdown: [[/rosOpenpose] Reason: new node registered with same name]
rosOpenposeSync: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const sensor_msgs::CameraInfo_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = const sensor_msgs::CameraInfo_<std::allocator<void> >*]: Assertion `px != 0' failed.
shutdown request: [/echo] Reason: new node registered with same name
```

3. Yes, my test image come from the recorded video and they have same resolution.
[image]

4. I'm not add `--net_resolution -1x128` in `01_body_from_image.py` file.
[image]

ravijo, 2022-03-09:
@akamob 

Seeing your screenshot, I am quite sure that the images you are getting are of poor quality. Because OpenPose can detect it SOMETIMES ONLY.

You can either (a) try to improve the image quality, or (b) play with the OpenPose configurations such as `net_resolution` etc. Basically, we do not want to resize down images in this scenario.

Please let me know your progress in this specific case.

ravijo, 2022-03-27:
Closing this issue as no further comments were received. Feel free to comment if the issue still exists.

closed 2022-03-27 by ravijo