ravijo/ros_openpose #62: problem with run.launch

niccovezzi, 2022-05-09:
hi
i have a problem when i try to launch run.launch.
i have a openpose with only cpu, the file test openpose.bin work for me (very very slowly, if i use --net_resolution -1x32 run with 4 fps).
but when i launch the file run.launch, it gives me error:
roslaunch ros_openpose run.launch 
... logging to /home/nicco/.ros/log/ccdfaedc-cfa5-11ec-9b59-9cfce8af62c0/roslaunch-nicco-VivoBook-ASUSLaptop-X509JA-X509JA-18553.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nicco-VivoBook-ASUSLaptop-X509JA-X509JA:44959/

SUMMARY
========

PARAMETERS
 * /echo/pub_topic: /frame
 * /rosOpenpose/cam_info_topic: cam_info_topic
 * /rosOpenpose/color_topic: /camera/image_raw
 * /rosOpenpose/depth_topic: depth_topic
 * /rosOpenpose/frame_id: no_depth
 * /rosOpenpose/no_depth: True
 * /rosOpenpose/pub_topic: /frame
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    echo (ros_openpose/echo.py)
    rosOpenpose (ros_openpose/rosOpenposeAsync)

ROS_MASTER_URI=http://localhost:11311

process[rosOpenpose-1]: started with pid [18571]
process[echo-2]: started with pid [18572]
[ INFO] [1652107174.632115228]: Starting ros_openpose...
---------------------------------- WARNING ----------------------------------
We have introduced an additional boost in accuracy in the CUDA version of about 0.2% with respect to the CPU/OpenCL versions. We will not port this to CPU given the considerable slow down in speed it would add to it. Nevertheless, this accuracy boost is almost insignificant so the CPU/OpenCL versions can be safely used.
-------------------------------- END WARNING --------------------------------
================================================================================
REQUIRED process [rosOpenpose-1] has died!
process has died [pid 18571, exit code -11, cmd /home/nicco/Desktop/ros_ws/devel/lib/ros_openpose/rosOpenposeAsync --model_folder /home/nicco/openpose/models/ --net_resolution -1x32 --model_pose COCO __name:=rosOpenpose __log:=/home/nicco/.ros/log/ccdfaedc-cfa5-11ec-9b59-9cfce8af62c0/rosOpenpose-1.log].
log file: /home/nicco/.ros/log/ccdfaedc-cfa5-11ec-9b59-9cfce8af62c0/rosOpenpose-1*.log
Initiating shutdown!
================================================================================
[echo-2] killing on exit
[rosOpenpose-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Any suggestion?

ravijo, 2022-05-10:
@niccovezzi 

I am sorry that you couldn't make it work! It is very rare to see that the rosOpenpose process dies. Unfortunately, the logs shared by you are not enough to debug the issue.

1. Did you execute `sudo make install` inside OpenPose to install it system-wide?
2. Are you sure that the package complication went without any errors?
3. Can you run the code inside a debugger for better understanding?

niccovezzi, 2022-05-10:
@ravijo 
yes, i execute sudo make install.
i run gdb for debug rosOpenposeAsync and this is the output:

Starting program: /home/nicco/Desktop/ros_ws/devel/lib/ros_openpose/rosOpenposeAsync 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffc42ec700 (LWP 14682)]
[New Thread 0x7fffc3aeb700 (LWP 14683)]
[New Thread 0x7fffc32ea700 (LWP 14684)]
[New Thread 0x7fffc2ae9700 (LWP 14685)]
[FATAL] [1652186829.908395134]: Missing 'pub_topic' info in launch file. Please make sure that you have executed 'run.launch' file.
[Thread 0x7fffc2ae9700 (LWP 14685) exited]
[Thread 0x7fffc32ea700 (LWP 14684) exited]
OpenCV Error: Assertion failed (tlsSlots.size() > slotIdx) in releaseSlot, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/system.cpp, line 1092
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/system.cpp:1092: error: (-215) tlsSlots.size() > slotIdx in function releaseSlot

Thread 1 "rosOpenposeAsyn" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.

Maybe the problem is about the version of OpenCV?
(p.s. the pub_topic missed because was not running run.launch)

ravijo, 2022-05-11:
@niccovezzi 

> Maybe the problem is about the version of OpenCV?

Yes. The problem is reported [here](https://answers.ros.org/question/352756/).

Please make sure to use the same OpenCV in OpenPose and ROS.

niccovezzi, 2022-05-11:
@ravijo 

Perfect, i modify a .cmake file of cv_bridge and now it works. 
Thank you very much for yours help ravijo.

ravijo, 2022-05-11:
@niccovezzi 

I am glad it worked!

Can you please mention in detail what you did? It will be helpful for users if they get the same error.

niccovezzi, 2022-05-11:
@ravijo 

Sure. I follow this solution:

I had a similar problem, and solved it by modifying cv_bridgeConfig.cmake file.

cd /opt/ros/melodic/share/cv_bridge/cmake
sudo gedit cv_bridgeConfig.cmake

Replaced line 96
set(_include_dirs "include;/usr/include;/usr/include/opencv2")

and line 119#
set(libraries "cv_bridge;/usr/local/lib/libopencv_core.so.4.5.4;/usr/local/lib/libopencv_imgproc.so.4.5.4;/usr/local/lib/libopencv_imgcodecs.so.4.5.4")

Need only to take care about OpenCV version (in my case 4.5.5). After that i return to my ros_ws, i do catkin_make and that's it.

Thanks again ravijo!

ravijo, 2022-05-11:
@niccovezzi 

Thank you very much. I formatted your response. I am closing this issue now.

goncalofurtado1, 2022-06-20:
Hi @ravijo, I am facing a similar issue.

I had to change the **run.launch** file slightly because my **realsense2_camera** is in ROS2 so the file now just waits for the topics to exist and doesnt launch the realsense2_camera since I launch it in ROS2.

What is happening is that whenever I launch my version of the run.launch it shows the visualizer as it is suposed to and as soon as I point the camera to me it shows me this error which seems to be the same as published by @niccovezzi.

```
... logging to /home/goncalo/.ros/log/dbd3965a-f094-11ec-8bb4-b3552ce940b9/roslaunch-goncalo-GL552VW-22372.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://goncalo-GL552VW:42069/

SUMMARY
========

PARAMETERS
 * /echo/pub_topic: /frame
 * /rosOpenpose/cam_info_topic: /camera/color/cam...
 * /rosOpenpose/color_topic: /camera/color/ima...
 * /rosOpenpose/depth_topic: /camera/aligned_d...
 * /rosOpenpose/frame_id: camera_color_opti...
 * /rosOpenpose/no_depth: False
 * /rosOpenpose/pub_topic: /frame
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /visualizer/frame_id: camera_color_opti...
 * /visualizer/id_text_offset: -0.05
 * /visualizer/id_text_size: 0.2
 * /visualizer/pub_topic: /frame
 * /visualizer/skeleton_hands: False
 * /visualizer/skeleton_line_width: 0.01

NODES
  /
    echo (ros_openpose/echo.py)
    rosOpenpose (ros_openpose/rosOpenposeAsync)
    visualizer (ros_openpose/visualizer.py)

auto-starting new master
process[master]: started with pid [22387]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dbd3965a-f094-11ec-8bb4-b3552ce940b9
process[rosout-1]: started with pid [22405]
started core service [/rosout]
process[rosOpenpose-2]: started with pid [22412]
process[visualizer-3]: started with pid [22413]
process[echo-4]: started with pid [22414]
[ INFO] [1655728268.348597539]: Starting ros_openpose...
Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.
[ WARN] [1655728268.382687147]: Waiting for color image frame...
[ WARN] [1655728268.382963156]: Waiting for datum...
[INFO] [1655728269.581108]: []

[INFO] [1655728269.693738]: []

[INFO] [1655728269.806744]: []

[INFO] [1655728269.929786]: []

================================================================================REQUIRED process [rosOpenpose-2] has died!
process has died [pid 22412, exit code -11, cmd /home/goncalo/catkin_ws/devel/lib/ros_openpose/rosOpenposeAsync --model_folder /home/goncalo/openpose/models/ --net_resolution -1x192 __name:=rosOpenpose __log:=/home/goncalo/.ros/log/dbd3965a-f094-11ec-8bb4-b3552ce940b9/rosOpenpose-2.log].
log file: /home/goncalo/.ros/log/dbd3965a-f094-11ec-8bb4-b3552ce940b9/rosOpenpose-2*.log
Initiating shutdown!
================================================================================
[echo-4] killing on exit
[visualizer-3] killing on exit
[rosOpenpose-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I don't believe it comes from a bad configuration of the launch file since the only thing I did was to remove the launch of the realsense2_camera and I also dont believe it comes from the **cv_bridgeConfig** configuration as mentioned in issue #14 since I already tested it by following the steps mentioned in the issue. As I mentioned, this error happens as soon as it tries to compute the human pose. 

When I run OpenPose with the same resolution I input here (-1x192) it is slow but works correctly. So no problem there. I had to uncheck **USE_CUDNN** because OpenPose was not working with it but it works fine without it.

Also when I try to debug like this:

    g++ -g src/ros_openpose/src/rosOpenposeAsync.cpp -o rosOpenposeAsync
    
It shows me this error:

```
 src/ros_openpose/src/rosOpenposeAsync.cpp:19:10: fatal error: ros_openpose/Frame.h: No such file or directory
 #include <ros_openpose/Frame.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Is it possible the whole reason it doesn't run is because it cant find the Frame.h file?

What are your suggestions?

I hope I made my problem understandable, Thank You for any answer!

ravijo, 2022-06-21:
@goncalofurtado1 

Can you please share necessary information such as the version of OpenPose, OpenCV, ROS, Ubuntu, etc?

You said that you are using ROS2 for realsense camera. Where is your ros_openpose running? 

ros_openpose currently designed only for ROS 1. So If you want to integrate it with ROS 2, please check out [ros1_bridge](https://github.com/ros2/ros1_bridge) package.

The logs shared by you shows that ros_openpose somehow could not receive a frame from the camera. It could be related to OpenCV. For example, the OpenCV should be installed by ROS (apt install blah blah) automatically. But if you are using a custom OpenCV, then you need to build ROS on that OpenCV.

I am opening this issues, for now.

goncalofurtado1, 2022-06-21:
Thank You for your answer!

Looks like the issue really could be related to the OpenCV after all. I am having other problems with other packages which don't allow me to test this issue right know but I'll give you a more concrete answer as soon as they are resolved.

If you find it useful I'll leave it here the corresponding versions:
- Ubuntu: **20.04**
- ROS: **noetic and foxy**
- Note: yes **ros_openpose** is installed in noetic and I launch the **realsense2_camera** node in foxy and then I run **ros1_bridge** which passes the topics correctly to **ros_openpose** and I can see the output in the **visualizer**.
- OpenPose: **1.7.0**
- OpenCV: I installed **4.2.0** but whenever I ran `python3 -c "import cv2; print(cv2.__version__)"` the output was **4.5.5** that's why the problem should be a mismatch between the versions. I uninstalled all versions and reinstalled 4.2.0 and made sure the command now returned **4.2.0**

ravijo, 2022-06-23:
@goncalofurtado1 

Sorry for the trouble. OpenPose 1.7.0 on Ubuntu 20.04 using ROS Noetic should work without issues. I remember checking it (compilation only) [here](https://github.com/ravijo/ros_openpose/issues/61#issuecomment-1086774037).

I suspect some configuration or dependencies mismatch issue on your side, though.

How about trying a fresh install to confirm my guess?

goncalofurtado1, 2022-06-23:
Ok I'll reinstall **OpenPose** and **ros_openpose** again and report back once it's done.

Thank You for your suggestion.

goncalofurtado1, 2022-06-24:
@ravijo 

I reinstalled both packages and remain with the same issue but I believe your are right about having some sort of bad configuration on my machine.

Do you have any suggestion on best practices to debug this issue?

ravijo, 2022-06-25:
@goncalofurtado1 

Thanks for the update.

> Do you have any suggestion on best practices to debug this issue?

Basically, in this situation, it is better to check the followings things:
1. Make sure that OpenPose is installed correctly by running the openpose.bin demo file.
2. Make sure that ROS is appropriately installed.
   * Specifically, ros_openpose depends on ROS to get images. Internally, ROS is using OpenCV by making use of cv_bridge. Therefore, the cv_bridge must be compatible with ROS and OpenCV too.
   * When we install ROS from binaries, the compatible cv_bridge and OpenCV are installed automatically.
   * Unfortunately, when using a custom OpenCV, the ROS must be built from the source!
3. Ensure the camera driver and corresponding ROS package (say Intel RealSense) are well installed.
   * To verify this, run the ROS package of your camera and use RViz to visualize input images from the camera.

If you feel this is too much to do, go for a fresh install of Ubuntu, then things should work smoothly.

goncalofurtado1, 2022-06-25:
@ravijo 

I did all the previous checks and everything is alright but I discovered something that is quite embarrassing... I forgot to set the **aligned_depth** option of the Intel RealSense camera to **True**. I think this might be the root of the problem.

As soon as I can I'll test this hypotheses and report back.

ravijo, 2022-06-26:
@goncalofurtado1 

Yeah, please make sure that depth is aligned to the color frame. This is why ros_openpose provides config_realsense.launch as shown below:

https://github.com/ravijo/ros_openpose/blob/3e34c4810a3d34d975f7758dad45fa907958b8b1/launch/config_realsense.launch#L12-L13

Furthermore, please check the depth encoding. If the depth is in millimeters, we must convert it to meters. See below how it is implemented:
https://github.com/ravijo/ros_openpose/blob/3e34c4810a3d34d975f7758dad45fa907958b8b1/scripts/ros_openpose_synchronous.py#L55

https://github.com/ravijo/ros_openpose/blob/3e34c4810a3d34d975f7758dad45fa907958b8b1/src/cameraReader.cpp#L80-L85

goncalofurtado1, 2022-06-28:
@ravijo 

It's solved! Thank you for your help.

In fact it was because of a mismatch between two versions of OpenCV but towards the end I just forgot to activate the **aligned_depth** feature.

ravijo, 2022-06-28:
@goncalofurtado1 

Thanks for the update. Glad it worked!!!

closed 2022-06-28 by ravijo