ravijo/ros_openpose #66: Increasing frame rate for Azure Kinect

yepw, 2022-06-26:
Hi, I was trying to make this pose detection node run at a higher rate (~30Hz). I successfully increased the image frame rate of the Azure Kinect to 30Hz, but the rate of `human_skeleton` is only about 10Hz.
I tried both the C++ and Python sync nodes and tried to tune the parameters of ApproximateTimeSynchronizer
https://github.com/ravijo/ros_openpose/blob/3e34c4810a3d34d975f7758dad45fa907958b8b1/scripts/ros_openpose_synchronous.py#L67
but none of them works. Any ideas are appreciated. 

### How to reproduce 
To increase the frame rate of RGB and depth images, add the following args when running `$(find azure_kinect_ros_driver)/launch/driver.launch`
https://github.com/ravijo/ros_openpose/blob/3e34c4810a3d34d975f7758dad45fa907958b8b1/launch/config_azurekinect.launch#L60
```
<arg name="depth_mode" value="WFOV_2X2BINNED"/>
<arg name="fps" value="30"/>
```
After this, `/rgb/image_raw` and ` /depth_to_rgb/image_raw` are published at about 30Hz.
However, the rate of `human_skeleton` is only about 10Hz.

ravijo, 2022-06-26:
@yepw 

Thanks for properly mentioning all the details. In the past, while using the Azure Kinect, I noticed that the image resolution was much more than what Intel RealSense was providing (probably 4K).

I guess that OpenPose is taking a significant time to consume an image frame (due to an enormous resolution). Therefore, I suggest you run the openpose.bin on a video captured by Azure Kinect and monitor the FPS.

I preassume that your OpenPose is compiled with CUDA. In any case, please check OpenPose documentation to speed up the inference process.

Please feel free to share the status of this task.

yepw, 2022-06-26:
Thanks, @ravijo 
It seems the bottleneck is within OpenPose. I am using a 1080Ti and get ~10 FPS. [Others](https://github.com/CMU-Perceptual-Computing-Lab/openpose/issues/1665) can get ~12 FPS with a 2070.
To get a higher FPS, one way is to change the `net_resolution` flag. 
https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/973b9139c042b5228bed42a0e0867f342b571f54/include/openpose/flags.hpp#L99
I am able to get ~30 FPS after setting `net_resolution` to `-1x176`.
It's not ideal since it reduces the output resolution. If anyone find a better solution, please let me know.

closed 2022-06-26 by yepw

ravijo, 2022-06-27:
@yepw 

Glad to hear your response.

Getting 10 FPS from 1080Ti is good enough. Furthermore, you correctly found that we can lower the `net_resolution` to get a higher FPS.