ravijo/ros_openpose #71: Frame_id + Publishing depth information

aabsz, 2022-08-09:
Dear Rajivo,

Many thanks for your fantastic code, which I found really helpful.
I am running the code with an Xtion RGB-D camera. It is almost working correctly. However, I'm a bit unsure whether I have done everything wholly correct or not. My main issue is with "frame_id," which should be modified in the launch.config file. What is it exactly? Do you know what it is for Xtion? I'm using the depth topic that the Xtion publishes as the "frame_id", and getting an almost fine skeleton in Rviz and user's body parts detected using opnepose on RGB image. However, the output of Rviz is not perfect, and I think it might be due to not using a correct "frame_id".

I also have another question. I am only getting joints' 2D positions as a stream of data. How can I get the depth data of each joint along with its 2D position too?

ravijo, 2022-08-10:
@aabsz 

Thank you very much for your warm message. I appreciate it. I think the credit goes to [our wonderful contributors](https://github.com/ravijo/ros_openpose#thanks).

> ... getting an almost fine skeleton in Rviz ... the output of Rviz is not perfect,...

I would suggest sharing a screenshot of your RViz, please.

> I think it might be due to not using a correct "frame_id".

We have covered this topic in [the FAQ](https://github.com/ravijo/ros_openpose#faq). May I request to look at the same, please? Furthermore, check out the existing configs for various cameras inside the [launch](https://github.com/ravijo/ros_openpose/tree/master/launch) subdirectory. Nevertheless, please feel free to write back if it does not answer your question. I would love to improve the documentation based on your feedback.

>  How can I get the depth data of each joint along with its 2D position too?

I am sorry, do you mean each joint's 3D coordinate (x, y, and z values)?

On a side note, once you finish this task, may I ask you to share your `config_xtion.launch`? This valuable information will be made available to everyone.

aabsz, 2022-08-11:
Dear Rajivo,

Thank you for the prompt reply. 
I'll share the RVIZ screen ASAP. 

I had already checked the FAQ and all config.launch files of other sensors, but I still don't know what it is exactly for Xtion. I guess I should search more about it. 

Regarding the 3D positions, yes, exactly. I need access to each joint's 3D position data and feed these data as input to a DL model. Currently, the 2D positions are published, but I also need access to the depth information. Has this already been included in the code, or do I have to do it myself? Do you have any suggestions on how to do it? 

I just made some minor changes to the config.launch code to make it compatible with Xtion. However, I would be happy to share the final code with you so that everyone can use it :-)

ravijo, 2022-08-13:
Thanks @aabsz for the update

> Regarding the 3D positions, yes, exactly. 

This is available already. Right now, in the terminal, you can see the pixels only just to keep the terminal cleaner. Please see the current implementation below:

https://github.com/ravijo/ros_openpose/blob/56dfd62f16ca9bae53263b614c6b196a94fe0e75/scripts/echo.py#L14-L15

>  I need access to each joint's 3D position data and feed these data as input to a DL

You should subscribe to the "/frame" topic to receive the 3D position data. 

> I just made some minor changes to the config.launch code to make it compatible with Xtion. However, I would be happy to share the final code with you so that everyone can use it :-)

Please share necessary files and documentation. If you are not sure about GitHub, share via email, please. My email is mentioned [here](https://github.com/ravijo/ros_openpose/blob/master/package.xml).

aabsz, 2022-08-27:
Dear ravijo,

Sorry for the long delay. I figured out everything regarding the frame_id for the Xtion RGB-D camera, which is now solved. I will share the corresponding Config file for this camera.
Regarding the depth information of each joint, I can not get the depth information from the depth sensor yet. Is it possible to guide me more on how to get this data? 
I'm getting each joint's pixel information on the terminal. I want to get the depth information too.

Many thanks

ravijo, 2022-08-27:
@aabsz 

> I can not get the depth information from the depth sensor yet

I think you mean ~depth sensor~ ros_openpose.

> I want to get the depth information too.

>> You should subscribe to the "/frame" topic to receive the 3D position data.

I suggested you to get the 3D position data from "/frame" topic. Just execute the following in terminal:

```console
rostopic echo /frame
``` 
Or if you want to do it via code, change [this line](https://github.com/ravijo/ros_openpose/blob/56dfd62f16ca9bae53263b614c6b196a94fe0e75/scripts/echo.py#L14) as following

```python
text = [bodyPart.point for person in msg.persons for bodyPart in person.bodyParts]
```

ravijo, 2022-09-11:
Closing this issue because of no further comments. However, feel free to comment if the problem still exists.

closed 2022-09-11 by ravijo