ravijo/ros_openpose #75: Hand skeleton error

igurucr, 2022-11-10:
We are unable to get the skeleton of hands. Whenever we try to do it, it's throwing an error. 
[image]
@ravijo

ravijo, 2022-11-11:
@igurucr 

Thank you very much for creating this issue. Please look at the following snippet:

https://github.com/ravijo/ros_openpose/blob/afecab60b39ea74a0673088589a377df49c5526e/launch/run.launch#L27-L28

The argument `skeleton_hands` enables visualization of the hands of the skeleton. To enable hand tracking, you need to pass `--hand` as OpenPose command line argument. May I request to check [Configuration#2](https://github.com/ravijo/ros_openpose#configuration)?

Please let me know if the issue persists.

Cheers

igurucr, 2022-11-13:
@ravijo 

 Thank you for the response. It's working for the hand detection now, but the output seems to be a bit different as its giving negative values. Can you please elaborate regarding the output? 
[image]

ravijo, 2022-11-14:
@igurucr 

> its giving negative values.

It is difficult to say if this is a problem. These values depend on the "Fixed Frame" inside RViz. Please check the frame once and let me know if it makes sense.

ravijo, 2023-02-03:
Closing this issue as no further comments were received. Feel free to comment if the problem still exists.

closed 2023-02-03 by ravijo