ravijo/ros_openpose #79: Directory for openpose python node

mobeeniqbal101, 2022-11-24:
Hi, I looked into the build directory as suggested by you under the issue : https://github.com/ravijo/ros_openpose/issues/54

But I cannot find the required files, can you please guide further ? Because I even tried to build open pose configuration again with Build python Checked. Screenshot below for reference:

[image]

mobeeniqbal101, 2022-11-25:
In the openpose/build/python folder I can see the following file pyopenpose.cpython-36m-x86_64-linux-gnu.so  but importing in python as suggested by you doesnt work.

[image]

@ravijo  If you could please guide further, it would be a great help.

ravijo, 2022-11-25:
@mobeeniqbal101 

Sorry for the delay in my response. I am on vacation. So I can not work on it. I apologize for the inconvenience. 

Unfortunately, this issue is related to OpenPose. You should check their documentation or create an issue on their GitHub page.

Thanks for your kind understanding.

mobeeniqbal101, 2022-11-29:
Hi @ravijo , I already created an issue on their page, but it seems they are no longer responding to issues, if you could help us it would be great.

ravijo, 2022-11-30:
@mobeeniqbal101 mobeeniqbal101

As I am on vacation, I can't work on it. I will reply with proper answer later on!

I appreciate your kind cooperation.

mobeeniqbal101, 2022-12-08:
Hi, I figured it out, I was missing some advanced checkmarks while building openpose via cmake-gui as mentioned below:

https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/03_python_api.md#compatibility

which I suppose are necessary for the python library.

closed 2022-12-08 by mobeeniqbal101