ravijo/ros_openpose #80: encoding MONO16 support

furuya-kenji, 2022-11-29:
Dear Rajivo,

I use ros_openpose with RGBcamera and ToF camera. 
The encoding of depth data from ToF camera is MONO16.
When doing [rostopic echo /frame] , z data unit is not adequate, compared to RealSense.

So, I want to change a part of source code as below.
------------------
src/cameraReader.cpp (line 82)
before:
if (depthMsg->encoding == sensor_msgs::image_encodings::TYPE_16UC1)

after:
if (depthMsg->encoding == sensor_msgs::image_encodings::TYPE_16UC1 || depthMsg->encoding == sensor_msgs::image_encodings::MONO16)
------------------

Is this right and valid?
Thank you and regards.

ravijo, 2022-11-30:
@furuya-kenji 

In the case of '16UC1' encoding, the depth values are represented in millimeters. Therefore a conversion from millimeters to meters is performed by multiplying every element of the depth image by 0.001.

> after:
> 
> ```
> if (depthMsg->encoding == sensor_msgs::image_encodings::TYPE_16UC1 || depthMsg->encoding == sensor_msgs::image_encodings::MONO16)
> ```
> 
> Is this right and valid? Thank you and regards.

This seems correct. However, I can not test it as of now. Please compile it and make sure to execute CPP node. The repository contains python node as well.

Let me know, how it goes!

Cheers

furuya-kenji, 2022-12-01:
@ravijo 

Thank you for replying.
The modified code works well.(point z values show in meters)

If you don't mind, is it possible to merge the source code?
Thanks and regards.

[image]

ravijo, 2022-12-01:
@furuya-kenji 

Thank you very much for the update. I am glad it works!

Yes, please create a pull request. Thank you for the great contribution!

furuya-kenji, 2022-12-01:
@ravijo 
Thank you for replying.
I created pull request below. 
https://github.com/ravijo/ros_openpose/pull/81

if any problem, please leave a comment.
Thanks and regards.

ravijo, 2022-12-01:
@furuya-kenji 

:100: done

Thanks a lot

closed 2022-12-01 by ravijo