ravijo/ros_openpose #83: Error of run.launch

osumilab, 2022-12-07:
# Summary
I introduced ros_openpose based on README and ran the run.launch, but it failed.

# Versions

- ROS Driver versions: noetic
- openpose versions: 1.7
- opencv versions: 4.2.0
- Camera to be used: Realsense D435i

# Issue details
It gives me error. Sorry for the length of message.
```
System-Product-Name:~/catkin_ws$ roslaunch ros_openpose run.launch
... logging to /home/osumi/.ros/log/bc743e38-7477-11ed-8877-813910df0dba/roslaunch-osumi-System-Product-Name-1426961.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.91.72.99:34445/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
[... 124 lines cut ...]
    import shutil
  File "/usr/lib/python3.8/shutil.py", line 29, in <module>
    import lzma
  File "/usr/lib/python3.8/lzma.py", line 27, in <module>
    from _bz2 import BZ2Compressor, BZ2Decompressor
KeyboardInterrupt
    from _lzma import *
KeyboardInterrupt
[ INFO] [1670398898.257948628]:  
 07/12 16:41:41,822 WARNING [140183102027520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
shutting down processing monitor...
... shutting down processing monitor complete
done

```
# Supplement
I saw several issues of situation similar to me, and tried the following methods listed there, but they did not the problem.
1. Execute ```sudo make install``` inside OpenPose 
2. Confirm to match opencv version in ROS and openpose. 
```print(cv2__version__)``` that output was 4.2.0, ```dpkg -l | grep libopencv``` that output was 4.2.0, Version of line 119 in ```cv_bridgeConfig.cmake``` is 4.2.0.
3. Confirm that there is ```openpose/build/caffe/lib/libcaffe.so``` and write ```set(CAFFE_LIB_FOLDER /home/openpose/build/caffe/lib)``` 
```link_directories(${CAFFE_LIB_FOLDER})``` in ``` CmakeLists.txt``` 

Also, regarding the configuration of launch file in README, should I describe ```<arg name="openpose_args" value="--model_folder /home/ravi/openpose/models/"/>``` or  ```<arg name="openpose_args" value="--face --hand"/>```?
I cannot describe both because they have the same```arg name```

ravijo, 2022-12-07:
@osumilab 

It seems that the problem is with OpenPose installation. Please see the error message below:

```
/home/osumi/catkin_ws/devel/lib/ros_openpose/rosOpenposeAsync: error while loading shared libraries: libcaffe.so.1.0.0: cannot open shared object file: No such file or directory
```

Can you please run `openpose.bin` file with sample video as suggested in the documentation of [OpenPose](https://github.com/CMU-Perceptual-Computing-Lab/openpose#quick-start-overview)

osumilab, 2022-12-08:
I ran ``` ~/openpose$ ./build/examples/openpose/openpose.bin ```, image was displayed.
And, I ran ```~/catkin_ws$ roslaunch ros_openpose run.launch``` in the state, but it gives me error like this and image changed [Camera disconnected, reconnecting...]
```
process[rosOpenpose-1]: started with pid [2180435]
/home/osumi/catkin_ws/devel/lib/ros_openpose/rosOpenposeAsync: error while loading shared libraries: libcaffe.so.1.0.0: cannot open shared object file: No such file or directory
process[visualizer-2]: started with pid [2180436]
process[echo-3]: started with pid [2180438]
process[camera/realsense2_camera_manager-4]: started with pid [2180439]
process[camera/realsense2_camera-5]: started with pid [2180440]
process[rviz-6]: started with pid [2180441]
================================================================================REQUIRED process [rosOpenpose-1] has died!
process has died [pid 2180435, exit code 127, cmd /home/osumi/catkin_ws/devel/lib/ros_openpose/rosOpenposeAsync --model_folder home/osumi/openpose/models/ __name:=rosOpenpose __log:=/home/osumi/.ros/log/bc743e38-7477-11ed-8877-813910df0dba/rosOpenpose-1.log].
log file: /home/osumi/.ros/log/bc743e38-7477-11ed-8877-813910df0dba/rosOpenpose-1*.log
Initiating shutdown!
================================================================================
[ INFO] [1670472897.351375390]: Initializing nodelet with 16 worker threads.
[ INFO] [1670472897.428212157]: RealSense ROS v2.3.2
[ INFO] [1670472897.428240739]: Built with LibRealSense v2.50.0
[ INFO] [1670472897.428253098]: Running with LibRealSense v2.50.0
[rviz-6] killing on exit
[camera/realsense2_camera-5] killing on exit
[camera/realsense2_camera_manager-4] killing on exit
[echo-3] killing on exit
[visualizer-2] killing on exit
[rosOpenpose-1] killing on exit
Traceback (most recent call last):
  File "/home/osumi/catkin_ws/src/ros_openpose-master/scripts/echo.py", line 9, in <module>
    import rospy
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/__init__.py", line 49, in <module>
    from .client import spin, myargv, init_node, \
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 52, in <module>
Traceback (most recent call last):
  File "/home/osumi/catkin_ws/src/ros_openpose-master/scripts/visualizer.py", line 10, in <module>
    import roslib
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/__init__.py", line 50, in <module>
    import rospy
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/__init__.py", line 49, in <module>
    from roslib.launcher import load_manifest  # noqa: F401
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/launcher.py", line 42, in <module>
    from .client import spin, myargv, init_node, \
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 52, in <module>
    import rospkg
  File "/usr/lib/python3/dist-packages/rospkg/__init__.py", line 41, in <module>
    import roslib
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest  # noqa: F401
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/launcher.py", line 42, in <module>
    from .manifest import InvalidManifest, Manifest, parse_manifest_file
  File "/usr/lib/python3/dist-packages/rospkg/manifest.py", line 40, in <module>
    import rospkg
  File "/usr/lib/python3/dist-packages/rospkg/__init__.py", line 42, in <module>
    import xml.dom.minidom as dom
  File "/usr/lib/python3.8/xml/dom/minidom.py", line 23, in <module>
    from .rospack import expand_to_packages, get_package_name, \
  File "/usr/lib/python3/dist-packages/rospkg/rospack.py", line 37, in <module>
    from xml.dom.xmlbuilder import DOMImplementationLS, DocumentLS
  File "/usr/lib/python3.8/xml/dom/xmlbuilder.py", line 3, in <module>
    from xml.etree.cElementTree import ElementTree
  File "/usr/lib/python3.8/xml/etree/cElementTree.py", line 3, in <module>
    import copy
    from xml.etree.ElementTree import *
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 844, in exec_module
  File "<frozen importlib._bootstrap>", line 914, in _find_spec
  File "<frozen importlib._bootstrap_external>", line 976, in get_code
  File "<frozen importlib._bootstrap_external>", line 1407, in find_spec
[... 17 lines cut ...]
Configuring OpenPose...
Starting thread(s)...
Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.
F1208 13:25:41.268456 2188158 cudnn_conv_layer.cpp:53] Check failed: status == CUDNN_STATUS_SUCCESS (4 vs. 0)  CUDNN_STATUS_INTERNAL_ERROR
*** Check failure stack trace: ***
    @     0x7f8f6b0581c3  google::LogMessage::Fail()
    @     0x7f8f6b05d25b  google::LogMessage::SendToLog()
    @     0x7f8f6b057ebf  google::LogMessage::Flush()
    @     0x7f8f6b0586ef  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f8f6ac7f0f5  caffe::CuDNNConvolutionLayer<>::LayerSetUp()
    @     0x7f8f6ad63e4d  caffe::Net<>::Init()
    @     0x7f8f6ad66365  caffe::Net<>::Net()
    @     0x7f8f6b73559c  op::NetCaffe::initializationOnThread()
    @     0x7f8f6b6e0ce3  op::FaceExtractorCaffe::netInitializationOnThread()
    @     0x7f8f6b6e2625  op::FaceExtractorNet::initializationOnThread()
    @     0x7f8f6b77b297  op::Worker<>::initializationOnThreadNoException()
    @     0x7f8f6b77b3e8  op::SubThread<>::initializationOnThread()
    @     0x7f8f6b77cc00  op::Thread<>::initializationOnThread()
    @     0x7f8f6b7801bc  op::Thread<>::threadFunction()
    @     0x7f8f6b396de4  (unknown)
    @     0x7f8f6b096609  start_thread
    @     0x7f8f6b1d2133  clone
中止 (コアダンプ)

```

osumilab, 2022-12-08:
I completely uninstalled Openpose and caffe , and re-installed its based on documentation of Openpose,but I still get the error ```F1208 14:10:00.528839 2225162 cudnn_conv_layer.cpp:53] Check failed: status == CUDNN_STATUS_SUCCESS (4 vs. 0)  CUDNN_STATUS_INTERNAL_ERROR```
I run ```./build/examples/openpose/openpose.bin``` was success, I run ```./build/examples/openpose/openpose.bin --video examples/media/video.avi``` was success too ,but I run ```./build/examples/openpose/openpose.bin --face --hand``` was failed
How did people who have been in the same situation overcome...

ravijo, 2022-12-08:
I understand.

The `CUDNN_STATUS_INTERNAL_ERROR` error is a bit tricky to solve. There are many potential causes for it, such as low GPU RAM, incorrect cuDNN version, and so on.

I think, you should spend some time going through OpenPose documentation. or ask at their GitHub page.

xiang2020git, 2022-12-08:
I also had the problem with `[libcaffe.so.1.0.0](libcaffe.so.1.0.0: cannot open shared object file: No such file or directory)` when running the launch file run.launch.

A quick fix is to load the path to your compiled libcaffe.so.1.0.0 before running the launch file:

```
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/xc/penpose/build/caffe/lib
export LD_LIBRARY_PATH
```
For a long-time fix: maybe we need to modify the CMakeLists.txt or make a symbolic link of the libcaffe.so.1.0.0 into some system path.

osumilab, 2022-12-09:
I load the path to ```libcaffe.so.q.0.0```
I run ```~/openpose$ ./build/examples/openpose/openpose.bin --video examples/media/video.avi --face --hand --write_json output_json_folder/```, it gives me error like this.
```F1209 15:41:14.624164 14581 cudnn_conv_layer.cpp:53] Check failed: status == CUDNN_STATUS_SUCCESS (1 vs. 0)  CUDNN_STATUS_NOT_INITIALIZED *** Check failure stack trace: ***```
But, I run ```roslaunch ros_openpose run.launch```, It gives me message and image was displayed.
What does it mean?

```
... logging to /home/osumi/.ros/log/66d2fe66-778c-11ed-9bc9-b118c59ab965/roslaunch-osumi-System-Product-Name-13921.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.91.72.99:34217/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
[... 123 lines cut ...]
[ INFO] [1670568043.531367706]: Motion Module was found.
[ INFO] [1670568043.531420901]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1670568043.531623221]: Add Filter: pointcloud
[ INFO] [1670568043.531950937]: num_filters: 2
[ INFO] [1670568043.531966224]: Setting Dynamic reconfig parameters.
[ INFO] [1670568046.470178005]: Done Setting Dynamic reconfig parameters.
[ INFO] [1670568046.470502202]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1670568046.471197360]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1670568046.471973349]: setupPublishers...
[ INFO] [1670568046.473949952]: Expected frequency for depth = 30.00000
[ INFO] [1670568046.494646996]: Expected frequency for color = 30.00000
[ INFO] [1670568046.506072003]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1670568046.518897625]: setupStreams...
 09/12 15:40:46,719 WARNING [139697963656960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1670568046.770152328]: SELECTED BASE:Depth, 0
[ INFO] [1670568046.775493215]: RealSense Node Is Up!
[ WARN] [1670568046.838260599]: 
 09/12 15:40:46,838 WARNING [139697963656960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1670568046.843057164]: publishing unordered pointcloud.
 09/12 15:40:46,889 WARNING [139697963656960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 09/12 15:40:46,940 WARNING [139697963656960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 09/12 15:40:47,110 WARNING [139697963656960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 09/12 15:40:47,160 WARNING [139697963656960] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
```
[image]

xiang2020git, 2022-12-09:
I haven't met the error regarding Cudnn yet therefore I can't give too much help regarding this issue.

For the error 
`
Prototxt file not found: home/osumi/openpose/models/pose/body_25/pose_deploy.prototxt.
`
In run.launch, you need a slash at the beginning of the path:
`
<arg name="openpose_args" value="--model_folder /home/osumi/openpose/models/pose/body_25/pose_deploy.prototxt />
`

osumilab, 2022-12-10:
I changed run.launch as follows, and I get error as well.
Is there a root cause somewhere?
ros.launch
```
<?xml version="1.0"?>
<!--
run.launch: this file does no more than calling other launch file
Author: Ravi Joshi
Date: 2019/11/23
-->
<launch>

  <!-- set this flag to switch ros_openpose to a synchronous setting -->
  <arg name="synchronous" default="false" />

  <!-- specify path to python openpose library (specifying this will switch synchronous setting from c++ to python) -->
  <arg name="py_openpose_path" default="none" />

  <!-- the value of this argument can only be one of the following: realsense, kinect, zed2, nodepth -->
  <arg name="camera" default="realsense" />

  <!-- set this flag to see invoke RViz -->
  <arg name="rviz" default="true" />

  <!-- set this flag to invoke the visualization of the skeleton -->
  <arg name="skeleton" default="true" />

  <!-- set this flag to see the pointcloud in the visualization inside RViz  -->
  <arg name="pointcloud" default="true" />

  <!-- set this flag to invoke the visualization of the hands of the skeleton -->
  <arg name="skeleton_hands" default="false" />

  <!--
   the standard openpose command-line arguments are also supported.
   to do so, please change the value, as shown in the example below.
  -->
  <?ignore
    <arg name="openpose_args" value="--hand --disable_multi_thread --model_folder /home/ravi/openpose/models/" />
  ?>
  <!-- full path to the model dir of openpose -->
  <arg name="openpose_args" value="--model_folder /home/osumi/openpose/models/pose/body_25/pose_deploy.prototxt" />

  <!-- rostopic to publish the 3D skeleton data -->
  <arg name="pub_topic" default="/frame" />

  <!-- thickness of the line used to draw skeleton for visualization inside RViz -->
  <arg name="skeleton_line_width" default="0.01" />

  <!-- size of the text used to indicate the id of a skeleton for visualization inside RViz -->
  <arg name="id_text_size" default="0.2" />

  <!-- offeset of the text used to indicate the id of a skeleton for visualization inside RViz -->
  <arg name="id_text_offset" default="-0.05" />

  <!-- flag to enable printing ros message to terminal -->
  <arg name="print" default="true" />

  <include file="$(find ros_openpose)/launch/config_$(arg camera).launch" >
    <arg name="rviz" value="$(arg rviz)" />
    <arg name="print" value="$(arg print)" />
    <arg name="skeleton" value="$(arg skeleton)" />
    <arg name="pub_topic" value="$(arg pub_topic)" />
    <arg name="pointcloud" value="$(arg pointcloud)" />
    <arg name="id_text_size" value="$(arg id_text_size)" />
    <arg name="openpose_args" value="$(arg openpose_args)" />
    <arg name="skeleton_hands" value="$(arg skeleton_hands)" />
    <arg name="id_text_offset" value="$(arg id_text_offset)" />
    <arg name="skeleton_line_width" value="$(arg skeleton_line_width)" />
    <arg name="synchronous" value="$(arg synchronous)" />
[... 226 lines cut ...]
Error occurred on a thread. OpenPose closed all its threads and then propagated the error to the main thread. Error description:

Prototxt file not found: /home/osumi/openpose/models/pose/body_25/pose_deploy.prototxt/pose/body_25/pose_deploy.prototxt.
Possible causes:
	1. Not downloading the OpenPose trained models.
	2. Not running OpenPose from the root directory (i.e., where the `model` folder is located, but do not move the `model` folder!). E.g.,
		Right example for the Windows portable binary: `cd {OpenPose_root_path}; bin/openpose.exe`
		Wrong example for the Windows portable binary: `cd {OpenPose_root_path}/bin; openpose.exe`
	3. Using paths with spaces.

Coming from:
- /home/osumi/openpose/src/openpose/net/netCaffe.cpp:ImplNetCaffe():61
- /home/osumi/openpose/src/openpose/net/netCaffe.cpp:ImplNetCaffe():97
- /home/osumi/openpose/src/openpose/pose/poseExtractorCaffe.cpp:addCaffeNetOnThread():106
- /home/osumi/openpose/src/openpose/pose/poseExtractorCaffe.cpp:netInitializationOnThread():196
- /home/osumi/openpose/src/openpose/pose/poseExtractorNet.cpp:initializationOnThread():102
- /home/osumi/openpose/src/openpose/pose/poseExtractor.cpp:initializationOnThread():34
- /home/osumi/openpose/include/openpose/pose/wPoseExtractor.hpp:initializationOnThread():57
- /home/osumi/openpose/include/openpose/thread/worker.hpp:initializationOnThreadNoException():77
- [All threads closed and control returned to main thread]
- /home/osumi/openpose/src/openpose/utilities/errorAndLog.cpp:checkWorkerErrors():280
- /home/osumi/openpose/include/openpose/thread/threadManager.hpp:stop():243
- /home/osumi/openpose/include/openpose/wrapper/wrapper.hpp:stop():455

```

osumilab, 2022-12-10:
Of course, ```pose_deploy.prototxt``` exists here.
[image]

osumilab, 2022-12-10:
Sorry, I described ```run.launch``` as```<arg name="openpose_args" value="--model_folder /home/osumi/openpose/models/" />``` , and rviz and openpose started up.

Also, I want to do hand tracking, regarding the configuration of launch file in README, should I describe ```<arg name="openpose_args" value="--model_folder /home/osumi/openpose/models/"/>``` or ```<arg name="openpose_args" value="--face --hand"/>```?
I cannot describe both because they have the same ```arg name```

osumilab, 2022-12-15:
Sorry to keep pestering you. 
Does this error mean that there is not enough memory?

```
[ WARN] [1671146881.665090298]: Waiting for datum...
point: ['0.291', '0.064', 0.9410000000000001]
F1216 08:28:01.971036 142037 syncedmem.cpp:78] Check failed: error == cudaSuccess (2 vs. 0)  out of memory
*** Check failure stack trace: ***
    @     0x7f5f3955c1c3  google::LogMessage::Fail()
    @     0x7f5f3956125b  google::LogMessage::SendToLog()
    @     0x7f5f3955bebf  google::LogMessage::Flush()
    @     0x7f5f3955c6ef  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f5f392af20a  caffe::SyncedMemory::gpu_data()
    @     0x7f5f391470f6  caffe::Blob<>::gpu_data()
    @     0x7f5f392e7a01  caffe::ConvolutionLayer<>::Forward_gpu()
    @     0x7f5f3926e292  caffe::Net<>::ForwardFromTo()
    @     0x7f5f3a2a2c19  op::NetCaffe::forwardPass()
    @     0x7f5f3a24dde7  op::FaceExtractorCaffe::forwardPass()
    @     0x7f5f3a248208  op::WFaceExtractorNet<>::work()
    @     0x7f5f3a2e952f  op::Worker<>::checkAndWork()
    @     0x7f5f3a2e96bb  op::SubThread<>::workTWorkers()
    @     0x7f5f3a2f723d  op::SubThreadQueueInOut<>::work()
    @     0x7f5f3a2ee271  op::Thread<>::threadFunction()
    @     0x7f5f398b9de4  (unknown)
    @     0x7f5f39a61609  start_thread
    @     0x7f5f396b1133  clone
================================================================================REQUIRED process [rosOpenpose-2] has died!
process has died [pid 141937, exit code -6, cmd /home/osumi/catkin_ws/devel/lib/ros_openpose/rosOpenposeAsync --face --hand --model_folder /home/osumi/openpose/models/ __name:=rosOpenpose __log:=/home/osumi/.ros/log/119671d2-7cd0-11ed-9bc9-b118c59ab965/rosOpenpose-2.log].
log file: /home/osumi/.ros/log/119671d2-7cd0-11ed-9bc9-b118c59ab965/rosOpenpose-2*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[camera/realsense2_camera-6] killing on exit
[visualizer-3] killing on exit
[rosOpenpose-2] killing on exit
[echo-4] killing on exit
[camera/realsense2_camera_manager-5] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

ravijo, 2022-12-16:
Yes! You need more memory in your graphic card. Or you can lower the net resolution parameter. Please check the documentation

osumilab, 2022-12-16:
Thank you for answering my many question!
I am learning by trial and error!

closed 2022-12-16 by osumilab