ravijo/ros_openpose #85: (OpenPose 1.7.0:16877): Gtk-ERROR **

TinkerYxs, 2022-12-14:
$ roslaunch ros_openpose run.launch camera:=rfly_nodepth 
... logging to /home/nvidia/.ros/log/1a1933e0-7b71-11ed-8f1b-48b02d694004/roslaunch-ubuntu-16853.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:43407/

SUMMARY
========

PARAMETERS
 * /echo/pub_topic: /frame
 * /rosOpenpose/cam_info_topic: cam_info_topic
 * /rosOpenpose/color_topic: /camera/rgb
 * /rosOpenpose/depth_topic: /camera/depth
 * /rosOpenpose/frame_id: no_depth
 * /rosOpenpose/no_depth: True
 * /rosOpenpose/pub_topic: /frame
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /
    echo (ros_openpose/echo.py)
    rosOpenpose (ros_openpose/rosOpenposeAsync)

auto-starting new master
process[master]: started with pid [16861]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1a1933e0-7b71-11ed-8f1b-48b02d694004
process[rosout-1]: started with pid [16871]
started core service [/rosout]
process[rosOpenpose-2]: started with pid [16877]
process[echo-3]: started with pid [16879]
[ INFO] [1670996123.218492638]: Starting ros_openpose...
[ INFO] [1670996123.219720352]: 1
[ INFO] [1670996123.219963073]: 2
Auto-detecting all available GPUs... Detected 1 GPU(s), using 1 of them starting at GPU 0.
[ WARN] [1670996123.241249413]: Waiting for color image frame...
[ INFO] [1670996123.242236999]: 3
[ WARN] [1670996123.242357895]: Waiting for datum...

(OpenPose 1.7.0:16877): Gtk-ERROR **: 13:35:23.243: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
================================================================================REQUIRED process [rosOpenpose-2] has died!
process has died [pid 16877, exit code -5, cmd /home/nvidia/32Tracking_ws/devel/lib/ros_openpose/rosOpenposeAsync --model_folder /home/nvidia/32Tracking_ws/src/openpose/models/ __name:=rosOpenpose __log:=/home/nvidia/.ros/log/1a1933e0-7b71-11ed-8f1b-48b02d694004/rosOpenpose-2.log].
log file: /home/nvidia/.ros/log/1a1933e0-7b71-11ed-8f1b-48b02d694004/rosOpenpose-2*.log
Initiating shutdown!
================================================================================
[rosOpenpose-2] killing on exit
[echo-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

ravijo, 2022-12-14:
@TinkerYxs 

The GTK error is most probably coming from OpenCV. I guess that you are using a custom OpenCV. To debug your issue, can you please do the following:

1. Run OpenPose as a standalone program by giving an example video mentioned in [OpenPose/Quick Start Overview](https://github.com/CMU-Perceptual-Computing-Lab/openpose#quick-start-overview).
2. Please make sure GUI inside ROS works fine by executing the following command:
    ```console
    rosrun image_view image_view image:=/your_camera/image_topic
    ```

BTW, are you defining your new camera `rfly_nodepth`? If so, we can add your contributions to this repository after discussions with you.

TinkerYxs, 2022-12-15:
> From your command line output I assume that you use a Nvidia Jetson board. I had this gtk version issue with my AGX Orin. My wild guess is, this is due to some openCV version mismatch from cv_bridge and openpose, see [here](https://github.com/CMU-Perceptual-Computing-Lab/openpose/issues/117)
> 
> I did the following to solve this:
> 
>     1. Build and install openCV 4.6 with CUDA enabled. The preinstalled openCV in Jetson Orin does not support CUDA.
> 
>     2. Build openpose with openCV 4.6. You can do this by specifying `OPENCV_DIR` to the installed path of openCV in `cmake-gui` interface, see [this Reference](https://github.com/CMU-Perceptual-Computing-Lab/openpose/blob/master/doc/installation/2_additional_settings.md#custom-opencv)
> 
>     3. Edit `cv_bridgeConfig.cmake` to tell `catkin_make` to find the newly installed openCV, see [this link](https://github.com/ravijo/ros_openpose/issues/62#issuecomment-1123430014)
>        
>        * I am not sure if this step is mandatory because i tried many ways to make the whole system work. But in the end it works.
> 
>     4. If you installed ROS using `àpt`, then it is probably that the `cv_bridge` package is also built with another openCV. To solve this, you can build the `vision_opencv` with `ros_openpose` using openCV 4.6:
>        
>        1. Assume that your workspace of `ros_openpose` is located in `~/catkin_ws/src`.
>        2. Download the source code of `vision_opencv`, which contains `cv_bridge`
>           ```
>           cd ~/catkin_ws/src
>           git clone https://github.com/ros-perception/vision_opencv.git
>           cd vision_opencv
>           git checkout noetic
>           ```
>        3. build `vision_opencv` and `ros_openpose` together:
>           ```
>           cd ~/catkin_ws
>           catkin_make
>           ```
> 
> 
> Then it should work.

yes, i use jetson agx ogin and my opencv version is 4.5.0(default and not compiled by cuda),  let me try your method.

TinkerYxs, 2022-12-15:
> From your command line output I assume that you use a Nvidia Jetson board. I had this gtk version issue with my AGX Orin. My wild guess is, this is due to some openCV version mismatch from cv_bridge [...]

y

TinkerYxs, 2022-12-15:
yes, i use jetson agx ogin and my opencv version is 4.5.0(default and not compiled by cuda), let me try your method.

TinkerYxs, 2022-12-15:
> @TinkerYxs
> 
> The GTK error is most probably coming from OpenCV. I guess that you are using a custom OpenCV. To debug your issue, can you please do the following:
> 
>     1. Run OpenPose as a standalone program by giving an example video mentioned in [OpenPose/Quick Start Overview](https://github.com/CMU-Perceptual-Computing-Lab/openpose#quick-start-overview).
> 
>     2. Please make sure GUI inside ROS works fine by executing the following command:
>        ```
>        rosrun image_view image_view image:=/your_camera/image_topic
>        ```
> 
> 
> BTW, are you defining your new camera `rfly_nodepth`? If so, we can add your contributions to this repository after discussions with you.

my openpose runs just fine and i  just use  a simulation camera

TinkerYxs, 2022-12-16:
> From your command line output I assume that you use a Nvidia Jetson board. I had this gtk version issue with my AGX Orin. My wild guess is, this is due to some openCV version mismatch from cv_bridge [...]

Thank you very much, your method slove this problem, I install opencv4.6.0 and cover the default opencv4.5.4,  after this I meet another problem " error while loading shared libraries: libopencv_highgui.so.406: cannot open shared object file: No such file or directory", I "sudo apt-get install libopencv-highgui-dev", then my openpose and ros_openpose  run correctly.

closed 2022-12-16 by TinkerYxs