ravijo/ros_openpose #88: REQUIRED process [rosOpenpose-2] has died!

vin3697, 2023-05-11:
Hello Ravi,

Thank you very much for your ROS openpose work.

The ROS openpose works fine with my USB camera ::  roslaunch ros_openpose run.launch camera:=nodepth 

But,
 
I am having issue with kinect xbox 360 camera.
--

The changes made by me in the launch files are as below::

1. run.launch file

	In this file i just changed the below arguments:
		-	
  			<arg name="camera" default="azurekinect" />
  		
  		-   
        	<arg name="openpose_args" value="--disable_multi_thread --net_resolution 128x96 --model_folder /openpose/models/ --model_pose COCO" />

2. config_azurekinect.launch

					<?xml version="1.0"?>
					
					<launch>
					  
					  <arg name="color_topic" default="/camera/rgb/image_raw" />

					  
					  <arg name="depth_topic" default="/camera/depth_to_rgb/image_raw" />

					  
					  <arg name="cam_info_topic" default="/camera/rgb/camera_info" />

					  
					  <arg name="frame_id" default="openni_rgb_optical_frame" />

					  
					  <arg name="no_depth" default="false" />

					  
					  <arg name="rviz" default="" />
					  <arg name="print" default="" />
					  <arg name="skeleton" default="" />
					  <arg name="pub_topic" default="" />
					  <arg name="pointcloud" default="" />
					  <arg name="id_text_size" default="" />
					  <arg name="openpose_args" default="" />
					  <arg name="id_text_offset" default="" />
					  <arg name="skeleton_hands" default="" />
					  <arg name="skeleton_line_width" default="" />
					  <arg name="synchronous" default="" />
					  <arg name="py_openpose_path" default="" />

					  <include file="$(find ros_openpose)/launch/core.launch" >
						<arg name="print" value="$(arg print)" />
						<arg name="no_depth" value="$(arg no_depth)" />
						<arg name="skeleton" value="$(arg skeleton)" />
						<arg name="frame_id" value="$(arg frame_id)" />
						<arg name="pub_topic" value="$(arg pub_topic)" />
						<arg name="color_topic" value="$(arg color_topic)" />
						<arg name="depth_topic" value="$(arg depth_topic)" />
						<arg name="id_text_size" value="$(arg id_text_size)" />
						<arg name="openpose_args" value="$(arg openpose_args)" />
						<arg name="id_text_offset" value="$(arg id_text_offset)" />
						<arg name="skeleton_hands" value="$(arg skeleton_hands)" />
						<arg name="cam_info_topic" value="$(arg cam_info_topic)" />
						<arg name="skeleton_line_width" value="$(arg skeleton_line_width)" />
						<arg name="synchronous" value="$(arg synchronous)" />
						<arg name="py_openpose_path" value="$(arg py_openpose_path)" />
					  </include>

					  
					  <group if="$(arg pointcloud)" >
						<include file="$(find freenect_launch)/launch/freenect.launch" >
						  
						</include>
						
						<group if="$(arg rviz)" >
						  <node pkg="rviz" type="rviz" respawn="false" required="true" name="rviz" args="-d $(find ros_openpose)/files/person_pointcloud_azurekinect.rviz" />
						</group>
					  </group>

					  
					  <group unless="$(arg pointcloud)" >
						<include file="$(find freenect_launch)/launch/freenect.launch" >
						  
						</include>
						
						<group if="$(arg rviz)" >
						  <node pkg="rviz" type="rviz" respawn="false" required="true" name="rviz" args="-d $(find ros_openpose)/files/only_person_azurekinect.rviz" />
						</group>
					  </group>
					</launch>

----
As you will notice that here I am not using the Azure_Kinect_ROS_Driver as I am having issues with it :: [ERROR] [1683817163.396425478]: Failed to open a K4A device. Cannot continue. 
Also with Openni ROS driver the kinect xbxo 360 device is not getting detecting.

thats why I have used freenect ROS driver, and made the above changes accordingly in the launch file. I have commented this below part::
										
						  				

I havent made any changes in the " /files/ " folder particularly in this file -> person_pointcloud_azurekinect.rviz

--

The problem is, whenever I launch 
		
		roslaunch ros_openpose run.launch 

everything works fine untill a person is detected in a frame, as soon as a person is present in the frame the process dies and below is the error:
--

[INFO] [1683815831.862595]: []

[INFO] [1683815831.896609]: []

[INFO] [1683815831.930698]: []

================================================================================
REQUIRED process [rosOpenpose-2] has died!
process has died [pid 409, exit code -11, cmd /root/openpose_ros/devel/lib/ros_openpose/rosOpenposeAsync --disable_multi_thread --net_resolution 128x96 --model_folder /openpose/models/ --model_pose COCO __name:=rosOpenpose __log:=/root/.ros/log/43193d1e-f009-11ed-8359-9408533a4973/rosOpenpose-2.log].
log file: /root/.ros/log/43193d1e-f009-11ed-8359-9408533a4973/rosOpenpose-2*.log
Initiating shutdown!
================================================================================
[rviz-29] killing on exit

I am not able to find below file, might be getting deleted after creation::
/root/.ros/log/43193d1e-f009-11ed-8359-9408533a4973/rosOpenpose-2*.log

-- -- --

Below is my GPU utilization

-- -- --

+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.182.03   Driver Version: 470.182.03   CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   50C    P0    18W /  N/A |   1544MiB /  3911MiB |     81%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1033      G   /usr/lib/xorg/Xorg                  4MiB |
|    0   N/A  N/A      1901      G   /usr/lib/xorg/Xorg                  4MiB |
|    0   N/A  N/A      5615      C   ...openpose/rosOpenposeAsync     1534MiB |
+-----------------------------------------------------------------------------+
Thu May 11 16:37:12 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.182.03   Driver Version: 470.182.03   CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   50C    P0    18W /  N/A |     10MiB /  3911MiB |     81%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

-- -- --

Thank you for reading thorugh this long text.

ravijo, 2023-05-12:
@vin3697 

Is the `freenect` driver still supported? BTW, there are many places where things could go wrong. So, please debug the issue step by step as mentioned below:

1. Please make sure you are using correctly formatted XML. Most of the XML you shared above was incorrect. I tried correcting it but left in between.
2. Can you please run the standalone OpenPose to confirm that OpenPose is installed correctly and works well? Please run the command as mentioned in [this section](https://github.com/CMU-Perceptual-Computing-Lab/openpose#quick-start-overview).
3. Using the freenect driver, can you please make sure depth and color frames are received properly? You may use RViz to view these topics.
4. Please check [Camera Run Instructions](https://github.com/ravijo/ros_openpose#camera-run-instructions) and [FAQ](https://github.com/ravijo/ros_openpose#faq) sections as well.

Thanks

vin3697, 2023-05-15:
Dear @ravijo ,

I created a launch file named config_kinectxbox.launch where I have made changes as below,
    
    <arg name="color_topic" default="/camera/rgb/image_raw" />
  
    
    <arg name="depth_topic" default="/camera/depth_to_rgb/image_raw" />
  
    
    <arg name="cam_info_topic" default="/camera/rgb/camera_info" />
  
    
    <arg name="frame_id" default="camera_rgb_frame" />

    
    <include file="$(find freenect_launch)/launch/freenect.launch" >
      <arg name="publish_tf" value="true" />
    </include>
    
    
    <group if="$(arg rviz)" >
      <node pkg="rviz" type="rviz" respawn="false" required="true" name="rviz"/>
    </group>

I am just launching the RVIZ without any prior settings.

Still I am getting the same issue.
At present, to keep the work going on, I have made a change in the config_nodepth.launch file (gave the RGB image topic), and its working fine with it.

If i need any further help I will open this issue again.
Thank you for your help!

Regards
Vin

closed 2023-05-15 by vin3697

vin3697, 2023-05-27:
Hello Ravi,

I am able to run your ROS openpose wrapper with Kinect Xbox 360 camera.
Sharing these files if someone is working with it:
For RVIZ files, I have just compared with other files and created with some changes.

1. config_kinectxbox.launch

<?xml version="1.0"?>

<launch>

  <param name="/use_sim_time" value="true"/>

  
  <arg name="color_topic" default="/camera/rgb/image_raw" />

  
  <arg name="depth_topic" default="/camera/depth_registered/image_raw" />

  
  <arg name="cam_info_topic" default="/camera/rgb/camera_info" />

  
  <arg name="frame_id" default="camera_depth_optical_frame" />

  
  <arg name="no_depth" default="false" />

  
  <arg name="rviz" default="" />
  <arg name="print" default="" />
  <arg name="skeleton" default="" />
  <arg name="pub_topic" default="" />
  <arg name="pointcloud" default="" />
  <arg name="id_text_size" default="" />
  <arg name="openpose_args" default="" />
  <arg name="id_text_offset" default="" />
  <arg name="skeleton_hands" default="" />
  <arg name="skeleton_line_width" default="" />
  <arg name="synchronous" default="" />
  <arg name="py_openpose_path" default="" />

  <include file="$(find ros_openpose)/launch/core.launch" >
    <arg name="print" value="$(arg print)" />
    <arg name="no_depth" value="$(arg no_depth)" />
    <arg name="skeleton" value="$(arg skeleton)" />
    <arg name="frame_id" value="$(arg frame_id)" />
    <arg name="pub_topic" value="$(arg pub_topic)" />
    <arg name="color_topic" value="$(arg color_topic)" />
    <arg name="depth_topic" value="$(arg depth_topic)" />
    <arg name="id_text_size" value="$(arg id_text_size)" />
    <arg name="openpose_args" value="$(arg openpose_args)" />
    <arg name="id_text_offset" value="$(arg id_text_offset)" />
    <arg name="skeleton_hands" value="$(arg skeleton_hands)" />
    <arg name="cam_info_topic" value="$(arg cam_info_topic)" />
    <arg name="skeleton_line_width" value="$(arg skeleton_line_width)" />
    <arg name="synchronous" value="$(arg synchronous)" />
    <arg name="py_openpose_path" value="$(arg py_openpose_path)" />
  </include>

  

  
  <group if="$(arg pointcloud)" >
    <include file="$(find freenect_launch)/launch/freenect.launch" >
      <arg name="publish_tf" value="true" />
    </include>
    
    <group if="$(arg rviz)" >
[... 229 lines cut ...]
    Saved: ~
Window Geometry:
  Displays:
    collapsed: false
  Height: 1056
  Hide Left Dock: false
  Hide Right Dock: true
  QMainWindow State: 000000ff00000000fd00000004000000000000019900000394fc0200000008fb0000001200530065006c0065006300740069006f006e00000001e10000009b0000006400fffffffb0000001e0054006f006f006c002000500072006f007000650072007400690065007302000001ed000001df00000185000000a3fb000000120056006900650077007300200054006f006f02000001df000002110000018500000122fb000000200054006f006f006c002000500072006f0070006500720074006900650073003203000002880000011d000002210000017afb000000100044006900730070006c006100790073010000002800000394000000dd00fffffffb0000002000730065006c0065006300740069006f006e00200062007500660066006500720200000138000000aa0000023a00000294fb00000014005700690064006500530074006500720065006f02000000e6000000d2000003ee0000030bfb0000000c004b0069006e0065006300740200000186000001060000030c00000261000000010000010f00000394fc0200000003fb0000001e0054006f006f006c002000500072006f00700065007200740069006500730100000041000000780000000000000000fb0000000a00560069006500770073000000002800000394000000b000fffffffb0000001200530065006c0065006300740069006f006e010000025a000000b200000000000000000000000200000490000000a9fc0100000001fb0000000a00560069006500770073030000004e00000080000002e100000197000000030000073f00000040fc0100000002fb0000000800540069006d006501000000000000073f000002f600fffffffb0000000800540069006d00650100000000000004500000000000000000000005a00000039400000004000000040000000800000008fc0000000100000002000000010000000a0054006f006f006c00730100000000ffffffff0000000000000000
  Selection:
    collapsed: false
  Time:
    collapsed: false
  Tool Properties:
    collapsed: false
  Views:
    collapsed: true
  Width: 1855
  X: 1975
  Y: 14

4. roslaunch ros_openpose run.launch camera:=kinectxbox

I have these minor issues which are unimportant atleast for now #78 and #41 / #31

Thank you for your work.