ravijo/ros_openpose #89: Lunch file operation error

Parkeunseok, 2023-06-07:
Hello, I'm a beginner in this field using your code.
The basic open pose works, but when you operate your launch file, you get the following error. Please solve this problem. I'm sending you a screenshot of the error and lunch file I came out with.
Thank you.

<?xml version="1.0"?>

<launch>

  
  <arg name="synchronous" default="false" />

  
  <arg name="py_openpose_path" default="none" />

  
  <arg name="camera" default="nodepth" />

  
  <arg name="rviz" default="true" />

  
  <arg name="skeleton" default="true" />

  
  <arg name="pointcloud" default="true" />

  
  <arg name="skeleton_hands" default="false" />

  
  <?ignore
    <arg name="openpose_args" value="--hand --disable_multi_thread --model_folder /home/ravi/openpose/models/" />
  ?>
  
  <arg name="openpose_args" value="--model_folder $(env HOME)/openpose/models/" />

  
  <arg name="pub_topic" default="/frame" />

  
  <arg name="skeleton_line_width" default="0.01" />

  
  <arg name="id_text_size" default="0.2" />

  
  <arg name="id_text_offset" default="-0.05" />

  
  <arg name="print" default="true" />

  <include file="$(find ros_openpose)/launch/config_$(arg camera).launch" >
    <arg name="rviz" value="$(arg rviz)" />
    <arg name="print" value="$(arg print)" />
    <arg name="skeleton" value="$(arg skeleton)" />
    <arg name="pub_topic" value="$(arg pub_topic)" />
    <arg name="pointcloud" value="$(arg pointcloud)" />
    <arg name="id_text_size" value="$(arg id_text_size)" />
    <arg name="openpose_args" value="$(arg openpose_args)" />
    <arg name="skeleton_hands" value="$(arg skeleton_hands)" />
    <arg name="id_text_offset" value="$(arg id_text_offset)" />
    <arg name="skeleton_line_width" value="$(arg skeleton_line_width)" />
    <arg name="synchronous" value="$(arg synchronous)" />
    <arg name="py_openpose_path" value="$(arg py_openpose_path)" />
  </include>
</launch>

[image]

This is the default open pose motion screen.
[image]

ravijo, 2023-06-08:
Hello @Parkeunseok 

Thanks for reporting the issue. It seems that the error is coming from OpenPose installation.
```
error while loading shared libraries: libcaffe.so.1.0.0: cannot open shared object file: No such file or directory
```
 Please check [this link](https://github.com/CMU-Perceptual-Computing-Lab/openpose/issues/148) and set `LD_LIBRARY_PATH`. Also, please check the following commented lines: https://github.com/ravijo/ros_openpose/blob/21e6c3e23bd49340e04aee1341e894142d9f66e7/CMakeLists.txt#L82-L85

You may have to set `CAFFE_LIB_FOLDER` to the location of `libcaffe.so` file.

Parkeunseok, 2023-06-09:
A new error has appeared. Can you solve this, too?
[image]

ravijo, 2023-06-09:
It seems that the output topic, where you get the processed data is set incorrectly.

The easiest way to confirm my hypothesis is to add a print statement at line 23:
https://github.com/ravijo/ros_openpose/blob/21e6c3e23bd49340e04aee1341e894142d9f66e7/scripts/echo.py#L22-L24

```
    rospy.loginfo('frame_topic: %s' % frame_topic)
```

If so, then you need to set it. Probably you have modifed the launch files incorrectly.

Parkeunseok, 2023-06-09:
What is the solution to this error??
[image]

first computer
RAM : 16GB
CPU : 12th Gen Intel® Core™ i7-12700KF × 20 
GPU : NVIDIA GeForce RTX 3080 

second computer
RAM : 16GB
CPU : 9th Gen Intel® Core™ i7-9750H 
GPU : NVIDIA GeForce GTX 1660Ti

ravijo, 2023-06-09:
The error shown in your screenshot complaints about insufficient memory in your graphic card.

There are various ways to deal with it. I suggest checking the documentation please

ravijo, 2023-12-02:
Closing this issue as no further comments have been received. However, feel free to comment if the problem still exists.

closed 2023-12-02 by ravijo