ravijo/ros_openpose #92: run.launch error occurr

viorinus, 2024-09-15:
Hello @ravijo,

I've encountered a similar issue to @SM30000.

When I run：`roslaunch ros_openpose run.launch`
I get the following error:

[visualizer-2] process has died [pid 38959, exit code 1, cmd /home/r/openpose/src/ros_openpose/scripts/visualizer.py __name:=visualizer __log:=/home/r/.ros/log/b300b026-734d-11ef-ba19-e954e920a64d/visualizer-2.log].
log file: /home/r/.ros/log/b300b026-734d-11ef-ba19-e954e920a64d/visualizer-2*.log
================================================================================REQUIRED process [echo-3] has died!
process has died [pid 38960, exit code 1, cmd /home/r/openpose/src/ros_openpose/scripts/echo.py __name:=echo __log:=/home/r/.ros/log/b300b026-734d-11ef-ba19-e954e920a64d/echo-3.log].
log file: /home/r/.ros/log/b300b026-734d-11ef-ba19-e954e920a64d/echo-3*.log
Initiating shutdown!
================================================================================

However, when I run: `roslaunch ros_openpose run.launch camera:=realsense skeleton:=false print:=false`

it works fine, and the `Color topic` `Depth topic ` `Camera Info. topic`  are printing messages as expected.

But I'm still unable to run `visualizer.py`or `echo.py`. Even after changing the shebang to `#!/usr/bin/env python3` and running them directly with `python3 src/ros_openpose/scripts/visualizer.py`, I encounter the following error:

Traceback (most recent call last):
  File "src/ros_openpose/scripts/visualizer.py", line 200, in <module>
    frame_topic = rospy.get_param('~pub_topic')
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 467, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/msproxy.py", line 123, in __getitem__
    raise KeyError(key)
KeyError: '~pub_topic'

I'm running on Ubuntu 20.04 with ROS Noetic, OpenPose 1.6, CUDA 12.6, and cuDNN 9.Could you please guide me on how to resolve this issue?

ravijo, 2024-09-15:
@viorinus 

It seems that the script is unable to find `pub_topic` argument because you are running them directly instead of using `rosrun`.  The `visualizer.py` and `echo.py` requires some ROS parameters to run. Please do not invoke them directly with python.

BTW, make sure these scripts are executable as mentioned at point 3 in [installation](https://github.com/ravijo/ros_openpose?tab=readme-ov-file#installation)

viorinus, 2024-09-15:
Hello @ravijo,

Thank you for your prompt response.

The scripts are executable, and when I attempt to run `rosrun ros_openpose visualizer.py`, I encounter the following error:

Traceback (most recent call last):
  File "/home/r/openpose/src/ros_openpose/scripts/visualizer.py", line 10, in <module>
    import rospy
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/_Bool.py", line 6, in <module>
    import genpy
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 48, in <module>
    import yaml
ImportError: No module named yaml

Initially, I thought this issue might be due to a Python version conflict in my system, so I omitted the related description. Could you please let me know if my understanding is correct?

ravijo, 2024-09-15:
@viorinus 

It seems a problem with your ROS installtion. Either you are running on incorrect python version or ROS isn't installed properly.

viorinus, 2024-09-15:
Thank you for your response. I will try to install the correct version of Python first, and I will report back on my progress soon!

viorinus, 2024-09-16:
@ravijo 

I've tried setting up a Python 3.8 virtual environment to run my code, 

```
(python38) r@r:~/openpose$ python3 -V
Python 3.8.19
```

but it's still throwing errors. 

Traceback (most recent call last):
File "/home/r/openpose/src/ros_openpose/scripts/visualizer.py", line 10, in
import rospy
File "/opt/ros/noetic/lib/python3/dist-packages/rospy/init.py", line 47, in
from std_msgs.msg import Header
File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/init.py", line 1, in
from ._Bool import *
File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/_Bool.py", line 6, in
import genpy
File "/opt/ros/noetic/lib/python3/dist-packages/genpy/init.py", line 34, in
from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 48, in
import yaml
ImportError: No module named yaml

When I ran the `conda list` command, I can see that `yaml` and `pyyaml `are indeed installed. 

[image]

I try to reinstalling yaml, and I got :

```
(python38) r@r:~$ pip install pyyaml
Requirement already satisfied: pyyaml in ./.conda/envs/python38/lib/python3.8/site-packages (6.0.1)

```

I attempted to install pyyaml in the error path:

```
(python38) r@r:~$ sudo pip install --target=/opt/ros/noetic/lib/python3/dist-packages pyyaml
Collecting pyyaml
  Downloading PyYAML-6.0.2-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (746 kB)
     |████████████████████████████████| 746 kB 190 kB/s 
Installing collected packages: pyyaml
Successfully installed pyyaml-6.0.2
```

but then it started giving me a new error:

```
Traceback (most recent call last):
  File "/home/r/openpose/src/ros_openpose/scripts/visualizer.py", line 10, in <module>
    import rospy
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/_Bool.py", line 6, in <module>
    import genpy
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 48, in <module>
    import yaml
  File "/opt/ros/noetic/lib/python3/dist-packages/yaml/__init__.py", line 362
    class YAMLObject(metaclass=YAMLObjectMetaclass):
                                                   ^
SyntaxError: invalid syntax
```
 Is there any good solution to this? Should I consider reinstalling my system and work environment? Thanks.

ravijo, 2025-12-27:
Closing this issue due to no further response. Please feel free to reopen if the problem persists.

closed 2025-12-27 by ravijo